espressif/esp-idf #12634: Order of bonded BLE devices

ztefanjo, 2023-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

There is a maximum of 15 devices (or 8 depending on BTM_SEC_MAX_DEVICE_RECORDS) that can be stored in the database of bonded BLE devices. When the database gets full, the device that was added first is removed when a new device is added (pure FIFO). This is problematic as the first added device might very well be the device that is used the most. It would be much better if a device gets top priority (is moved first in the FIFO list or how to look at it) at every new connection. The device that hasn't been used for the longest time should be the one that is thrown away.

zhp0406, 2023-11-22:
Thank you for your report and your suggestions. We will evaluate them.

zhp0406, 2023-12-05:
@ztefanjo 
This feature has been implemented. Please cherry-pick the following commit or update to the latest version:

Commit: fdc35e772baf9314d9ba1ef9983660eceffaba94
Commit: 061cd291689651f9688c139b7d3c1deb496104ce

ztefanjo, 2023-12-05:
@zhp0406 , great news, thank you!
Closing the issue.

closed 2023-12-05 by ztefanjo