espressif/esp-idf #12635: esp_https_ota_finish does not return errors correctly

huhaifan, 2023-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-89-g5c61c89308

### Espressif SoC revision.

ESP32-S2FNR2 (revision v1.0)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

When esp_ota_set_boot_partition encounters an error, it is desired for esp_https_ota_finish to return correctly.

### What is the actual behavior?

esp_https_ota_finish just return ESP_OK

### Steps to reproduce.

Call OTA without otadata partition

mahavirj, 2023-11-21:
@huhaifan 

Could you please try removing `esp_err_t` prefix from following line in the code:

https://github.com/espressif/esp-idf/blob/a7fbf452fa181452f266b16c49063e2ff069f119/components/esp_https_ota/src/esp_https_ota.c#L668

huhaifan, 2023-11-21:
> @huhaifan
> 
> Could you please try removing `esp_err_t` prefix from following line in the code:
> 
> https://github.com/espressif/esp-idf/blob/a7fbf452fa181452f266b16c49063e2ff069f119/components/esp_https_ota/src/esp_https_ota.c#L668

Yes,I have removed ```esp_err_t``` from my local esp-idf.

closed 2023-11-27 by espressif-bot