espressif/esp-idf #12638: openthread Thread Safety issues

KEEPHEADU, 2023-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I implemented a demo of tcp communication based on openthread using ot_cli example. I used a global variable to store the ip address and port information of tcp, which I initialized in app_main. But when I tried to use this information in another task I found that the data became a jumble.

I wonder if any OpenThread-related function is affecting the value of the global variable?

chip：esp32h2
[esp_ot_cli.txt](https://github.com/espressif/esp-idf/files/13425192/esp_ot_cli.txt)

zwx1995esp, 2023-11-22:
No, for OpenThread-related function, they never affect any value of the global variable. For your code, I see that you create a  `tcp_client_receive_task` task at the line: 255, and pass a NULL to the function `tcp_client_receive_task`
```
 if (pdPASS != xTaskCreate(tcp_client_receive_task, "tcp_client_receive", 4096, tcp_client_member, 4, NULL)) {
 ```
 But in the function of `tcp_client_receive_task`, at the line 171,
 ```
static void tcp_client_receive_task(void *pvParameters)
{
    char rx_buffer[128];
    int len = 0;
    TCP_CLIENT *tcp_client_member = (TCP_CLIENT *)pvParameters;
.....
 ```
  you try to use a NULL pointer which pass in the function. Could this be the cause of your issue?

KEEPHEADU, 2023-11-22:
Thank you for your answer. I checked the function about task creation after your suggestion. In fact, when tcp_socket_client_task was created, the parameter was incorrectly written in the position of the handle. line : 162
```
xTaskCreate(tcp_socket_client_task, "tcp_socket_client_task", 10240,  xTaskGetCurrentTaskHandle(), 10, (void*)&mytcp_client_member);
```

closed 2023-11-22 by KEEPHEADU