espressif/esp-idf #12641: idf.py command line tools only work when provided full path (Windows)

krupis, 2023-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. If I want to use some idf.py commands using command line, I must provide full path and that is quite frustrating.

For example reading adc_info (https://docs.espressif.com/projects/esptool/en/latest/esp32/espefuse/adc-info-cmd.html).

If I try to run command ```espefuse.py adc_info``` in command line, I will get the following error:

```
C:\Users\petrikas.lu>espefuse.py adc_info
'espefuse.py' is not recognized as an internal or external command,
operable program or batch file.
```

However, if I provide full python.exe path and also the tool (in my case espefuse.py) since I want to read adc_info, then it will work (there is some error due to boot mode but atleast the command goes through)

```
C:\Users\petrikas.lu>C:\Users\petrikas.lu\.espressif5\python_env\idf5.0_py3.11_env\Scripts\python.exe C:\Users\petrikas.lu\esp5\esp-idf\components\esptool_py\esptool\espefuse.py -p COM9 adc_info
espefuse.py v4.6.2
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Please make sure that you have specified the right port with the --port argument
```

Does anyone else have this issue? How to fix it so I do not need to provide full path as its really annoying to do that everytime.

dobairoland, 2023-11-21:
Duplicates https://github.com/espressif/esp-idf/issues/9786

krupis, 2023-11-21:
> Duplicates #9786

Ur right. I l close it. My bad!

krupis, 2023-11-21:
Closed due to duplicate

closed 2023-11-21 by krupis