espressif/esp-idf #12642: Hot Reloading

mmackh, 2023-11-21:
### Is your feature request related to a problem?

Changing a single line, compiling and flashing a larger project can take up to 30s with each iteration and adds up over a day. Other programming environments like Android Studio offer hot reloading, increasing productivity by a lot. 

### Describe the solution you'd like.

Please introduce a idf.py command that allows for hot reloading - based on the changes in the binary, write only the relevant flash locations and reboot the system. 

There is already a working prototype located here by cnlohr: https://github.com/cnlohr/esp32s2-cookbook/tree/master/idf_sandbox

A demo video has been posted on twitter as well: https://x.com/cnlohr/status/1546445702636584960?s=12

This would be game changing if it was directly built into esp-idf and supported all different chips.

igrr, 2023-11-21:
Thanks for opening this issue request! That's a cool idea, and something that we have also been discussing internally.

The initial version of this solution based on [esp-elfloader](https://components.espressif.com/components/espressif/elf_loader) will appear in [arduino-esp32](https://github.com/espressif/arduino-esp32) project hopefully within the next few weeks/months. Based on experience and users' feedback, we will consider bringing it into ESP-IDF as well.

chipweinberger, 2023-11-30:
wow that video by cnlohr is amazing

I had no idea it would be that easy to implement. 

this would have saved me months worth of time.

mmackh, 2024-01-30:
Has there been progress in this area?

ammaree, 2024-03-04:
+99 for ESP-IDF v5.x please

chipweinberger, 2024-03-06:
This is the single most important feature ESP-IDF could add. It should have an engineer assigned to it.

and3rson, 2024-03-25:
This would be awesome. Furthermore, it would be great to be able to use elf-loader from arduino-esp32 environment.

Hello1024, 2024-03-29:
One way to achieve this:

  * Update the linker to have an option to take a previous .elf file, and if possible, prefer to locate symbols at the same location.
  
  * Then, with the 'flash' command, provide the previous .bin file too.    Have the command only erase and write places where the files differ.     In most cases the difference should be tiny, because you probably will only be rewriting one function.

chipweinberger, 2024-03-29:
@cnlohr maybe youd like to be in this thread too

cnlohr, 2024-04-03:
I appreciate the ping (I had no idea this thread existed), but don't know if I have anything to offer?  It's how I do most of my ESP32-S2 development recently.  But, it doesn't "fit" into the IDF mentality very well, so I never even considered trying to upstream anything.

mmackh, 2024-09-03:
Has there been any progress on this front?

NotSimple, 2024-10-05:
This would bring a huge increase in productivity.  Those build times really add up, slowing down development quite a bit. Would love to see this

B-l-u-e-f-i-r-e, 2024-12-06:
+1 Can't understand why this is not a thing yet.

alba-ado, 2025-01-03:
+1 As the flash size of the modules keeps increasing, so do the firmware sizes. It takes ages to upload firmware if it includes some assets such as images or fonts. This would be a perfect addition.

playmiel, 2025-03-26:
what is the current status of the project?