espressif/esp-idf #12643: ESP32C3: DR_REG_SPI_BASE should be DR_REG_I2S0_BASE?

moefear85, 2023-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In `soc/esp32c3/include/soc/soc.h:24`
```
#define REG_SPI_MEM_BASE(i)                     (DR_REG_SPI_BASE - (i) * 0x1000)
```

but I'm getting compile errors, because `DR_REG_SPI_BASE` isn't defined anywhere for the C3, unlike 32/S2/S3. What does exist in `components/soc/esp32c3/include/soc/reg_base.h:44`:
```
#define DR_REG_I2S0_BASE                        0x6002D000
```

If I replace `DR_REG_I2S_BASE` with `DR_REG_I2S0_BASE` in soc.h, it compiles fine. Is this an oversight, or am I missing something here?

AxelLin, 2024-01-11:
@L-KAYA Any update?

L-KAYA, 2024-01-11:
Fixing

closed 2024-01-30 by espressif-bot