espressif/esp-idf #12645: ESP32S3: Slow connection to Mesh WPA2 WiFi network

andreywis, 2023-11-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi All,

I recently upgraded a project from the ESP32-WROOM-32D to the ESP32-S3-WROOM-1U-N16R2 module and I am experiencing a slower WiFi connection times.

I am using platformIO and everything is up to date: framework-espidf @ 3.50101.230828 (5.1.1) 

In my code I save the last BSSID and channel and use it for the next connection, this is my code:

`
uint8_t bssid[6] = {
	readNVSIntValue(TAG_NVS, "bssid1"),
	readNVSIntValue(TAG_NVS, "bssid2"),
	readNVSIntValue(TAG_NVS, "bssid3"),
	readNVSIntValue(TAG_NVS, "bssid4"),
	readNVSIntValue(TAG_NVS, "bssid5"),
	readNVSIntValue(TAG_NVS, "bssid6")
};
uint8_t channel = readNVSIntValue(TAG_NVS, "wifi_channel");

wifi_config_t config;
ESP_ERROR_CHECK(esp_wifi_get_config(WIFI_IF_STA, &config));

memcpy(config.sta.bssid, bssid, sizeof(config.sta.bssid));
config.sta.bssid_set = 1;
config.sta.channel = channel;
config.sta.scan_method = 0;

ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &config) );
`

This is my log:
`
I (1663) WIFI: WIFI_EVENT_STA_START
I (1667) wpa: BSS: Add new id 0 BSSID xx:xx:xx:xx:xx:xx SSID 'SSIDName' chan 1
I (1674) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2403) wifi:state: init -> auth (b0)
I (2406) wifi:state: auth -> assoc (0)
I (2410) wifi:state: assoc -> run (10)
I (2425) wifi:connected with SSIDName, aid = 1, channel 1, BW20, bssid = xx:xx:xx:xx:xx:xx
I (2426) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
I (2429) wifi:pm start, type: 0
I (2431) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (2441) WIFI: WIFI_EVENT_STA_CONNECTED
`

Is the ~730ms added related to [PMK generation](https://github.com/espressif/esp-idf/issues/3784#issuecomment-512833624)? 

Could it be related to: [#3784](https://github.com/espressif/esp-idf/issues/3784)?

kapilkedawat, 2023-12-04:
@andreywis what's the IDF commit you are using?

andreywis, 2023-12-04:
Hi @kapilkedawat, I am using platformio 6.4.0 - https://github.com/platformio/platform-espressif32/releases/tag/v6.4.0

It uses esp-idf release 5.1.1 - https://github.com/espressif/esp-idf/releases/tag/v5.1.1

I believe the commit is https://github.com/espressif/esp-idf/commit/e088c3766ba440e72268b458a68f27b6e7d63986

andreywis, 2023-12-04:
Just upgraded to 5.1.2 and it's a lot faster now to connect.

Since I am using platformIO I added this to the platformio.ini:

platform_packages =
        framework-espidf @ https://github.com/tasmota/esp-idf/releases/download/v5.1.2-org/esp-idf-v5.1.2-org.zip

My current consumption in deep sleep now is 94uA, it was 18uA only. But it's probably not related to this issue.

closed 2023-12-04 by andreywis