espressif/esp-idf #12646: If I set wifi auth==wpa ,node can't connect to root in esp mesh network

fzboffice, 2023-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32_DEVKIT_V4

### Power Supply used.

USB

### What is the expected behavior?

node connect to root

### What is the actual behavior?

node can not connect to root

### Steps to reproduce.

I use esp-idf\examples\mesh\internal_communication project.
Call esp_mesh_fix_root(true);
Call esp_mesh_set_type(MESH_NODE);(another development board call call esp_mesh_set_type(MESH_ROOT);)
Set Mesh AP Authentication Mode to CONFIG_WIFI_AUTH_WPA_PSK in "Example Configuration" in menuconfig.
Compile esp-idf\examples\mesh\internal_communication and flash.

### Debug Logs.

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7140
load:0x40078000,len:15500
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080640
I (29) boot: ESP-IDF v5.2-dev-3903-g66992aca7a 2nd stage bootloader
I (29) boot: compile time Nov 23 2023 09:39:41
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
[... 99 lines cut ...]
I (2514) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:7
I (2824) mesh: find root:ESPM_3EECF5, root_cap:1(max:300), new channel:1, old channel:0
I (2824) mesh: [FIND][ch:1]AP:8, otherID:0, MAP:1, idle:0, candidate:0, root:1[00:00:00:00:00:00][FIXED-ROOT]
I (2824) mesh: [FIND:7]find a network, channel:1, cfg<channel:1, router:12345678, 00:00:00:00:00:00>

I (2834) mesh_main: <MESH_EVENT_FIND_NETWORK>new channel:1, router BSSID:00:00:00:00:00:00
I (2834) wifi:mode : sta (c4:4f:33:3e:cb:01) + softAP (c4:4f:33:3e:cb:02)
W (2854) wifi:<MESH AP>adjust channel:1, secondary channel offset:1(40U)
I (2864) wifi:Total power save buffer number: 16
W (2874) wifi:<MESH AP>adjust channel:1, secondary channel offset:1(40U)
I (2884) wifi:Total power save buffer number: 16
I (3184) mesh: [SCAN][ch:1]AP:8, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (3184) mesh: 7370[selection]try rssi_threshold:-78, backoff times:0, max:5<-78,-82,-85>
I (3194) mesh: [DONE]connect to parent:ESPM_3EECF5, channel:1, rssi:-22, c4:4f:33:3e:ec:f6[layer:1, assoc:0], my_vote_num:0/voter_num:0, rc[00:00:00:00:00:00/-120/0]
I (4734) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, non-root, vote(,)<><>
I (4734) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (6234) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, non-root, vote(,)<><>
I (6234) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (7744) mesh: [wifi]disconnected reason:201(), continuous:3/max:12, non-root, vote(,)<><>
I (7744) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (9244) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, non-root, vote(,)<><>
I (9244) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (10744) mesh: [wifi]disconnected reason:201(), continuous:5/max:12, non-root, vote(,)<><>
I (10744) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
```

fzboffice, 2023-11-23:
If I set wifi auth==WPA/WPA2, it works.

fzboffice, 2023-11-23:
This issue does not seem to be caused by the auth mode. The error with this issue is that my AP password is randomly selected between "!" and "~". When I am randomly selected between "0" and "9", no matter which authentication method is used, it can work normally.

fzboffice, 2023-11-23:
update!
I can connect when password include '0'-'9' and auth is wpa/wpa2
I can't connect when password include '!'-'~' and auth is wpa/wpa2
I can't connect when auth is wpa,no matter what password

zhangyanjiaoesp, 2023-11-23:
@fzboffice Can you reproduce it just running the `internal_communication` project? I have tested on the `internal_communication` example, and set the password to `mypass!word`, `mypass~word`, `!!!!~~~~`, `!~!~!~!~`, they all can connect successfully.

fzboffice, 2023-11-23:
@zhangyanjiaoesp not just '!''~',it is start from'!' to '~'.

fzboffice, 2023-11-23:
@zhangyanjiaoesp Do you set auth mode to wpa?

zhangyanjiaoesp, 2023-11-24:
> @zhangyanjiaoesp Do you set auth mode to wpa?

Just test with the authmode set to WPA, and the node device can't found the AP indeed. And we will solve it ASAP.

fzboffice, 2023-12-05:
@zhangyanjiaoesp Has this issue been solved?

zhangyanjiaoesp, 2023-12-05:
The fix has been merged internally, and it will take some time to sync to Github.

closed 2023-12-06 by espressif-bot