espressif/esp-idf #12647: 为什么ble mesh订阅了指定的组地址只会收到一条publish message

zksite, 2023-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

这是我的订阅代码
`ESP_ERROR_CHECK(
                esp_ble_mesh_model_subscribe_group_addr(esp_ble_mesh_get_primary_element_address(), VND_CID,
                                                        VND_MODEL_ID_CLIENT, VND_GROUP_EXE_RULE));`

下面是回调
`static void ble_mesh_custom_model_cb(esp_ble_mesh_model_cb_event_t event,
                                          esp_ble_mesh_model_cb_param_t *param) {
    switch (event) {
        case ESP_BLE_MESH_CLIENT_MODEL_RECV_PUBLISH_MSG_EVT: {
            uint16_t *addr = esp_ble_mesh_is_model_subscribed_to_group(&vnd_models[0], VND_GROUP_EXE_RULE);
            ESP_LOGI(VND_MESH, "ESP_BLE_MESH_CLIENT_MODEL_RECV_PUBLISH_MSG_EVT %d,%d",
                     param->client_recv_publish_msg.ctx->addr, *addr);
        }
        default:
            break;
    }
}`
我在其它节点发布订阅消息，只会收到一条，只有我重启该订阅节点才能再次收到一条消息。这个VND_MODEL_ID_CLIENT模型没有做任何与订阅相关的配置，直接订阅的组地址，如果需要每次都能收到发布的消息，需要做其它额外的配置吗？

forx157, 2023-11-23:
Please set the log level of BLE Mesh to INFO following the path below and provide me with the device logs:
`(Top) → Component config → ESP BLE Mesh Support → BLE Mesh STACK DEBUG LOG LEVEL → BLE_MESH_STACK`

zksite, 2023-11-23:
**subscriber node logs:**
I (808) VN_MESH: bluetooth_init success
I (868) BLE_MESH: Restored mesh device role: Node
I (868) BLE_MESH: Primary address 0x0063, element count 1
I (868) BLE_MESH: Restored Primary Address 0x0063
I (868) BLE_MESH: Restored DevKey dd5fb97f4d565c0f17ded95d2dca30bb
I (878) BLE_MESH: Restored IV Index 0x00000000 (IV Update Flag 0) duration 96 hours
I (888) BLE_MESH: Restored Sequence Number 0x00008c
I (898) BLE_MESH: Restored RPL entry 0x0001: seq 0x000a17, old_iv 0
I (898) BLE_MESH: Restored RPL entry 0x0066: seq 0x000009, old_iv 0
I (908) BLE_MESH: Restored RPL entry 0x006e: seq 0x00000a, old_iv 0
I (908) BLE_MESH: Restored RPL entry 0x006f: seq 0x000009, old_iv 0
I (918) BLE_MESH: Restored RPL entry 0x0073: seq 0x00002f, old_iv 0
I (928) BLE_MESH: Restored NetKeyIndex 0x000
I (928) BLE_MESH: Restored NetKey 869916606fe3d53b818ca363b5d4f0e5
I (938) BLE_MESH: Restored AppKeyIndex 0x000, NetKeyIndex 0x000
I (948) BLE_MESH: Restored AppKey 0aa7a702dffba66bdf9a7793661ac00f
I (958) BLE_MESH: Restored Model Bound AppKey, index 0000ffffffff
I (958) BLE_MESH: Restored Model Subscription, address 01c000000000
I (968) BLE_MESH: Restored Model Bound AppKey, index 0000ffffffff
I (968) BLE_MESH: Restored Model Subscription, address 01c000000000
I (978) BLE_MESH: Not support publication, model_id 0x0000, cid 0x02e5
I (988) BLE_MESH: Settings commit, sub[0].net_idx 0x000
W (988) BLE_MESH: bt_mesh_proxy_server_prov_disable, Already
I (1008) VN_MESH: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (1008) VN_MESH: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (1008) BLE_MESH: Group address 0xc001 already exists
I (1018) main_task: Returned from app_main()
I (9398) BLE_MESH: net_idx 0x000 iv_index 0x00000000 current iv_index 0x00000000
I (20998) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (51008) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (61008) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (71018) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (81018) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (84468) BLE_MESH: Settings store, waiting 0 seconds
I (84488) BLE_MESH: recv, app_idx 0x0000 src 0x0073 dst 0xc001
I (84488) BLE_MESH: recv, len 7: d0e50200010200
I (84488) VN_MESH: ESP_BLE_MESH_CLIENT_MODEL_RECV_PUBLISH_MSG_EVT 115,49153
I (91028) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (101028) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (131038) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (141038) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00

**publish node logs:**
I (551) EXAMPLE_NVS: Open namespace done, name "mesh_example"
I (611) BLE_MESH: Restored mesh device role: Node
I (611) BLE_MESH: Primary address 0x0073, element count 1
I (611) BLE_MESH: Restored Primary Address 0x0073
I (621) BLE_MESH: Restored DevKey d944a67cf5fcfb7d8f5889338bdf9636
I (621) BLE_MESH: Restored IV Index 0x00000000 (IV Update Flag 0) duration 96 hours
I (631) BLE_MESH: Restored Sequence Number 0x000037
I (641) BLE_MESH: Restored RPL entry 0x0001: seq 0x000a40, old_iv 0
I (641) BLE_MESH: Restored NetKeyIndex 0x000
I (651) BLE_MESH: Restored NetKey 869916606fe3d53b818ca363b5d4f0e5
I (661) BLE_MESH: Restored AppKeyIndex 0x000, NetKeyIndex 0x000
I (661) BLE_MESH: Restored AppKey 0aa7a702dffba66bdf9a7793661ac00f
I (671) BLE_MESH: Restored Model Bound AppKey, index 0000ffffffff
I (681) BLE_MESH: Settings commit, sub[0].net_idx 0x000
W (681) BLE_MESH: bt_mesh_proxy_server_prov_disable, Already
I (701) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (701) EXAMPLE: net_idx: 0x0000, addr: 0x0073
I (701) EXAMPLE: flags: 0x00, iv_index: 0x00000000
W (701) BOARD: led green is already off
I (711) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (721) EXAMPLE_NVS: Restore, key "onoff_client" not exists
W (721) BLE_MESH: bt_mesh_prov_enable, Already
I (731) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code -120
I (731) EXAMPLE: BLE Mesh Node initialized
I (741) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
I (751) BUTTON: Registering button input: 9
I (10441) BLE_MESH: net_idx 0x000 iv_index 0x00000000 current iv_index 0x00000000
I (13661) BUTTON: 9 DOWN
I (13831) BUTTON: 9 UP
I (13831) BLE_MESH: Publish Retransmit Count 0 Interval 50ms
I (13831) BLE_MESH: send, app_idx 0x0000 src 0x0073 dst 0xc001
I (13831) BLE_MESH: send, len 7: d0e50200010200
I (13841) BLE_MESH: Settings store, waiting 0 seconds
W (13851) EXAMPLE: ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT
I (20561) BUTTON: 9 DOWN
I (20691) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (20741) BUTTON: 9 UP
I (20741) BLE_MESH: Publish Retransmit Count 0 Interval 50ms
I (20741) BLE_MESH: send, app_idx 0x0000 src 0x0073 dst 0xc001
I (20741) BLE_MESH: send, len 7: d0e50201010201
I (20751) BLE_MESH: Settings store, waiting 0 seconds
W (20761) EXAMPLE: ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT
I (33381) BUTTON: 9 DOWN
I (33611) BUTTON: 9 UP
I (33611) BLE_MESH: Publish Retransmit Count 0 Interval 50ms
I (33611) BLE_MESH: send, app_idx 0x0000 src 0x0073 dst 0xc001
I (33611) BLE_MESH: send, len 7: d0e50200010202
I (33621) BLE_MESH: Settings store, waiting 0 seconds
W (33631) EXAMPLE: ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT
I (50701) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (60701) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (90701) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (100711) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (120711) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (150721) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
I (160721) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00

forx157, 2023-11-24:
hi, @zksite 
I conducted a test using the vendor model after making the modifications included in the attachment and found that the nodes can send and receive messages properly. I have included my changes in the attachment for your reference.
[0001-test-ble_mesh-publish-and-subscribe-test.zip](https://github.com/espressif/esp-idf/files/13456039/0001-test-ble_mesh-publish-and-subscribe-test.zip)

zksite, 2023-11-24:
Thanks your code @forx157 . Using your example to run in the official project is not a problem. I tried to copy the code to my project, but the same problem still exists. Is this related to the configuration of sdkconfig?

zksite, 2023-11-25:
I copied the official configuration and tested it again without any problems. Thank you

closed 2023-11-25 by zksite