espressif/esp-idf #12648: ESP32-H2 Deep Sleep Wakeup Resources

Bosemani, 2023-11-23:
Hai 
ESP-IDF Version V.5.1.2
Chip: ESP32-H2

I want to implement deep sleep in esp32-h2. I tried Deep Sleep Example wakeup using timer, its working fine.
I'm not able to Enable GPIO wakeup in esp32-h2.

**what are the wakeup mode esp32-h2 support? (Like Timer, EXT0, EXT1, GPIO..)**
**How to enable GPIO wakeup?**

10086loutianhao, 2023-11-23:
hello, GPIO wakeup can be used in ESP32H2 when deepsleep, May I ask where you found information stating that the H2 can be woken up from deep sleep using GPIO wakeup?

10086loutianhao, 2023-11-23:
esp-32h2 support GPIO wakeup, EXT1 wakeup and LP timer wakeup when lightsleep, and support EXT1 wakeup and LP timer wakeup when deepsleep.

Bosemani, 2023-11-23:
> hello, GPIO wakeup can be used in ESP32H2 when deepsleep, May I ask where you found information stating that the H2 can be woken up from deep sleep using GPIO wakeup?

Hai @10086loutianhao 
 In deep sleep example, [Readme file ](https://github.com/espressif/esp-idf/tree/master/examples/system/deep_sleep#:~:text=GPIO%3A%20Pads%20powered%20by%20VDD3P3_RTC%20can%20be%20used%20to%20trigger%20a%20wake%20up%20from%20deep%20sleep.%20You%20may%20choose%20the%20pin%20and%20trigger%20level%20in%20menuconfig.%20(This%20wake%20up%20source%20is%20unavailable%20on%20ESP32%2C%20ESP32%2DS2%2C%20and%20ESP32%2DS3.))  `(This wake-up source is unavailable on ESP32, ESP32-S2, and ESP32-S3.)` In this H2 was not mentioned. So, I thought It will support.

Bosemani, 2023-11-23:
> esp-32h2 support GPIO wakeup, EXT1 wakeup and LP timer wakeup when lightsleep, and support EXT1 wakeup and LP timer wakeup when deepsleep.

Hai @10086loutianhao 
  Thank you for your Information.
I want to use wakeup using EXT1 from deep sleep mode. What are the gpio can use for that?
We already designed Board for GPIO_11 and one more board with GPIO_5

10086loutianhao, 2023-11-23:
> Hai @10086loutianhao Thank you for your Information. I want to use wakeup using EXT1 from deep sleep mode. What are the gpio can use for that? We already designed Board for GPIO_11 and one more [...]

hello, you can choose GPIO7-14 to use EXT1, but GPIO_5 is unavailable. Please refer to deepsleep example, and you can use idf.py menuconfig ->Example Configuration ->EXT1 wakeup configuration

10086loutianhao, 2023-11-23:
> Hai @10086loutianhao In deep sleep example, [Readme file [...]

Thanks

Bosemani, 2023-11-23:
> hello, you can choose GPIO7-14 to use EXT1, but GPIO_5 is unavailable. Please refer to deepsleep example, and you can use idf.py menuconfig ->Example Configuration ->EXT1 wakeup configuration

Hai @10086loutianhao 
  Thank you for your quick support.
I tried Deep sleep example with wakeup source EXT1. 
It's working. Thank you. I'm closing this.

closed 2023-11-23 by Bosemani