espressif/esp-idf #12650: ESP32 hci controller timeout in host

MrSumi, 2023-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

ESP32 run hci controller example(BTDM mode), and communicate with BlueZ in Linux via 4 wried UART.

### What is the actual behavior?

The ESP32 successfully attach with `btattach -B /dev/ttymxc0 -S 921600 -P h4` and work fine in the beginning.
But after running a few minutes or hours, it returns `Can't read local name on hci0: Connection timed out (110)` with `hciconfig -a`.

[image]

### Steps to reproduce.

1. ESP32 run the [hci controller example in idf-v5.1.1](https://github.com/espressif/esp-idf/tree/release/v5.1/examples/bluetooth/hci/controller_hci_uart_esp32).
2. Linux host attach the ESP32 `btattach -B /dev/ttymxc0 -S 921600 -P h4 &`, and set hci up `hciconfig hci0 up`.
3. Linux host setup a GATT service and characteristic, and start adv using `bluetoothctl` (here use the BLE).
4. Linux host try query a BT device's name `hcitool name <MAC>`, and connect its NAP server `bt-network -c <MAC> nap &` per five seconds (here use the Classic BT).
5. Use another ESP device(in my case, ESP32S3) run the [gatt client example in idf-v5.1.1](https://github.com/espressif/esp-idf/tree/release/v5.1/examples/bluetooth/bluedroid/ble/gatt_client), and try connect the Linux Host and write/read the GATT characteristic.

### Debug Logs.

```plain
### Here's the Linux host `hciconfig -a` returns ###

[root@superx5a:~]#
[root@superx5a:~]#
[root@superx5a:~]# hciconfig -a
hci0:   Type: Primary  Bus: UART
        BD Address: A0:B7:65:4A:93:AA  ACL MTU: 1021:9  SCO MTU: 255:4
        UP RUNNING PSCAN ISCAN
        RX bytes:7514 acl:52 sco:0 events:225 errors:0
        TX bytes:3789 acl:53 sco:0 commands:127 errors:0
        Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV3
        Link policy: RSWITCH SNIFF
        Link mode: PERIPHERAL ACCEPT
        Name: 'APC2_320E99'
        Class: 0x000108
        Service Classes: Unspecified
        Device Class: Computer, Server
        HCI Version: 4.2 (0x8)  Revision: 0x30e
        LMP Version: 4.2 (0x8)  Subversion: 0x30e
        Manufacturer: RivieraWaves S.A.S (96)

[root@superx5a:~]# hciconfig -a
hci0:   Type: Primary  Bus: UART
        BD Address: A0:B7:65:4A:93:AA  ACL MTU: 1021:9  SCO MTU: 255:4
        UP RUNNING PSCAN ISCAN
        RX bytes:24834 acl:186 sco:0 events:591 errors:0
        TX bytes:10679 acl:191 sco:0 commands:275 errors:0
        Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV3
        Link policy: RSWITCH SNIFF
        Link mode: PERIPHERAL ACCEPT
        Name: 'APC2_320E99'
        Class: 0x000108
        Service Classes: Unspecified
        Device Class: Computer, Server
        HCI Version: 4.2 (0x8)  Revision: 0x30e
        LMP Version: 4.2 (0x8)  Subversion: 0x30e
        Manufacturer: RivieraWaves S.A.S (96)

[root@superx5a:~]# hciconfig -a
hci0:   Type: Primary  Bus: UART
        BD Address: A0:B7:65:4A:93:AA  ACL MTU: 1021:9  SCO MTU: 255:4
        UP RUNNING PSCAN ISCAN
        RX bytes:85437 acl:504 sco:0 events:950 errors:0
        TX bytes:65756 acl:509 sco:0 commands:310 errors:0
        Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV3
        Link policy: RSWITCH SNIFF
        Link mode: PERIPHERAL ACCEPT
Can't read local name on hci0: Connection timed out (110)
[root@superx5a:~]# hciconfig hci0 reset
Can't init device hci0: Cannot allocate memory (12)
[root@superx5a:~]#
```

### More Information.

Here's the ESP32 serial output:
[ESP32 Serial Debug 2023-11-23 121310.txt](https://github.com/espressif/esp-idf/files/13446761/ESP32.Serial.Debug.2023-11-23.121310.txt)

Here's the Linux host `btmon` output:
[btmon_20231123_121234.txt](https://github.com/espressif/esp-idf/files/13446764/btmon_20231123_121234.txt)

Here's the Linux host `hciconfig -a` output:
[hciconfig_20231123_121253.txt](https://github.com/espressif/esp-idf/files/13446779/hciconfig_20231123_121253.txt)

Actually, i had created an issue in [esp-hosted](https://github.com/espressif/esp-hosted/issues/279).
But i found the same problem while using [esp-idf example](https://github.com/espressif/esp-idf/tree/release/v5.1/examples/bluetooth/hci/controller_hci_uart_esp32) for further test.

mantriyogesh, 2023-11-23:
ESP-Hosted reference : https://github.com/espressif/esp-hosted/issues/279

zhp0406, 2023-11-24:
Thank you for reporting the issue; I will check it.

zhp0406, 2023-11-27:
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/13475932/libbtdm_app.zip)

This is a fixed library. The usage involves replacing the file "esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a" with this updated library.

Both of the methods mentioned above are aimed at resolving the issues caused by the "Read Link Quality" command. Kindly conduct some testing and share your feedback.

thanks!

MrSumi, 2023-11-28:
Hi @zhp0406 

I had tried your fixed library, but it seems to cause a core crash.

### Logs
ESP32 serial output:
[ESP32 Serial Debug 2023-11-28 102942.txt](https://github.com/espressif/esp-idf/files/13482816/ESP32.Serial.Debug.2023-11-28.102942.txt)

Linux host `btmon` output:
[btmon_20231128_102842.txt](https://github.com/espressif/esp-idf/files/13482824/btmon_20231128_102842.txt)

Linux host `hciconfig -a` output:
[hciconfig_20231128_102930.txt](https://github.com/espressif/esp-idf/files/13482832/hciconfig_20231128_102930.txt)

zhp0406, 2023-11-28:
@MrSumi 
I will check the cause of the crash.Have you tried prohibiting the host from sending "Read Link Quality" to the controller when using BLE?

MrSumi, 2023-11-30:
Hi @zhp0406 

I guess the "Read Link Quality" may send to the controller by calling `hcitool lq` with BLE address.
Now, i call `hcitool lq` only with Classic BT address, and it works fine in the last 12 hours.

To ensure the problem is solved, i will do futher testing.

thinks a lot!

zhp0406, 2023-11-30:
@MrSumi 
Looking forward to your further feedback.

Indastri, 2024-01-09:
Since there was no response from the user closing this issue. If the issue still persists please feel free to reopen this issue or create a new one. Cheers.

closed 2024-01-09 by Indastri