espressif/esp-idf #12653: 0 ohms resistor not mentionned in ESP S3 DevKitM 1 documentation

imabot2, 2023-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm not sure to be at the right place, but I noticed a 0 Ohm resistor on the  ESP-S3-DevKitM-1 schematics. I didn't find any mention about the purpose of this resistor in the documentation. Could someone tell me what this resistor is made for ?

imabot2, 2023-11-28:
Thank you for the reply. I know a 0 Ohm resistor is a wire. My question is what configuration does it change on the ESP32-S3-DevKitM-1 ?

atemiz, 2023-11-28:
For example, GPIO48 is present in the pin header and is also connected to the RGB LED. If you desolder R17, you can get rid of the connection with this LED.
For the connection between USB-UART Bridge and ESP32, R7 (RX) and R9 (TX) are used.
R24 may be for insulation purposes. By desoldering this, it is possible to power only ESP32-S3-MINI-1 via the ESP_3V3 pin on the header.

ESP-Marius, 2024-04-18:
I think we can consider this ticket closed, as it is not really related to ESP-IDF.

Thanks to @atemiz for providing the answer!

closed 2024-04-18 by ESP-Marius