espressif/esp-idf #12654: register_chipv7_phy does not return, and makes the watchdog to call panic()

fbucafusco, 2023-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.3.0

### Espressif SoC revision.

ESP32-D0WD (revision v1.0)

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

A bit of context:
- we have a partition called factory image and an application partition
- we have a custom bootloader, that decides which partition to run. Basically, when there are too many non-controlled restarts while running the app, the factory image kicks in. The factory image has the ota process running, so when running it allows to write the app partition.
- both the factory image and app use the idf 4.3 but the factory image is being built using Platformio and the app uses the idf.

When the esp is first flashed, only with the bl + fi partitions, everything goes well. The fi runs, the ota process works fine, etc.
When you load an application and the application calls esp_restart() for some reason, then the BL decides whether to start the FI or the app again. 

I made an app image that deliberately calls to esp_restart();
After some tries, the FI should be running again allowing us to OTA the app with a diff image.

### What is the actual behavior?

When the FI runs again after, what happens is that the watchdog kicks and the system gets restarted (I configured the watchdog to call the panic() handler)
I dug into and I found that the execution is being stuck when
esp_phy_enable() -> esp_phy_load_cal_and_init()  -> register_chipv7_phy() is called.

### Steps to reproduce.

I don't have any way of sharing my binaries. 

I configure both FI and APP so the PHY is always configured to call 
register_chipv7_phy(init_data, cal_data, PHY_RF_CAL_FULL);

Cal data is not stored in the NVS, so I don;t have the partition for that.

### Debug Logs.

```plain
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[... 104 lines cut ...]

0x400820a9: _xt_lowint1 at xtensa_vectors.o:?

0x40087245: esp_dport_access_reg_read at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\esp32/dport_access.c:259

0x40125669: ram_check_noise_floor at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:2291

0x40126db5: noise_init at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:2433

0x40127363: bb_init at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:2796

0x40127d7e: register_chipv7_phy at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:3782

0x400db4c5: esp_phy_load_cal_and_init at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\esp_wifi\src/phy_init.c:669 (discriminator 13)

0x400db567: esp_phy_enable at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\esp_wifi\src/phy_init.c:242

0x4010c03a: wifi_hw_start at ??:?

0x4010c7a9: wifi_start_process at ??:?

0x4010b614: ieee80211_ioctl_process at ??:?

0x40091fbd: ppTask at ??:?
```

MaxwellAlan, 2023-11-29:
Hi @fbucafusco 

From the logs, we speculate that there might be some incorrect hardware states, causing the check to fail during the software restart process. 
As for the root cause, it is currently unknown, since we can not reproduce this issue locally, so it's difficult for us to debug now.

Could you try test with the latest release/v4.3 ? maybe some phy lib issue had been fixed.

fbucafusco, 2023-12-04:
Sorry for the delay.
Yeah, my only conclusion is also that some hw state provokes that. I'm sorry about not providing more info on how to reproduce it.
The problem is that my BL firmware is being built with [this version of idf](https://registry.platformio.org/platforms/platformio/espressif32?version=4.4.0) And that is the latest that platformio supports for 4.xxxx. I tried to upgrade everything to a newer version, but I have to make too many changes and currently, I don't have too much time. 
I also tried to update (by hand) the .a files in the phy and wifi lib (with [this](https://github.com/espressif/esp32-wifi-lib) and [this ](https://github.com/espressif/esp-phy-lib)), and I had a ton of issues, so I changed that back. 

I will try to make it reproducible with a simple code, and let you know. 
If you have any other advice on what to do, just tell me cos I am pretty lost about this. Since the phy and wifi libs are prebuilt, I am pretty stucked here. In general, I try to dig more into any code before posting some bug...

MaxwellAlan, 2023-12-08:
@fbucafusco 

Thanks for providing the details, we'll check related process.

MaxwellAlan, 2023-12-08:
@fbucafusco 

Could you help do below test?

1.  replace  $IDF_PATH/components/esp_wifi/lib/esp32/libphy.a with new libphy version 4780 [libphy.zip](https://github.com/espressif/esp-idf/files/13608573/libphy.zip), check the new phy still have the same issue.
2. If the new phy lib still have the same issue, could you add below code in esp_phy_load_cal_and_init func before register_chipv7_phy called.
`extern void phy_reg_check(void);`
`phy_reg_check();`
[image]
and then provide the full fail logs for we to analyse.
Thanks.

fbucafusco, 2023-12-08:
``Hi @MaxwellAlan 

I tried to replace the phy library, and the execution got a bit further, but stucks the same, after. This is the call stack (it is also shown in the logs below.):

[image]

Then I added the lines for getting the registers' state and this is the log:

`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7940
load:0x40078000,len:14412
load:0x40080400,len:4124
0x40080400: _init at ??:?

entry 0x400806a8
I (27) boot: ESP-IDF 4.3.2 2nd stage bootloader
I (27) boot: compile time 14:21:22
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data (Arduino FW forces this) 01 02 00009000 00005000
I (76) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (83) boot:  2 app0             OTA app          00 10 00010000 001f0000
I (91) boot:  3 factory          factory app      00 00 00200000 001f0000
I (98) boot:  4 nvs_storage      Other user NVS   01 02 003f0000 00010000
I (106) boot: End of partition table
I (110) boot: Reset Reason: 12 Boot CMD 0 attempts: 3
I (116) boot: Unstable app
I (119) boot_comm: chip revision: 1, min. application chip revision: 0
I (126) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=267a0h (157600) map
I (192) esp_image: segment 1: paddr=002267c8 vaddr=3ffb0000 size=04458h ( 17496) load
I (199) esp_image: segment 2: paddr=0022ac28 vaddr=40080000 size=053f0h ( 21488) load
I (208) esp_image: segment 3: paddr=00230020 vaddr=400d0020 size=9cf90h (642960) map
I (441) esp_image: segment 4: paddr=002ccfb8 vaddr=400853f0 size=118fch ( 71932) load
I (471) esp_image: segment 5: paddr=002de8bc vaddr=50000000 size=00010h (    16) load
I (482) boot: Loaded app from partition at offset 0x200000
I (482) boot: Disabling RNG early entropy source...
I (494) cpu_start: Pro cpu up.
I (494) cpu_start: Starting app cpu, entry point is 0x400813dc
0x400813dc: call_start_cpu1 at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\esp_system\port/cpu_start.c:150

I (481) cpu_start: App cpu up.
I (509) cpu_start: Pro cpu start user code
I (509) cpu_start: cpu freq: 160000000
I (509) cpu_start: Application information:
I (513) cpu_start: Project name:     factory_image
I (519) cpu_start: App version:      demo_bootloop-213-g9474af7-dirt
I (526) cpu_start: Compile time:     Dec  7 2023 10:32:38
I (532) cpu_start: ELF file SHA256:  d86a06d005454859...
I (538) cpu_start: ESP-IDF:          4.3.2
I (543) heap_init: Initializing. RAM available for dynamic allocation:
I (550) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (556) heap_init: At 3FFB83B8 len 00027C48 (159 KiB): DRAM
I (562) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (568) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (575) heap_init: At 40096CEC len 00009314 (36 KiB): IRAM
I (582) spi_flash: detected chip: gd
I (585) spi_flash: flash io: dio
W (589) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (604) cpu_start: Starting scheduler on PRO CPU.
[... 472 lines cut ...]
E (5763) task_wdt: Tasks currently running:
E (5763) task_wdt: CPU 0: wifi
E (5763) task_wdt: CPU 1: IDLE
E (5763) task_wdt: Aborting.

abort() was called at PC 0x401377f4 on core 0
0x401377f4: task_wdt_isr at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\esp_common\src/task_wdt.c:182 (discriminator 1)

Backtrace:0x400816db:0x3ffb07b0 0x4008a705:0x3ffb07d0 0x400913e2:0x3ffb07f0 0x401377f4:0x3ffb0860 0x40082089:0x3ffb0880 0x40126f06:0x3ffc1d60 0x40120293:0x3ffc1d80 0x401103ad:0x3ffc1da0 0x40110b01:0x3ffc1dc0 0x4010f964:0x3ffc1de0 0x40092e71:0x3ffc1e00
0x400816db: panic_abort at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\esp_system/panic.c:393

0x4008a705: esp_system_abort at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\esp_system/system_api.c:112

0x400913e2: abort at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\newlib/abort.c:46

0x401377f4: task_wdt_isr at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\esp_common\src/task_wdt.c:182 (discriminator 1)

0x40082089: _xt_lowint1 at C:\Users\Franco\.platformio\packages\framework-espidf@3.40302.0\components\freertos\port\xtensa/xtensa_vectors.S:1105

0x40126f06: hal_init at ??:?
0x40120293: ic_enable at ??:?
0x401103ad: wifi_hw_start at ??:?
0x40110b01: wifi_start_process at ??:?
0x4010f964: ieee80211_ioctl_process at ??:?
0x40092e71: ppTask at ??:?`

MaxwellAlan, 2023-12-13:
Hi @fbucafusco 

From the phy logs, it seems that some PHY modules Clock not enabled.

Could you add below code before phy_reg_check called, we need to check related module's clock status.

`#include "soc/dport_reg.h"`
`#include "soc/dport_access.h"`
`printf("periph clk %08x \n", DPORT_READ_PERI_REG(DPORT_WIFI_CLK_EN_REG)); // to see the status`

then go on phy_reg_check ...

Thanks.

fbucafusco, 2023-12-14:
Hi!

The log is the same, so I only copy the new line added. 

```I (873) phy_init: periph clk fffc87cf```  

The weird thing is that that hexa value fffc87cf  is the same in both situations: when it works and when it doesn't.

MaxwellAlan, 2023-12-18:
Hi @fbucafusco 

From the logs you provide, we have eliminated the possibility of a clock-related issue, and upload a new phy debug lib: [libphy_ESP32_20231215_debug.zip](https://github.com/espressif/esp-idf/files/13703431/libphy_ESP32_20231215_debug.zip)

please re-test and upload the full test logs.

Thanks.

fbucafusco, 2023-12-27:
Hi Again! Sorry for the delay..... Christmas happened.

There you have the new log. 

[nonworking_2.txt](https://github.com/espressif/esp-idf/files/13776853/nonworking_2.txt)

Thanks

MaxwellAlan, 2023-12-27:
Hi @fbucafusco 

From the logs, we suspect the root cause is power down phy related digital modules.

Could you add below two line to force power up before esp_phy_load_cal_and_init
`CLEAR_PERI_REG_MASK(RTC_CNTL_DIG_PWC_REG, RTC_CNTL_WIFI_FORCE_PD);
    CLEAR_PERI_REG_MASK(RTC_CNTL_DIG_ISO_REG, RTC_CNTL_WIFI_FORCE_ISO);`

[image]

fbucafusco, 2023-12-27:
Hi again! 
Seems to be working now! 
I also tested the original `lib_phy.a` and also worked.
So what I am going to do is to create a patch for `phy_init.c.`
It seems that 4.3.0 is not gonna have any more support, so I guess I will have to move to 4.4.0 or even further.
:) 
Thanks for the help!

fbucafusco, 2024-01-11:
just closing this, you were really helpful Thanks

closed 2024-01-11 by fbucafusco

AxelLin, 2024-01-26:
@MaxwellAlan 
Is this issue fixed in esp-idf?

MaxwellAlan, 2024-01-26:
Hi @AxelLin 

We have OTA test in IDF release flow, OTA from different IDF version and roll back, including customer used version,but we can not reproduce customer's issue, so we actually not know how this issue happened, maybe related with the customer's bootloader or something else, only this one case.
So we not have this patch in IDF now.