espressif/esp-idf #12655: When initializing WiFi AP mode not letting anything connect

andresragot, 2023-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.1 & 5.1.2

### Espressif SoC revision.

ESP32-WROVER-IE

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

When declaring and initializing the `wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();`

and setting the explicit number of connections to the `wifi_config_t` in here:
`esp_wifi_set_mode(WIFI_MODE_AP);

        wifi_config_t wifi_config_ap = { 0 };
        wifi_config_ap.ap.ssid_hidden = 0;
        wifi_config_ap.ap.channel = 0;
        wifi_config_ap.ap.max_connection = 3;
        wifi_config_ap.ap.beacon_interval = 100;
        wifi_config_ap.ap.pmf_cfg.capable = true;
        wifi_config_ap.ap.pmf_cfg.required = true;
        wifi_config_ap.ap.authmode = WIFI_AUTH_WPA2_PSK;
        memset(&wifi_config_ap.ap.ssid, 0, sizeof(wifi_config_ap.ap.ssid) * sizeof(char));
        memset(&wifi_config_ap.ap.password, 0, sizeof(wifi_config_ap.ap.password) * sizeof(char));

        memcpy(wifi_config_ap.ap.ssid, nvs_struct.ap_ssid, strlen(nvs_struct.ap_ssid));
        memcpy(wifi_config_ap.ap.password, nvs_struct.ap_pass, strlen(nvs_struct.ap_pass));
        esp_wifi_set_config(WIFI_IF_AP, &wifi_config_ap);`

gave this error in the CMD running the code in debug.
`W (1661) wifi:Affected by the ESP-NOW encrypt num, set the max connection num to 0`
**"FIXED"**
adding this line of code:

` wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
cfg.espnow_max_encrypt_num = 0;`

### What is the actual behavior?

When connecting from another devide it doesn't let you connect.
Gives error "Wrong Password" or "Could not connect"

### Steps to reproduce.
[... 131 lines cut ...]
I (1631) wifi_init: tcp mss: 1440
I (1641) wifi_init: WiFi/LWIP prefer SPIRAM
I (1641) wifi_init: WiFi IRAM OP enabled
I (1651) SF:    > Starting WiFi...
I (1651) wifi:Set ps type: 0, coexist: 0

I (1661) SF: AP_SSID _SingularFlower_
I (1661) SF: AP_PASS _12345678_
W (1661) wifi:Affected by the ESP-NOW encrypt num, set the max connection num to 0
I (1841) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16
I (1901) wifi:mode : softAP (90:38:0c:aa:d8:a5)
I (1901) wifi:Total power save buffer number: 5
I (1901) wifi:Init max length of beacon: 752/752
I (1901) wifi:Init max length of beacon: 752/752
W (1911) SF: ****************************************** CORE
I (1911) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.29.1
I (1921) SF: WIFI_EVENT_AP_START
I (1921) SF: > Initializating 8 flowers...
I (1931) SF: > Entering main loop...
```

### More Information.

We have tried in a UNIX system and the same problem.
Other ESP32-WROVER-IE chips and the problem is the same.

Xiehanxin, 2023-12-06:
hi @andresragot , I try your ap setting in softAP example, sorry i can't reprudoce your issue, could you provide the a minimal reproduce code or clarify if there are other use cases besides AP?

kepler0308, 2023-12-22:
Hello brother,Have you resolved this issue?

closed 2024-03-22 by Alvin1Zhang