espressif/esp-idf #12657: esp32c6 cannot join Zigbee network

valdisz, 2023-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm not able to join the Zigbee network. I have tried several demos, my own code, and different coordinators.

Here is a summary:

An end device (esp32c6) is trying to join the network. I'm using `HA_on_off_light` demo code, but also my own code or different examples behave the same. Here is a typical log example from the end device:

```
I (385) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (385) phy_init: phy_version 230,9f68e96,Oct  8 2023,10:40:34
I (435) phy: libbtbb version: 90c587c, Oct  8 2023, 10:41:21
I (435) main_task: Returned from app_main()
I (445) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (445) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (445) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (19995) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (20005) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (41755) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (41765) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (63525) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (63535) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (85295) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (85305) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (107035) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (107045) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (128795) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (128805) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (149465) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (149475) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (171215) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (171225) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (213995) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (256795) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
```
It starts the network steering, and then it fails.

On the other end I was using `HA_on_off_switch` demo code and Sonoff Zigbee Bridge (flashed with Tasmota) + Home Assistant.

The `HA_on_off_switch` gives out such log:
```
I (469) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_OK
I (469) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
W (469) ESP_ZB_ON_OFF_SWITCH: Network(0x2c26) closed, devices joining not allowed.
I (479) ESP_ZB_ON_OFF_SWITCH: Start network formation
I (479) ESP_ZB_ON_OFF_SWITCH: Formed network successfully (Extended PAN ID: 15:d3:78:6f:29:6d:90:4a, PAN ID: 0x2c26, Channel:13, Short Address: 0x0000)
I (1089) ESP_ZB_ON_OFF_SWITCH: Network(0x2c26) is open for 180 seconds
I (1089) ESP_ZB_ON_OFF_SWITCH: Network steering started
I (20909) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (22759) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (22759) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (41489) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (44669) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (44669) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (63249) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (65849) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (65849) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (85029) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (87769) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (87769) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
```

It means that the end device connects to the network, the coordinator permits access, but then the end device fails. The same was happening with the Tasmota as well.

I'm using the latest version of IDF and libraries.
Both devices are about 20cm apart from each other.

No matter what I do, the end device cannot join the network. I have spent on this three days already.

valdisz, 2023-11-23:
The production config was missing from the chip and it was logging `I (445) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL`

I flashed missing information, now it does not report problems `I (465) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_OK`

But that does not help, the device cannot join the network.

valdisz, 2023-11-23:
Attaching debug logs from the end device.

I see there messages like this one:
`ZBOSS: zdo/zdo_app.c:595   Have not got nwk key - authentication failed`

Also, there are warning messages:
- `0x42034bf6: zb_multimac_mlme_reset_request_proxy_send_req at nwk_multimac_proxy.c.obj:?`
- `0x42035310: zb_multimac_mlme_set_request_proxy_send_req at nwk_multimac_proxy.c.obj:?`
- `0x42035310: zb_multimac_mlme_set_request_proxy_send_req at nwk_multimac_proxy.c.obj:?`
- ... 

[detailed-logs.txt](https://github.com/espressif/esp-idf/files/13453675/detailed-logs.txt)

valdisz, 2023-11-24:
Looks like the Production Config I was writing to the chip was not correct.

closed 2023-11-24 by valdisz

JanneMantyharju, 2024-02-10:
@valdisz Do you remember exactly what you did with production config? I have the same problem:
`I (14400) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (16800) ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0x42c9 on channel 11, dev 0x0, iface_id 0
I (17310) ZBOSS: mac/mac_associate.c:448   received association response
I (17320) ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0xc14, iface_id 0, ch 11, rejoin 0
I (17320) ZBOSS: zdo/zdo_app_join.c:773   xpanid 29:d2:c5:99:8a:76:c1:bc
I (27370) ZBOSS: zdo/zdo_app.c:595   Have not got nwk key - authentication failed
I (27370) ZBOSS: nwk/nwk_main.c:4409   zb_nwk_do_leave param 15 rejoin 0
I (27370) ZBOSS: common/zb_address.c:1063   zb_address_delete ref 17`

I followed the mfg_tool example on https://github.com/espressif/esp-matter/tree/208ffca11d74517a864ada9ac900971655075493/tools/mfg_tool

but I still get the same result.

valdisz, 2024-02-13:
@JanneMantyharju I have not solved this; thankfully, in my case, that does not stop devices from functioning. Of course, you cannot go to the real production like that.

mukeshbauskar, 2024-06-26:
> Looks like the Production Config I was writing to the chip was not correct.

@valdisz can you please elaborate on Production Config. For me it is not at all pairing...