espressif/esp-idf #12659: Documentation regarding automatic light sleep

JVKran, 2023-11-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I can't seem to find elaborate information on how automatic light sleep works. For example, if I enable automatic light sleep by calling ```esp_pm_configure```, the code simply stops running. Every time I use a wakeup-source (such as a button for example), a few more characters are printed in the terminal. Once the entire log-message is printed and visible, the application keeps running as normal. 

So, from first to last, this happens when I press the button.

```
I (983) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (1003) sleep: Code start at 0x42000020, t
```

Press

```
I (983) main: Boot exited.
I (983) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (1003) sleep: Code start at 0x42000020, total
```

Press

```
I (983) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (1003) sleep: Code start at 0x42000020, total 1009351, da
```

Press

```
I (983) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 80, Light sleep: ENABLED
I (1003) sleep: Code start at 0x42000020, total 1009351, data start at 0x3c000000, total 33554432 Bytes
I (1003)
```

Am I supposed to take the ```ESP_PM_NO_LIGHT_SLEEP``` in this case? What should I do? Is this expected?

igrr, 2023-11-28:
Hi @JVKran, the logic used by the power management module is described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html#esp32-power-management-algorithm). If automatic light sleep is enabled, the system will automatically wake up if there is a task ready to run, or a software timer (esp_timer) ready to execute. If neither of these conditions are true, the system will stay in light sleep mode.

> Every time I use a wakeup-source (such as a button for example), a few more characters are printed in the terminal. 

I think what is happening is that before the system went to sleep the first time, some data was placed into UART FIFO. When the system enters light sleep, UART is suspended (and the peripheral is clock-gated). Each time the system wakes up, a few more characters get sent out from the FIFO before the system goes to sleep again.

However, note that the system will not enter light sleep while there is a task which is ready to run (i.e. not blocked on an RTOS synchronization primitive).

> Am I supposed to take the ESP_PM_NO_LIGHT_SLEEP in this case? What should I do? Is this expected?

This depends on what you are trying to achieve. If you can attach the source code you are trying to run, it would help us give you a suggestion.

JVKran, 2023-11-29:
`I think what is happening is that before the system went to sleep the first time, some data was placed into UART FIFO. When the system enters light sleep, UART is suspended (and the peripheral is clock-gated). Each time the system wakes up, a few more characters get sent out from the FIFO before the system goes to sleep again.`

Thanks to you putting it like this, I understand what's happening. It's a combination of application behaviour and the terminal not representing the most recent state. Thanks. Everything was defined.

JVKran, 2024-07-17:
Reopening as I've found the time to find the cause for this behaviour.

Starting a REPL for the esp_console results in the behaviour described above. When I comment this line, the behaviour described above does not occur anymore.

``` c
esp_console_new_repl_uart(&m_uart_config, &m_repl_config, &m_repl);
```

I'm not necesarilly interested in a console over UART, but I do need to be able to execute commands to esp_console_run. Is there an example that shows how to start the console but without REPL? Even better would be to have the REPL being able to work with automatic light sleep.

Best regards,

Jochem

ESP-Marius, 2024-07-23:
@JVKran As far as I see `esp_console_run` should work fine without the REPL, see [this test-case](https://github.com/espressif/esp-idf/blob/af25eb447e3330c21e3b38e91db16332056882b2/components/console/test_apps/console/main/test_console.c#L226) for an example.

Regarding console+light sleep you can read/get more information in https://github.com/espressif/esp-idf/issues/14137

ESP-Marius, 2024-08-22:
Marking this one as resolved, for any further questions regarding console + light sleep we can take that in the thread linked above

closed 2024-08-22 by ESP-Marius