espressif/esp-idf #12660: Issue initializing LAN8720A in v5.0

pedrorovi, 2023-11-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0 24b9d38a24e354102b524d70c0f112a799aee9f7

### Espressif SoC revision.

ESP32-WROOM-32E

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Olimex gateway Rev. G

### Power Supply used.

USB

### What is the expected behavior?

Init and run Ethernet on ESP32.

### What is the actual behavior?

The esp can't init Ethernet lan8720. I following the esp at instructions [here](https://docs.espressif.com/projects/esp-at/en/release-v3.2.0.0/esp32/AT_Command_Set/Ethernet_AT_Commands.html) but it's not working. I find the same issue with other version [here](https://github.com/espressif/esp-idf/issues/5425).

### Steps to reproduce.

1. ./build.py install
2. ./build.py menuconfig
3. Following the instructions to configure Ethernet
4. ./build.py -p (PORT) flash
5. ./build.py -p (PORT) monitor

### Debug Logs.

```plain
E (2099) eth_phy_802_3: eth_phy_802_3_pwrctl(164): power up timeout
E (2099) eth_phy_802_3: esp_eth_phy_802_3_basic_phy_init(331): power control failed
E (2103) lan87xx: lan87xx_init(292): failed to init PHY
E (2109) esp_eth: esp_eth_driver_install(229): init phy failed
```
```

pedrorovi, 2023-11-24:
Changing #define CONFIG_AT_ETH_PHY_ADDR         1 to 0 is working

windymonkey, 2023-11-29:
> Changing #define CONFIG_AT_ETH_PHY_ADDR 1 to 0 is working

This solve my problem, thanks.

kostaond, 2023-12-01:
Closing the issue. It was a configuration issue and user was able to find solution.

closed 2023-12-01 by kostaond