espressif/esp-idf #12661: HTTP stream to upload large file causes frequent connection failed, sock < 0

suryasid09, 2023-11-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using HTTP stream described in the website to upload large file (1 MB file). 
I have noticed that when I open connection with esp_http_client_open(), I am not able to established TCP connection. 
[image]

However, when not using HTTP stream and instead using esp_http_client_perform(), I instantly get TCP connection. 

This is the code I am using for HTTP stream: 

configuring http in main:

```
esp_http_client_config_t config_post = {
    .url = "http://192.168.10.21/uploads/upload_file.php",
    .method = HTTP_METHOD_POST,
    .cert_pem = NULL,
    .keep_alive_enable = 1,
    .timeout_ms = 600 * 1000};

client = esp_http_client_init(&config_post);

esp_http_client_set_header(client, "Content-Type", "application/octet-stream");
```

Reading from a usb and using http post:

```

static esp_err_t post_rest_function(void) {
  static char *post_data = buffer;
  ESP_ERROR_CHECK(esp_http_client_set_post_field(client, post_data, 8192));
  // esp_err_t err = esp_http_client_perform(client);
  esp_err_t err = esp_http_client_write(client, buffer, 8192);

  return err;
}

static void file_operations(void) {
  int64_t start_time, end_time, mid_time;

  static FILE *file;
  static esp_err_t err;
  static size_t read_bytes;
  static size_t last_sent_bytes;
  static const char *directory = "/usb/esp";
  static const char *file_path = "/usb/esp/book.txt";

  struct stat s = {0};
  bool directory_exists = stat(directory, &s) == 0;

  if (!directory_exists) {
    ESP_LOGE(TAGUSB, "No directory found errno: %s", strerror(errno));
  }
  ESP_LOGI(TAGUSB, "Reading file");
  file = fopen(file_path, "rb");
  if (file == NULL) {
    ESP_LOGE(TAGUSB, "Failed to open file for reading");
    return;
  }

  int i, j;

  start_time = esp_timer_get_time();
  esp_http_client_open(client, 10000);
  while (1) {
    memset(buffer, 0, sizeof(buffer));
    read_bytes = fread(buffer, 1, sizeof(buffer) - 2, file);
    buffer[8191] = '\0';

    if (read_bytes <= 0) {
      break;  // End of file reached
    } else {
      err = post_rest_function();
    }
  }
  end_time = esp_timer_get_time();
  fclose(file);
  esp_http_client_close(client);
  esp_http_client_cleanup(client);
  printf("time taken is %lld sec for 1MB\n", (end_time - start_time) / 1000000);
  printf("Transfer speed in MiB/s %f\n",
         (1 * 1000000) / (float)(end_time - start_time));
}

```
I want to use HTTP stream to increase my upload speed, but it doesn't seem to be working. I would appreciate any help.

suryasid09, 2023-11-28:
any update on this?

hmalpani, 2023-12-05:
Hello @suryasid09 
Can you provide with a minimal application which can be used to reproduce this? Please also share the sdkconfig file with which you are facing this issue.

Thanks!

suryasid09, 2023-12-05:
Hello @hmalpani 
Thank you for your reply.

some development since my last post -> I did some further experiment. When 1st time I am trying to upload 1 MB data using HTTP stream, Esp32 gets IP assigned immediately, but, often when I ping the IP address of esp32 (through windows command line interface), there is no response. Then, I preform the reset by pressing the button on esp32 dev board  the 2nd time to check again uploading of 1 MB file, it always fail to open new http connection and gives the above error. I need to wait for around 5 mins or so then it starts it work again. On the server side, I am using XAMPP for testing. 

Additionally, I have also tried connecting to different router thinking it is router issue but it doesn't solve the issue. 

Please find the cleaned code:

```
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "esp_http_client.h"
#include "esp_timer.h"

char buffer[4096];

static void wifi_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    switch (event_id)
    {
    case WIFI_EVENT_STA_START:
        printf("WiFi connecting ... \n");
        break;
    case WIFI_EVENT_STA_CONNECTED:
        printf("WiFi connected ... \n");
        break;
    case WIFI_EVENT_STA_DISCONNECTED:
        printf("WiFi lost connection ... \n");
        break;
    case IP_EVENT_STA_GOT_IP:
        printf("WiFi got IP ... \n\n");
        break;
    default:
        break;
    }
}

void wifi_connection()
{

    esp_netif_init();
    esp_event_loop_create_default();
    esp_netif_create_default_wifi_sta();
    wifi_init_config_t wifi_initiation = WIFI_INIT_CONFIG_DEFAULT();
    esp_wifi_init(&wifi_initiation);
    esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, wifi_event_handler, NULL);
    esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, wifi_event_handler, NULL);
    wifi_config_t wifi_configuration = {
        .sta = {
            .ssid = "your_ssid",
            .password = "your_password"}};
    esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_configuration);

    esp_wifi_start();
    esp_wifi_connect();

}

esp_err_t client_event_post_handler(esp_http_client_event_handle_t evt)
{
    switch (evt->event_id)
    {
    case HTTP_EVENT_ON_DATA:
        printf("HTTP_EVENT_ON_DATA: %.*s\n", evt->data_len, (char *)evt->data);
        break;

    default:
        break;
    }
    return ESP_OK;
}

void http_task()
{

	    esp_http_client_config_t config_post = {
	        .url = "http://192.168.10.21/uploads/upload_file.php",
	        .method = HTTP_METHOD_POST,
	        .cert_pem = NULL,
	        .event_handler = client_event_post_handler};

	    esp_http_client_handle_t client = esp_http_client_init(&config_post);

		if ((err = esp_http_client_open(client, 9998336)) != ESP_OK)								//open to transfer 10MB file
		{
		    ESP_LOGE("HTTP_custom_msg", "Failed to open HTTP connection: %s", esp_err_to_name(err));
		}

		for(i = 0; i< 2441 ; i++)
		{
			for (j =0; j<4096; j++)
			{
				buffer[j] = (char)(1+j%255);														//putting random characters
			}
			buffer[j] = '\0';
			esp_http_client_write(client,buffer,sizeof(buffer));

		}

	    esp_http_client_close(client);
	    esp_http_client_cleanup(client);

}

void app_main(void)
{
    nvs_flash_init();
    wifi_connection();
    vTaskDelay(5*1000 / portTICK_PERIOD_MS);
    printf("WIFI was initiated ...........\n\n");
    http_task();

}

```

and the sdkconfig file:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13557020/sdkconfig.txt)

hmalpani, 2023-12-05:
Hello @suryasid09 

I tested your code. I am able to upload whole 10M data to server without any error. I started a python http server. You can see the server setup in the code below:
```
import argparse
from http.server import HTTPServer, BaseHTTPRequestHandler

class S(BaseHTTPRequestHandler):
    def do_POST(self):
        content_length = int(self.headers['Content-Length']) # <--- Gets the size of data
        post_data = self.rfile.read(content_length) # <--- Gets the data itself        # Doesn't do anything with posted data
        # self._set_headers()
        # self.wfile.write(self._html("POST!"))

def run(server_class=HTTPServer, handler_class=S, addr="localhost", port=8000):
    server_address = (addr, port)
    httpd = server_class(server_address, handler_class)

    print(f"Starting httpd server on {addr}:{port}")
    httpd.serve_forever()

if __name__ == "__main__":

    parser = argparse.ArgumentParser(description="Run a simple HTTP server")
    parser.add_argument(
        "-l",
        "--listen",
        default="0.0.0.0",
        help="Specify the IP address on which the server listens",
    )
    parser.add_argument(
        "-p",
        "--port",
        type=int,
        default=8000,
        help="Specify the port on which the server listens",
    )
    args = parser.parse_args()
    run(addr=args.listen, port=args.port)
```

suryasid09, 2023-12-06:
@hmalpani thanks for looking into it. 
with the python server, it is working as intended, it seems my XAMP server was creating issue. I haven't done the thorough testing yet with data upload from usb or sd card. But, I will close the issue for now.