espressif/esp-idf #12662: host example msc_host_vfs_register error

RomanYakhnenko, 2023-11-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1U

### Power Supply used.

USB

### What is the expected behavior?

Flash is mounted

### What is the actual behavior?

Error mount flash in 
ESP_ERROR_CHECK( msc_host_vfs_register(msc_device, "/usb", &mount_config, &vfs_handle) );

### Steps to reproduce.

1. Take example from IDF 5.1.1  examples\peripherals\usb\host\msc
2. Set My chip
3. Set port
4. Run compile
5. Run...

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x186c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3130
entry 0x403c9934
I (27) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (27) boot: compile time Nov 24 2023 12:34:29  
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz     
I (44) boot.esp32s3: SPI Mode       : QIO        
I (49) boot.esp32s3: SPI Flash Size : 8MB        
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length  
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot:  3 storage          Unknown data     01 82 00110000 002f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0ef64h ( 61284) map
I (122) esp_image: segment 1: paddr=0001ef8c vaddr=3fc92900 size=0108ch (  4236) load
I (123) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=3629ch (221852) map
I (163) esp_image: segment 3: paddr=000562c4 vaddr=3fc9398c size=01854h (  6228) load
I (164) esp_image: segment 4: paddr=00057b20 vaddr=40374000 size=0e854h ( 59476) load
I (186) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
I (198) cpu_start: Multicore app
I (198) cpu_start: Pro cpu up.
I (198) cpu_start: Starting app cpu, entry point is 0x40375304
0x40375304: call_start_cpu1 at C:/Espressif/frameworks/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (216) cpu_start: Pro cpu start user code
I (216) cpu_start: cpu freq: 160000000 Hz
I (216) cpu_start: Application information:
I (219) cpu_start: Project name:     usb-msc
I (224) cpu_start: App version:      1
I (228) cpu_start: Compile time:     Nov 24 2023 12:37:58
I (234) cpu_start: ELF file SHA256:  b5a3f2892118bee4...
I (240) cpu_start: ESP-IDF:          v5.1.1
I (245) cpu_start: Min chip rev:     v0.0
I (250) cpu_start: Max chip rev:     v0.99 
I (255) cpu_start: Chip rev:         v0.1
I (260) heap_init: Initializing. RAM available for dynamic allocation:
I (267) heap_init: At 3FC95AB0 len 00053C60 (335 KiB): DRAM
I (273) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (280) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (286) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (293) spi_flash: detected chip: generic
I (297) spi_flash: flash io: qio
I (301) sleep: Configure to isolate all GPIO pins in sleep state
I (308) sleep: Enable automatic switching of GPIO sleep configuration
I (315) app_start: Starting scheduler on CPU0
I (320) app_start: Starting scheduler on CPU1
I (320) main_task: Started on CPU0
I (330) main_task: Calling app_main()
I (610) example: Waiting for USB stick to be connected
I (1000) example: MSC device connected
I (1000) example: connection...
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x930
idProduct 0x6545
bcdDevice 1.10
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 32
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 150mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0x8
        bInterfaceSubClass 0x6
        bInterfaceProtocol 0x50
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
Device info:
         Capacity: 7445 MB
         Sector size: 512
         Sector count: 15249407
         PID: 0x6545
         VID: 0x0930
         iProduct: USB Flash Memory
         iManufacturer:
         iSerialNumber: 00187D1174FEE190111E6C89 
ESP_ERROR_CHECK failed: esp_err_t 0x1701 (ERROR) at 0x4200958b
0x4200958b: app_main at C:/REPOSITARY/PU110/Tests/Storage/msc_ESPIDF_clean/main/msc_example_main.c:212 (discriminator 1)

file: "./main/msc_example_main.c" line 212
func: app_main
expression: msc_host_vfs_register(msc_device, "/usb", &mount_config, &vfs_handle)

abort() was called at PC 0x4037a2fb on core 0
0x4037a2fb: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf/components/esp_system/esp_err.c:50

Backtrace: 0x4037584e:0x3fc993d0 0x4037a305:0x3fc993f0 0x403806be:0x3fc99410 0x4037a2fb:0x3fc99480 0x4200958b:0x3fc994b0 0x42035ac3:0x3fc99600 0x4037ca09:0x3fc99630
0x4037584e: panic_abort at C:/Espressif/frameworks/esp-idf/components/esp_system/panic.c:452

0x4037a305: esp_system_abort at C:/Espressif/frameworks/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x403806be: abort at C:/Espressif/frameworks/esp-idf/components/newlib/abort.c:38

0x4037a2fb: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf/components/esp_system/esp_err.c:50

0x4200958b: app_main at C:/REPOSITARY/PU110/Tests/Storage/msc_ESPIDF_clean/main/msc_example_main.c:212 (discriminator 1)

0x42035ac3: main_task at C:/Espressif/frameworks/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)

0x4037ca09: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

Try with 3 different usb flash stick. Try change in [here](https://github.com/tore-espressif/esp-idf/commit/632e7d42d8af5432a0a05cd7b37c89cbf32834ff)

RomanYakhnenko, 2023-11-24:
espressif/usb_host_msc version 1.1.1

RomanYakhnenko, 2023-11-24:
When i make active 

`const esp_vfs_fat_mount_config_t mount_config = {
    .format_if_mount_failed = **true**,`
flash format and mount - now testing

RomanYakhnenko, 2023-11-24:
In this moment after format from esp32 all work. Not undestand - try format from ciomputer with fifferent parameters. If need open again

closed 2023-11-24 by RomanYakhnenko

tore-espressif, 2023-11-27:
Hi @RomanYakhnenko , the MSC driver supports only FAT file format. New USB flash disks with great size (>8GB) usually use exFAT. That is why reformatting to FAT helped.

fozzzgate, 2024-06-12:
@RomanYakhnenko Were you ever able to get this working with Windows FAT32 formatted USB drives? I am having this same problem and I can't figure out what is causing it. Windows formatted USB drives are not accepted (0x1701 - ESP_ERR_MSC_MOUNT_FAILED) and USB drives formatted by usb_host_msc can be read fine on ESP32, but are corrupt and not accepted on Windows.

I have downgraded to ESP IDF 5.0.4 & 4.4.7 / usb_host_msc 1.0.0 and it is still happening. Everything was working fine yesterday with Windows-formatted USB drives until I upgraded to ESP IDF 5.2.2 and now everything is broken and I can't even go back to a working version.