espressif/esp-idf #12667: [ESP32-S3: USB Host HID] How to handle 2 HID interface for the same device?

rtek1000, 2023-11-26:
### Is your feature request related to a problem?

Hi,

How to handle 2 HID interface for the same device?

In 'USB Host HID' project: Only the keyboard part works, but the touchpad part does not work. The multimedia keys don't work either.

The mini keyboard has an additional interface for the touchpad.

[image]

Below is the serial monitor dump for the project ../examples/peripherals/usb/host/hid/ (mini keyboard attached):
> Executing action: monitor
> Running idf_monitor in directory /home/user/esp/hid
> Executing "/home/user/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/user/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /home/user/esp/hid/build/hid.elf -m '/home/user/.espressif/python_env/idf5.3_py3.10_env/bin/python' '/home/user/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...
> --- esp-idf-monitor 1.3.3 on /dev/ttyACM0 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> SP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x178c
> load:0x403c9700,len:0x4
> load:0x403c9704,len:0xcb8
> load:0x403cc700,len:0x2d84
> entry 0x403c9914
> I (27) boot: ESP-IDF v5.3-dev-422-ga7fbf452fa 2nd stage bootloader
> I (27) boot: compile time Nov 26 2023 00:55:55
> I (27) boot: Multicore bootloader
> I (31) boot: chip revision: v0.2
> I (35) boot.esp32s3: Boot SPI Speed : 80MHz
> I (40) boot.esp32s3: SPI Mode       : DIO
> I (45) boot.esp32s3: SPI Flash Size : 2MB
> I (49) boot: Enabling RNG early entropy source...
> I (55) boot: Partition Table:
> I (58) boot: ## Label            Usage          Type ST Offset   Length
> I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (81) boot:  2 factory          factory app      00 00 00010000 00100000
> I (88) boot: End of partition table
> I (92) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0c5fch ( 50684) map
> I (110) esp_image: segment 1: paddr=0001c624 vaddr=3fc92400 size=027b0h ( 10160) load
> I (112) esp_image: segment 2: paddr=0001eddc vaddr=40374000 size=0123ch (  4668) load
> I (119) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1f494h (128148) map
> I (149) esp_image: segment 4: paddr=0003f4bc vaddr=4037523c size=0d158h ( 53592) load
> I (167) boot: Loaded app from partition at offset 0x10000
> I (167) boot: Disabling RNG early entropy source...
> I (178) cpu_start: Multicore app
> I (188) cpu_start: Pro cpu start user code
> I (188) cpu_start: cpu freq: 160000000 Hz
> I (188) cpu_start: Application information:
> I (191) cpu_start: Project name:     hid
> I (196) cpu_start: App version:      1
> I (200) cpu_start: Compile time:     Nov 26 2023 00:55:43
> I (206) cpu_start: ELF file SHA256:  c7a7b152f...
> I (212) cpu_start: ESP-IDF:          v5.3-dev-422-ga7fbf452fa
> I (218) cpu_start: Min chip rev:     v0.0
> I (223) cpu_start: Max chip rev:     v0.99 
> I (228) cpu_start: Chip rev:         v0.2
> I (232) heap_init: Initializing. RAM available for dynamic allocation:
> I (240) heap_init: At 3FC95420 len 000542F0 (336 KiB): RAM
> I (246) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
> I (252) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
> I (258) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
> I (265) spi_flash: detected chip: generic
> I (269) spi_flash: flash io: dio
> W (273) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> I (286) sleep: Configure to isolate all GPIO pins in sleep state
[... 131 lines cut ...]
> 		*** Endpoint descriptor ***
> 		bLength 7
> 		bDescriptorType 5
> 		bEndpointAddress 0x82	EP 2 IN
> 		bmAttributes 0x3	INT
> 		wMaxPacketSize 8
> 		bInterval 8

Board: YD-ESP32-23 (using USB C OTG cable too):

[image]

### Describe the solution you'd like.

It would be interesting if the USB Host HID example can already handle the two HID interfaces of this mini keyboard with touchpad, because there are many such keyboards for sale in Chinese markets (Aliexpress etc).

### Describe alternatives you've considered.

I would like tips on how to handle the second HID interface.

### Additional context.

As the mouse already works well with the project (USB Host HID), and the keyboard works well with the same project, the only thing missing is that both can be operated at the same time.

I made some modifications to a design for STM32F4 that allowed me to scan both interfaces, but I noticed that the ESP32-S3 performance seems to be better than the STM32 performance, which may be due to the use of interrupts in the ESP32, while STM32 appears to be operating in polling mode.

roma-jam, 2023-11-26:
Hi @rtek1000,

HID example already displayed the information, that mouse interface has been connected. 
Is there any display output when:
- Keys on keyboard were pressed;
- Touch pad is in use;
- Key that emulate left-right mouse buttons were pressed (they are on the left side of the keyboard);

Could you share the log during all these scenarious? 
Thanks.

rtek1000, 2023-11-27:
Hi, I added other flags to each subroutine that I think are relevant for monitoring the return of events, and as you can see, when you put a cable mouse into the USB port, the mouse routines are executed, but when you put the mini keyboard with touchpad, only keyboard subroutines are executed. 

At the bottom of this message the modified code has been added, similar to this one:
`printf("%lu - hid_host_keyboard_report_callback()\r\n", log_cnt1++);`

Log from 'idf.py -p /dev/ttyACM0 monitor':

> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x178c
> load:0x403c9700,len:0x4
> load:0x403c9704,len:0xcb8
> load:0x403cc700,len:0x2d84
> entry 0x403c9914
> I (27) boot: ESP-IDF v5.3-dev-422-ga7fbf452fa 2nd stage bootloader
> I (27) boot: compile time Nov 26 2023 00:55:55
> I (27) boot: Multicore bootloader
> I (31) boot: chip revision: v0.2
> I (35) boot.esp32s3: Boot SPI Speed : 80MHz
> I (40) boot.esp32s3: SPI Mode       : DIO
> I (45) boot.esp32s3: SPI Flash Size : 2MB
> I (49) boot: Enabling RNG early entropy source...
> I (55) boot: Partition Table:
> I (58) boot: ## Label            Usage          Type ST Offset   Length
> I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (81) boot:  2 factory          factory app      00 00 00010000 00100000
> I (88) boot: End of partition table
> I (92) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0c5fch ( 50684) map
> I (110) esp_image: segment 1: paddr=0001c624 vaddr=3fc92400 size=027b0h ( 10160) load
> I (112) esp_image: segment 2: paddr=0001eddc vaddr=40374000 size=0123ch (  4668) load
> I (119) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1f450h (128080) map
> I (149) esp_image: segment 4: paddr=0003f478 vaddr=4037523c size=0d158h ( 53592) load
> I (167) boot: Loaded app from partition at offset 0x10000
> I (167) boot: Disabling RNG early entropy source...
> I (178) cpu_start: Multicore app
> I (188) cpu_start: Pro cpu start user code
> I (188) cpu_start: cpu freq: 160000000 Hz
> I (188) cpu_start: Application information:
> I (191) cpu_start: Project name:     hid
> I (196) cpu_start: App version:      1
> I (200) cpu_start: Compile time:     Nov 26 2023 00:55:43
> I (206) cpu_start: ELF file SHA256:  58a98384c...
> I (212) cpu_start: ESP-IDF:          v5.3-dev-422-ga7fbf452fa
> I (218) cpu_start: Min chip rev:     v0.0
> I (223) cpu_start: Max chip rev:     v0.99 
> I (227) cpu_start: Chip rev:         v0.2
> I (232) heap_init: Initializing. RAM available for dynamic allocation:
> I (240) heap_init: At 3FC95420 len 000542F0 (336 KiB): RAM
> I (246) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
> I (252) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
> I (258) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
> I (265) spi_flash: detected chip: generic
> I (269) spi_flash: flash io: dio
> W (273) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> I (286) sleep: Configure to isolate all GPIO pins in sleep state
> I (293) sleep: Enable automatic switching of GPIO sleep configuration
> I (300) main_task: Started on CPU0
> I (310) main_task: Calling app_main()
> I (310) example: HID Host example
> I (310) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
> I (350) example: Waiting for HID Device to be connected
> 0 - hid_host_device_callback()
> 1 - hid_host_device_event()
> I (1970) example: HID Device, protocol 'MOUSE' CONNECTED  (regular mouse)
> 2 - hid_host_interface_callback()
> 3 - hid_host_mouse_report_callback()
[... 649 lines cut ...]
                ESP_ERROR_CHECK(usb_host_lib_info(&lib_info));
                if (lib_info.num_devices == 0) {
                    // End while cycle
                    break;
                } else {
                    ESP_LOGW(TAG, "To shutdown example, remove all USB devices and press button again.");
                    // Keep polling
                }
            }

            if (APP_EVENT_HID_HOST ==  evt_queue.event_group) {
                hid_host_device_event(evt_queue.hid_host_device.handle,
                                      evt_queue.hid_host_device.event,
                                      evt_queue.hid_host_device.arg);
            }
        }
    }

    ESP_LOGI(TAG, "HID Driver uninstall");
    ESP_ERROR_CHECK(hid_host_uninstall());
    gpio_isr_handler_remove(APP_QUIT_PIN);
    xQueueReset(app_event_queue);
    vQueueDelete(app_event_queue);
}
```

roma-jam, 2023-11-27:
Hi @rtek1000, 

> Where can I find information on how to generate the requested log?

You already provided the log I asked for. This is exactly the log from `idf.py -p /dev/ttyACM0 monitor`. Thanks.

Regarding the example, right now the HID Driver parses only reports when HID device support BOOT protocol.
Otherwise, the general output report should be present. 

Based on the USB descriptor and log output you have provided, both interfaces were connected and should work as a HID devices by boot protocol.

Could you annotate the forcing BOOT protocol for both (Keyboards and Mice) protocols and try the HID example again?
This could be achieved with the annotating lines 438-443 in [hid_host_example.c](https://github.com/espressif/esp-idf/blob/ab03c2ea13ecaac1510b75e93b32cf0c472640fb/examples/peripherals/usb/host/hid/main/hid_host_example.c#L438).
 
    ESP_ERROR_CHECK(hid_host_device_open(hid_device_handle, &dev_config));
    // if (HID_SUBCLASS_BOOT_INTERFACE == dev_params.sub_class) {
    //    ESP_ERROR_CHECK(hid_class_request_set_protocol(hid_device_handle, HID_REPORT_PROTOCOL_BOOT));
    //    if (HID_PROTOCOL_KEYBOARD == dev_params.proto) {
    //        ESP_ERROR_CHECK(hid_class_request_set_idle(hid_device_handle, 0, 0));
    //    }
    }
    ESP_ERROR_CHECK(hid_host_device_start(hid_device_handle));

And share the log again after: 
1. Attach device to the ESP32 Host (keyboard+touchpad)
2. Press several keys on a keyboard (for example `qwerty`)
3. Press one of the left buttons with mouse on them (left click or right click)
4. Press the touchpad with one finger
5. Press the touchpad with two fingers and slide them up-down
6. Press the touchpad with three fingers
7. Press "up-left-down-right" arrows on the joystick in the top-right of the device. 
8. Detach device.

Thanks.

rtek1000, 2023-11-27:
> Could you annotate the forcing BOOT protocol for both (Keyboards and Mice) protocols and try the HID example again? This could be achieved with the annotating lines 438-443 in [...]

- Ok, the log from ```idf.py -p /dev/ttyACM0 monitor```: 

> Executing action: monitor
> Running idf_monitor in directory /home/user/esp/hid
> Executing "/home/user/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/user/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /home/user/esp/hid/build/hid.elf -m '/home/user/.espressif/python_env/idf5.3_py3.10_env/bin/python' '/home/user/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...
> --- esp-idf-monitor 1.3.3 on /dev/ttyACM0 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> SP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x178c
> load:0x403c9700,len:0x4
> load:0x403c9704,len:0xcb8
> load:0x403cc700,len:0x2d84
> entry 0x403c9914
> I (27) boot: ESP-IDF v5.3-dev-422-ga7fbf452fa 2nd stage bootloader
> I (27) boot: compile time Nov 26 2023 00:55:55
> I (27) boot: Multicore bootloader
> I (31) boot: chip revision: v0.2
> I (35) boot.esp32s3: Boot SPI Speed : 80MHz
> I (40) boot.esp32s3: SPI Mode       : DIO
> I (45) boot.esp32s3: SPI Flash Size : 2MB
> I (49) boot: Enabling RNG early entropy source...
> I (55) boot: Partition Table:
> I (58) boot: ## Label            Usage          Type ST Offset   Length
> I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (81) boot:  2 factory          factory app      00 00 00010000 00100000
> I (88) boot: End of partition table
> I (92) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0c38ch ( 50060) map
> I (110) esp_image: segment 1: paddr=0001c3b4 vaddr=3fc92400 size=027b0h ( 10160) load
> I (112) esp_image: segment 2: paddr=0001eb6c vaddr=40374000 size=014ach (  5292) load
> I (119) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1f004h (126980) map
> I (149) esp_image: segment 4: paddr=0003f02c vaddr=403754ac size=0cee8h ( 52968) load
> I (167) boot: Loaded app from partition at offset 0x10000
> I (167) boot: Disabling RNG early entropy source...
> I (178) cpu_start: Multicore app
> I (188) cpu_start: Pro cpu start user code
> I (188) cpu_start: cpu freq: 160000000 Hz
> I (188) cpu_start: Application information:
> I (191) cpu_start: Project name:     hid
> I (195) cpu_start: App version:      1
> I (200) cpu_start: Compile time:     Nov 26 2023 00:55:43
> I (206) cpu_start: ELF file SHA256:  ac0b9e848...
> I (211) cpu_start: ESP-IDF:          v5.3-dev-422-ga7fbf452fa
> I (218) cpu_start: Min chip rev:     v0.0
> I (222) cpu_start: Max chip rev:     v0.99 
> I (227) cpu_start: Chip rev:         v0.2
> I (232) heap_init: Initializing. RAM available for dynamic allocation:
> I (239) heap_init: At 3FC95420 len 000542F0 (336 KiB): RAM
> I (245) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
> I (251) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
> I (257) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
> I (264) spi_flash: detected chip: generic
> I (268) spi_flash: flash io: dio
> W (272) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> I (286) sleep: Configure to isolate all GPIO pins in sleep state
> I (292) sleep: Enable automatic switching of GPIO sleep configuration
> I (300) main_task: Started on CPU0
> I (310) main_task: Calling app_main()
> I (310) example: HID Host example
> I (310) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
> I (350) example: Waiting for HID Device to be connected
> I (10090) example: HID Device, protocol 'KEYBOARD' CONNECTED
> I (10090) example: HID Device, protocol 'MOUSE' CONNECTED
> 
[... 203 lines cut ...]
> 104 - hid_host_interface_callback()
> 105 - hid_host_mouse_report_callback()
> 106 - hid_host_interface_callback()
> 107 - hid_host_mouse_report_callback()
> 108 - hid_host_interface_callback()
> 109 - hid_host_mouse_report_callback()
> 110 - hid_host_interface_callback()
> 111 - hid_host_mouse_report_callback()
> 112 - hid_host_interface_callback()
> 113 - hid_host_mouse_report_callback()
> 114 - hid_host_interface_callback()
> 115 - hid_host_mouse_report_callback()
> 116 - hid_host_interface_callback()
> 117 - hid_host_mouse_report_callback()
> E (111140) USBH: Device 1 gone
> 118 - hid_host_interface_callback()
> I (111140) example: HID Device, protocol 'KEYBOARD' DISCONNECTED
> 119 - hid_host_interface_callback()
> I (111150) example: HID Device, protocol 'MOUSE' DISCONNECTED

----

- An additional question, about using a USB HUB, to use a regular mouse in one port of the HUB and a regular keyboard in another port (without touchpad), is it possible?

- - I tried to use a HUB but it didn't recognize either the mouse or the keyboard. Although both keyboard and mouse are recognized if they are inserted directly into the ESP32-S3's USB port one at a time.

roma-jam, 2023-11-27:
Hi @rtek1000, 
Thanks for the additional info. 

Could you please make the same things on the clean example code? 
The idea here is to get the raw report data and see what we can do with it.

Right now it looks like both (Keyboard and Mouse) interfaces are working, the goal is to understand how to parse the data when touchpad is active.

> An additional question, about using a USB HUB, to use a regular mouse in one port of the HUB and a regular keyboard in another port (without touchpad), is it possible?

Technically, it is possible but right now it is not implemented in `usb_host` library, so any external HUBs is just a regular USB device for ESP32. 

But we have it in our TODO list and we have plans to implement it.

rtek1000, 2023-11-27:
> Hi @rtek1000, Thanks for the log, seems that the USB HID Driver works fine. The report handling is a part of a higher logic now and can be done above easily.

- Yes, thanks!

For STM32, some tags are used, which might be interesting to have a pattern for keys that do not have characters:

```C++
#define KEY_NONE                               0x00
#define KEY_ERRORROLLOVER                      0x01
#define KEY_POSTFAIL                           0x02
#define KEY_ERRORUNDEFINED                     0x03
#define KEY_A                                  0x04
#define KEY_B                                  0x05
#define KEY_C                                  0x06
#define KEY_D                                  0x07
#define KEY_E                                  0x08
#define KEY_F                                  0x09
#define KEY_G                                  0x0A
#define KEY_H                                  0x0B
#define KEY_I                                  0x0C
#define KEY_J                                  0x0D
#define KEY_K                                  0x0E
#define KEY_L                                  0x0F
#define KEY_M                                  0x10
#define KEY_N                                  0x11
#define KEY_O                                  0x12
#define KEY_P                                  0x13
#define KEY_Q                                  0x14
#define KEY_R                                  0x15
#define KEY_S                                  0x16
#define KEY_T                                  0x17
#define KEY_U                                  0x18
#define KEY_V                                  0x19
#define KEY_W                                  0x1A
#define KEY_X                                  0x1B
#define KEY_Y                                  0x1C
#define KEY_Z                                  0x1D
#define KEY_1_EXCLAMATION_MARK                 0x1E
#define KEY_2_AT                               0x1F
#define KEY_3_NUMBER_SIGN                      0x20
#define KEY_4_DOLLAR                           0x21
#define KEY_5_PERCENT                          0x22
#define KEY_6_CARET                            0x23
#define KEY_7_AMPERSAND                        0x24
#define KEY_8_ASTERISK                         0x25
#define KEY_9_OPARENTHESIS                     0x26
#define KEY_0_CPARENTHESIS                     0x27
#define KEY_ENTER                              0x28
#define KEY_ESCAPE                             0x29
#define KEY_BACKSPACE                          0x2A
#define KEY_TAB                                0x2B
#define KEY_SPACEBAR                           0x2C
#define KEY_MINUS_UNDERSCORE                   0x2D
#define KEY_EQUAL_PLUS                         0x2E
#define KEY_OBRACKET_AND_OBRACE                0x2F
#define KEY_CBRACKET_AND_CBRACE                0x30
#define KEY_BACKSLASH_VERTICAL_BAR             0x31
#define KEY_NONUS_NUMBER_SIGN_TILDE            0x32
#define KEY_SEMICOLON_COLON                    0x33
#define KEY_SINGLE_AND_DOUBLE_QUOTE            0x34
#define KEY_GRAVE ACCENT AND TILDE             0x35
#define KEY_COMMA_AND_LESS                     0x36
#define KEY_DOT_GREATER                        0x37
#define KEY_SLASH_QUESTION                     0x38
#define KEY_CAPS LOCK                          0x39
#define KEY_F1                                 0x3A
#define KEY_F2                                 0x3B
#define KEY_F3                                 0x3C
#define KEY_F4                                 0x3D
#define KEY_F5                                 0x3E
[... 142 lines cut ...]
#define KEY_KEYPAD_CLEAR                       0xD8
#define KEY_KEYPAD_CLEAR_ENTRY                 0xD9
#define KEY_KEYPAD_BINARY                      0xDA
#define KEY_KEYPAD_OCTAL                       0xDB
#define KEY_KEYPAD_DECIMAL                     0xDC
#define KEY_KEYPAD_HEXADECIMAL                 0xDD
#define KEY_LEFTCONTROL                        0xE0
#define KEY_LEFTSHIFT                          0xE1
#define KEY_LEFTALT                            0xE2
#define KEY_LEFT_GUI                           0xE3
#define KEY_RIGHTCONTROL                       0xE4
#define KEY_RIGHTSHIFT                         0xE5
#define KEY_RIGHTALT                           0xE6
#define KEY_RIGHT_GUI                          0xE7
```

Ref.: [usbh_hid_keybd.h](https://github.com/rtek1000/Datalogger_2039/blob/main/Software/Datalogger_2039%20-%20STM32/Datalogger2039_USB_Host_HID/Middlewares/ST/STM32_USB_Host_Library/Class/HID/Inc/usbh_hid_keybd.h)

> Yes, for the external HUBs support we have couple of tickets, this is the last one: [#12554 (comment)](https://github.com/espressif/esp-idf/issues/12554#event-10923708526)

Ok, thanks!

> Is there anything else I can help with or we can close this issue?

No, that's all, thanks for your attention!