espressif/esp-idf #12668: Heap Memory Allocation Failing

narangmayank, 2023-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

Heap memory should be allocated successfully as there is plenty of size available

### What is the actual behavior?

Heap memory allocation failed

### Steps to reproduce.

Just run the code snippet

```
#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "esp_log.h"

const char* TAG = "HELLO";

void app_main(void)
{
    printf("Hello world!\n");

    vTaskDelay(2000 / portTICK_PERIOD_MS);
    
    ESP_LOGI(TAG, "Free heap size: %lu\n", esp_get_free_heap_size());
    ESP_LOGI(TAG, "Free heap size (caps): %u\n", heap_caps_get_free_size(MALLOC_CAP_8BIT));

    char*ptr = (char*) malloc(170*1024);

    if(ptr == NULL)
    {
        ESP_LOGI(TAG, "early malloc failed.");
    }
    else 
    {
        free(ptr);
        ESP_LOGI(TAG, "early malloc success.");
    }
}
```

### Debug Logs.

```plain
I (0) cpu_start: App cpu up.
I (198) cpu_start: Pro cpu start user code
I (198) cpu_start: cpu freq: 160000000 Hz
I (198) cpu_start: Application information:
I (203) cpu_start: Project name:     hello_world
I (208) cpu_start: App version:      v5.0.2-dirty
I (214) cpu_start: Compile time:     Nov 26 2023 15:04:14
I (220) cpu_start: ELF file SHA256:  4b0533de59b6602a...
I (226) cpu_start: ESP-IDF:          v5.0.2-dirty
I (231) cpu_start: Min chip rev:     v0.0
I (236) cpu_start: Max chip rev:     v3.99
I (241) cpu_start: Chip rev:         v1.0
I (246) heap_init: Initializing. RAM available for dynamic allocation:
I (253) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (259) heap_init: At 3FFB2770 len 0002D890 (182 KiB): DRAM
I (265) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (271) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (278) heap_init: At 4008B864 len 0001479C (81 KiB): IRAM
I (285) spi_flash: detected chip: generic
I (289) spi_flash: flash io: dio
W (293) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (307) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
I (4311) HELLO: Free heap size: 301716

I (4311) HELLO: Free heap size (caps): 301716

I (4311) HELLO: early malloc failed.
```

### More Information.

NA

narangmayank, 2023-11-27:
@igrr Can you take a look into this.

SaketNer, 2023-11-28:
Here's @igrr said the issue is due to fragmentation. The heap_caps_get_free_size shows the total free size , not the continuous largest block. Hence even though you have enough space it is not continuous. The malloc command will allocate memory in continuous block else fail.

zikalino, 2023-11-28:
@narangmayank do you have any special need to allocate 170k? this is something we can't really fix, so we will probably close this ticket. Maybe if you could elaborate on your particular scenario, we could suggest a workaround?

narangmayank, 2023-11-29:
@zikalino I'm have a lot of data around 18KB and I was trying zlib decompression to reduce the data costs. I found a library to be helpful but this requires around 164KB of buffer to be allocated for its usage.

SoucheSouche, 2023-11-29:
@narangmayank, you could maybe use the external memory for this purpose. By enabling `SPIRAM` in `menuconfig`, you can use the external memory as a heap which on the esp32, represents approximately 4MB of available memory. You can then do the allocation with `heap_caps_malloc(your_size, MALLOC_CAP_SPIRAM)`. Other than that, I don't think you will find a heap big enough to allocate 170kB of data on esp32.

SoucheSouche, 2023-12-04:
@narangmayank, I will close the issue since it seems that you have now several options to handle your memory limitation problem.

Feel free to reopen if you have anu other concerns :)

closed 2023-12-04 by SoucheSouche