espressif/esp-idf #12669: External components behavior due to covert project from PIO to ESP-IDF

dm128, 2023-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf 5.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

PIO project contents component.mk files that are not connected to IDF builder. Recommended structure of the components folder doesn't contain folder of the components, however PIO successfully works with component.mk file inside components folder. Same situation with other component.mk files placed in main folder etc. Git cloned project was build in PIO with warnings and after that was imported in esp-idf with copy files option to esp-idf default workspace. CMakeList.txt file was generated during build PIO project and came into esp-idr without changes. As a import result metadata of the project lost parameters enrolled in all component.mk files. One of that parameters is COMPONENT_ADD_INCLUDEDIRS. I found only one mention of it in the official esp-idf documentation and have no idea how I can connect all component.mk files to use this and other parameter.
Github project link is here [https://github.com/Ebiroll/esp32_blackmagic](url)

### What is the actual behavior?

Project after import can't be built in a reason of external lib is not in include paths. Adding path in CMakeList.txt in folders makes other mistakes.

### Steps to reproduce.

1. Step is make clone of the Github project into PIO workspace and open it in VCS
2. Step is build the project in PIO/VCS with default environment 
3. Step is import project into esp-idf with copy files option into default esp-idf workspace
4. Step is adjust metadata of the project and try to build it in esp-idf
...

### Build or installation Logs.

```plain
At least error of unavailable resource will stop build:
esp32_blackmagic/main/gdb_if.h:25:10: fatal error: libopencm3/usb/usbd.h: No such file or directory
   25 | #include <libopencm3/usb/usbd.h>
```

### More Information.

I already seen discussions about more correct coo-existing metadata of projects environment in PIO and ESP-IDF. It's really great idea to make compatibility between different IDEs

zikalino, 2023-11-28:
@dm128 I will take a look into this issue and get back to you after looking into your github project!

ESP-Marius, 2024-11-19:
I dont see anything here that points at a specific bug with IDF :thinking: 

`component.mk` is using the old make build-system, which has been replaced with the new cmake based system now: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/build-system.html#migrating-from-gnu-make-build-system