espressif/esp-idf #12670: iBeacon example can not scan beacon signals from others in the receiver mode.

meko0917, 2023-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM-32E-N8

### Power Supply used.

USB

### What is the expected behavior?

I have tested ble_ibeacon example for the first time of using ESP32-WROOM-32E with ESP-IDF v5.1.2 newly installation.
I expected the result as described in that example's git page.
I (384) boot: Loaded app from partition at offset 0x10000
I (384) boot: Disabling RNG early entropy source...
I (385) cpu_start: Pro cpu up.\0x1b[0m
I (389) cpu_start: Starting app cpu, entry point is 0x40081010
I (0) cpu_start: App cpu up.
I (400) heap_init: Initializing. RAM available for dynamic allocation:
I (406) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (412) heap_init: At 3FFCCC88 len 00013378 (76 KiB): DRAM
I (418) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (425) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (431) heap_init: At 40090E58 len 0000F1A8 (60 KiB): IRAM
I (437) cpu_start: Pro cpu start user code\0x1b[0m
I (120) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (243) BTDM_INIT: BT controller compile version [44d04c1]

I (243) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (633) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
I (663) IBEACON_DEMO: register callback
I (329203) IBEACON_DEMO: ----------iBeacon Found----------

### What is the actual behavior?

I (409) boot: Loaded app from partition at offset 0x10000
I (409) boot: Disabling RNG early entropy source...
I (420) cpu_start: Multicore app
I (420) cpu_start: Pro cpu up.
I (421) cpu_start: Starting app cpu, entry point is 0x4008137c
0x4008137c: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (438) cpu_start: Pro cpu start user code
[... 231 lines cut ...]
I (465) cpu_start: ESP-IDF:          v5.1.2
I (470) cpu_start: Min chip rev:     v0.0
I (475) cpu_start: Max chip rev:     v3.99
I (480) cpu_start: Chip rev:         v3.1
I (485) heap_init: Initializing. RAM available for dynamic allocation:
I (492) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (498) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (504) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (510) heap_init: At 3FFC6E88 len 00019178 (100 KiB): DRAM
I (516) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (523) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (529) heap_init: At 400987C0 len 00007840 (30 KiB): IRAM
I (537) spi_flash: detected chip: gd
I (539) spi_flash: flash io: dio
I (544) coexist: coex firmware version: b6d5e8c
I (549) app_start: Starting scheduler on CPU0
I (554) app_start: Starting scheduler on CPU1
I (554) main_task: Started on CPU0
I (564) main_task: Calling app_main()
I (584) BTDM_INIT: BT controller compile version [ec4ac65]
I (584) BTDM_INIT: Bluetooth MAC: 64:b7:08:d5:14:46
I (594) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16
I (984) IBEACON_DEMO: register callback
I (984) main_task: Returned from app_main()
```

zhp0406, 2023-11-28:
Thank you for reporting the issue; I will check it.

zhp0406, 2023-11-28:
I (31) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (31) boot: compile time Nov 28 2023 16:07:46
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1f604h (128516) map
I (151) esp_image: segment 1: paddr=0002f62c vaddr=3ffbdb60 size=009ech (  2540) load
I (153) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7e024h (516132) map
I (343) esp_image: segment 3: paddr=000ae04c vaddr=3ffbe54c size=03ab8h ( 15032) load
I (349) esp_image: segment 4: paddr=000b1b0c vaddr=40080000 size=182fch ( 99068) load
I (403) boot: Loaded app from partition at offset 0x10000
I (403) boot: Disabling RNG early entropy source...
I (415) cpu_start: Multicore app
I (415) cpu_start: Pro cpu up.
I (415) cpu_start: Starting app cpu, entry point is 0x40081374
0x40081374: call_start_cpu1 at /home/zhanghaipeng/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (433) cpu_start: Pro cpu start user code
I (433) cpu_start: cpu freq: 160000000 Hz
I (433) cpu_start: Application information:
I (438) cpu_start: Project name:     ble_ibeacon_demo
I (443) cpu_start: App version:      v5.1.1-dirty
I (449) cpu_start: Compile time:     Nov 28 2023 16:10:48
I (455) cpu_start: ELF file SHA256:  20fd11d53d86a49b...
I (461) cpu_start: ESP-IDF:          v5.1.1-dirty
I (466) cpu_start: Min chip rev:     v0.0
I (471) cpu_start: Max chip rev:     v3.99 
I (476) cpu_start: Chip rev:         v3.1
I (481) heap_init: Initializing. RAM available for dynamic allocation:
I (488) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (494) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (500) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (506) heap_init: At 3FFC6B30 len 000194D0 (101 KiB): DRAM
I (512) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (519) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (525) heap_init: At 400982FC len 00007D04 (31 KiB): IRAM
I (533) spi_flash: detected chip: generic
I (536) spi_flash: flash io: dio
W (540) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (554) coexist: coex firmware version: 80b0d89
I (559) app_start: Starting scheduler on CPU0
I (563) app_start: Starting scheduler on CPU1
I (563) main_task: Started on CPU0
I (573) main_task: Calling app_main()
I (593) BTDM_INIT: BT controller compile version [946b762]
I (593) BTDM_INIT: Bluetooth MAC: 08:3a:8d:0d:9c:62
I (593) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1013) IBEACON_DEMO: register callback
I (1013) main_task: Returned from app_main()
I (1053) IBEACON_DEMO: ----------iBeacon Found----------
I (1053) IBEACON_DEMO: Device address:: 30 ae a4 80 56 0e 
I (1053) IBEACON_DEMO: Proximity UUID:: fd a5 06 93 a4 e2 4f b1 af cf c6 eb 07 64 78 25 
I (1063) IBEACON_DEMO: Major: 0x27b7 (10167)
I (1063) IBEACON_DEMO: Minor: 0xf206 (61958)
I (1073) IBEACON_DEMO: Measured power (RSSI at a 1m distance):-59 dbm
I (1073) IBEACON_DEMO: RSSI of packet:-22 dbm
I (1103) IBEACON_DEMO: ----------iBeacon Found----------
I (1103) IBEACON_DEMO: Device address:: 30 ae a4 80 56 0e 
I (1103) IBEACON_DEMO: Proximity UUID:: fd a5 06 93 a4 e2 4f b1 af cf c6 eb 07 64 78 25 
I (1113) IBEACON_DEMO: Major: 0x27b7 (10167)
I (1113) IBEACON_DEMO: Minor: 0xf206 (61958)
I (1123) IBEACON_DEMO: Measured power (RSSI at a 1m distance):-59 dbm
I (1123) IBEACON_DEMO: RSSI of packet:-22 dbm
I (1153) IBEACON_DEMO: ----------iBeacon Found----------
I (1153) IBEACON_DEMO: Device address:: 30 ae a4 80 56 0e 
I (1153) IBEACON_DEMO: Proximity UUID:: fd a5 06 93 a4 e2 4f b1 af cf c6 eb 07 64 78 25 
I (1163) IBEACON_DEMO: Major: 0x27b7 (10167)
I (1163) IBEACON_DEMO: Minor: 0xf206 (61958)
I (1173) IBEACON_DEMO: Measured power (RSSI at a 1m distance):-59 dbm
I (1173) IBEACON_DEMO: RSSI of packet:-18 dbm
I (1403) IBEACON_DEMO: ----------iBeacon Found----------

zhp0406, 2023-11-28:
I tested using the version you provided, with one instance in the sender role and another in the receiver role. In the sender role, the sender calls the esp_ble_gap_start_advertising API for broadcasting, while in the receiver role, it uses the esp_ble_gap_start_scanning API for scanning.

The test results indicate that the receiver successfully received the data, which is as expected.

Could you please check if there is no sender role nearby on your end?

[image]

meko0917, 2023-11-29:
I tested again on the other PC, the result is same as yours.
I guess It was the problem by the PC environment.
Sorry and thanks.

closed 2023-11-29 by Alvin1Zhang