espressif/esp-idf #12671: set_rtc_memory_crc() unfound for ESP32-S3

peteDDD, 2023-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

see_full_toolset_in_log_below

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

rExpect required functions, specifically set_rtc_memory_crc() (and others listed below), to be found when building examples/system/deep_sleep_wake_stub

### What is the actual behavior?

Expect to be able to build the examples/system/deep_sleep_wake_stub on ESP32-S3
Per the TCM,  10.6: RTC Boot, on page 565, setting the wake_stub, from within a wake_stub requires setting the rtc crc which in all examples is done using set_rtc_memory_crc() however this and other necessary code for working within wake-stubs is not found.

components/esp_hw_support/sleep_wake_stub.c requires  set_rtc_memory_crc() which should precumably be in 
components/esp_rom/esp32s3/ld/esp32s3.rom.ld  like it is for other esp32 processors but there is no definition of  set_rtc_memory_crc there (or anywhere else I can find)

Similarly, the source for the following are also not found:
rtc_cntl_ll_get_wakeup_cause
rtc_cntl_ll_sleep_enable
xt_utils_wait_for_intr
rtc_cntl_ll_time_to_count
rtc_cntl_ll_get_rtc_time

### Steps to reproduce.

1: simply try to build examples/system/deep_sleep_wake using visual studio code with platformio
2: even adding a copy of sleep_wake_stub.c to the source files does not help
3: see log below

### Build or installation Logs.

```plain
Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: espidf, arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - framework-espidf @ 3.40405.230623 (4.4.5)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\src\rtc_wake_stub_example.o
Linking .pio\build\esp32-s3-devkitc-1\firmware.elf
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/src/rtc_wake_stub_example.o:(.literal.wake_stub_example+0x18): undefined reference to `esp_wake_stub_get_wakeup_cause'
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/src/rtc_wake_stub_example.o:(.literal.wake_stub_example+0x20): undefined reference to `esp_wake_stub_uart_tx_wait_idle'
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/src/rtc_wake_stub_example.o:(.literal.wake_stub_example+0x24): undefined reference to `esp_wake_stub_set_wakeup_time'
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/src/rtc_wake_stub_example.o:(.literal.wake_stub_example+0x28): undefined reference to `esp_wake_stub_sleep'
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/src/rtc_wake_stub_example.o: in function `wake_stub_example':
C:/Users/peted/Documents/PlatformIO/Projects/Newer_sleep_wake_stub_test/src/rtc_wake_stub_example.c:44: undefined reference to `esp_wake_stub_get_wakeup_cause'
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Users/peted/Documents/PlatformIO/Projects/Newer_sleep_wake_stub_test/src/rtc_wake_stub_example.c:48: undefined reference to `esp_wake_stub_uart_tx_wait_idle'
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Users/peted/Documents/PlatformIO/Projects/Newer_sleep_wake_stub_test/src/rtc_wake_stub_example.c:65: undefined reference to `esp_wake_stub_set_wakeup_time'
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Users/peted/Documents/PlatformIO/Projects/Newer_sleep_wake_stub_test/src/rtc_wake_stub_example.c:68: undefined reference to `esp_wake_stub_uart_tx_wait_idle'
c:/users/peted/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Users/peted/Documents/PlatformIO/Projects/Newer_sleep_wake_stub_test/src/rtc_wake_stub_example.c:71: undefined reference to `esp_wake_stub_sleep'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1
```

### More Information.

As I search the whole source tree, I can't find source for these functions for ESP32-S3

peteDDD, 2023-11-26:
platformio.ini
```
[env:esp32-s3-devkitc-1]
platform = espressif32 ;@6.4.0
board = esp32-s3-devkitc-1
framework = espidf, arduino

board_build.arduino.memory_type = dio_opi
monitor_speed = 115200
monitor_filters = direct, esp32_exception_decoder

board_build.partitions = default_partitions.csv

build_flags = 
	-I./src/
	-DCORE_DEBUG_LEVEL=5
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-fmax-errors=5

lib_ldf_mode = deep
```

igrr, 2023-11-26:
Deep sleep wake stub example was added in https://github.com/espressif/esp-idf/commit/a6f7035c8c7badd83e5f348b92704b3c499dbea8, which is part of IDF v5.1 release. This example won't work with earlier versions of IDF, and you are using v4.4.5.

peteDDD, 2023-11-26:
> Deep sleep wake stub example was added in a6f7035

Thank you for your very prompt response.  There were other things that
broke after 4.4 which kept me there but I'll try to update now.

Thanks again for your help.

esp-lis, 2025-03-10:
@peteDDD Has this issue been resolved?

peteDDD, 2025-03-10:
> [@peteDDD](https://github.com/peteDDD) Has this issue been resolved?

This issue was active over 15 months ago... It did not work so I moved on and solved it using other code.

closed 2025-03-11 by Alvin1Zhang