espressif/esp-idf #12672: OTA fails after upgrade from v4.4.5 to v5.1.1

hobieboy, 2023-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We have just updated from v4.4.5 to v5.1.1 and find that OTA does not work anymore. To confirm it is not our implementation, I ran the Advanced HTTPS OTA example provided within v5.1.1 and exhibit exact same problem.

I have attached 2 screenshots - one reported the failure reason by ESP32-S2 (mbedtls_ ssl_ handshake returned -0×2700).
And the other from the server.

I have tried this with both the test certificate provided by the example or the valid certificate that we generated (and worked in v4.4.5) with same error messages. As for options, we tried both skip common name check = true and false and it doesn't make any difference.

Looking for help as to what is wrong here?

thanks.
[image]
[image]

mahavirj, 2023-11-28:
@hobieboy 

- Can you please share logs with mbedtls debug option set and log level set to verbose?
- FWIW, please refer to migration guide for 5.x [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/protocols.html#mbed-tls) to see if you are using any deprecated (or default disabled) cipher-suites

nileshkale123, 2023-12-06:
Hello @hobieboy

Partial OTA updates are facilitated by servers that support the HTTP range-header mechanism. For a more in-depth understanding of range-based requests, you can refer to [Range Retrieval Requests](https://www.ietf.org/archive/id/draft-ietf-httpbis-p5-range-09.html#range.retrieval.requests).

Consider initiating a Python-based HTTPS server instead of OpenSSL. You can find detailed instructions for launching a Python-based server using the pytest_simple_ota.py file in the corresponding README.md. Below are the specific steps:
 
Firstly, you have to enable pytest to start the server using this way. The steps for it are as follows:
1. Navigate to the main directory where you have cloned esp-idf.
2. Run the command to enable pytest
     - ' . ./install.sh --enable-pytest '
     - ' . ./export.sh '

Once the pytest got enabled, here are detailed steps for initiating the server:
1. Navigate to the 'simple_ota_example' directory.
2. Create a directory named 'server'.
3. Go into the 'server' directory.
4. Generate server certificates using the command specified in the README file:
     - 'openssl req -x509 -newkey rsa:2048 -keyout ca_key.pem -out ca_cert.pem -days 365 -nodes'
5. Copy the generated 'ca_cert.pem' file to 'advanced_https_ota/server_certs/ca_cert.pem' using command:
     - 'cp ./ca_cert.pem ./../../advanced_https_ota/server_certs/ca_cert.pem'
6. Navigate back to the 'simple_ota_example' directory
7. Execute the following command to start the server:
     - 'python3 pytest_simple_ota.py <binary file path> 8070 ./server/'

hmalpani, 2023-12-11:
Hello @hobieboy 

Have you tried OTA using python server as suggested above?

hobieboy, 2023-12-11:
Sorry for the late response; been over busy. I have not tried the Python method since I was able to make OTA work without making use of the range header request. I'd like to better understand the benefits/costs of running a Python HTTPS vs a simple OpenSSL before switching over.
But as for this particular thread, I think I can close it since my main issue is resolved.
Thanks to all for the help!

closed 2023-12-11 by hobieboy