espressif/esp-idf #12674: ETH drivers with custom registers read write API .

MrutyunjayPatel, 2023-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
IDF version : 5.1.2

We are using eth LAN8720 for a product in production.
due to the issue of LAN chips click interfering with boot strapping on pin GPIO0 of esp32. 
we have revised the new board with RTL8201 chip.

The the issue is LED indicators, as the default led behavior of the both the ethernet chips are different   

Is their any option in the driver to set the led properties as per the requirements, as The RTL chip has the option to set he led similar to LAN8720.      

if we could write LED_sel[1:0]	with value of "0x01"  the led will be same as lan 8720 chip.

From the RTL datasheet 
**LED  Function Selection RTL8201**

Address: 19:5~4  LED_sel[1:0]
 
[LEDSel == Page 7 Register 19 (Interrupt, WOL Enable, and LEDs Function Registers ]
_______________________

```
LED_sel[1:0]	00		01		10		11 (default)
LED0		ACTALL		LinkALL/ACTALL	Link10/ACTALL 	LINK10/ACT10
LED1		LINK100		LINK100		LINK100		LINK100/ACT100

```
_______________________
LAN8720 default 
```
LED 0   LinkALL/ACTALL
LED 1   LINK100/LINK10

```

Thanks and Regards
Mrutyunjay

kostaond, 2023-12-01:
Option to read/write PHY custom PHY registers has been actually recently added to ESP-IDF, please check https://github.com/espressif/esp-idf/commit/78f791d4d5072f084196d015ad8ddd682d6fd853

Will soon synced to v5.1 too.

ronie, 2023-12-02:
@kostaond am i correct in understanding you can only read/write registers up to 0x1F ?

for a project that i'm working on, i need read access to registers 0x171 to 0x018A of the PHY,
but when i try that the returned data is incorrect.

i'm not familiar with the esp-idf codebase, but after a bit of digging around,
i think the reg address is limited to 5 bits (so max 0x1F) here:
https://github.com/espressif/esp-idf/blob/76923a7a7f92ef81a2599d44c93c038143cc55f9/components/soc/esp32/include/soc/emac_mac_struct.h#L75

kostaond, 2023-12-05:
Closing the issue since infrastructure to access custom Ethernet PHY registers is already in place. Feel free to reopen if you face any issue to access it.

closed 2023-12-05 by kostaond

MrutyunjayPatel, 2024-01-02:
hi ,
for IDF 5.1 Added  following coed to achieve the led control settings. until we port the code to IDF 5.3x 

-------------------------------------------------------------------------------------------------
add new function to idf components files

esp-idf-v5.1\components\esp_eth\include\esp_eth_phy_802_3.h

esp-idf-v5.1\components\esp_eth\src\esp_eth_phy_802_3.c 

```
esp_err_t esp_eth_phy_802_3_update_ledFunction(phy_802_3_t *phy_802_3, uint32_t *ledfn )
{
    esp_err_t ret = ESP_OK;
    esp_eth_mediator_t *eth = phy_802_3->eth;
    uint32_t led  =0;
    uint32_t cpage=0;
    uint32_t mpage=0;
    uint32_t dpage=0;

		//ledfn valid values 0x0,0x1,0x2,0x3(default).

    // Applicable for RTL8201 led function
    // set address to read write to page 7 , write to reg 31(0x1F) value 7(0x07).
    // Read page7 register19 and clear bits 4,5 : AND ing with 0xFFCF to set bit4 bit 5 as 00 (1100 1111).
    // set the bit 4,5 accordingly : OR ing with 0x0X00,X=0,1,2,3'
		//Write back to page7 register 19.
		//read current page value. 
    ESP_GOTO_ON_ERROR(eth->phy_reg_read (eth, phy_802_3->addr, 31, &cpage   ), err, TAG, "read  reg 31  failed"   );
    //write to set up to read write to page to 7.
		mpage= 0x0007;
    ESP_GOTO_ON_ERROR(eth->phy_reg_write(eth, phy_802_3->addr, 31, mpage    ), err, TAG, "write reg 31 0x07 failed");
    //Read Page7 register19.
    ESP_GOTO_ON_ERROR(eth->phy_reg_read (eth, phy_802_3->addr, 19, &led     ), err, TAG, "read Ledfn failed"   );
    //ESP_LOGE("ETH_PHY_802_3_reg","LEDfn      read =0x%08X",led);
    // Clear bits 4,5
    led = led & 0xFFFFFFCF;// 0b1111 1111 1100 1111 == FFCF
  //ESP_LOGE("ETH_PHY_802_3_reg","LEDfn 4,5 clear =0x%08X",led);
    // read =0x0000C034   // 0b1100 0000 0011 0100 default
    // set bits 4,5       // 0b0000 0000 00XX 0000
    // select any one of following 4 values for bit 4,5
    //led = led | 0x00000000 ; // ledfn XX00XXXX 0  LED0=ACT-ALL              LED1=LINK100
      led = led | 0x00000100 ; // ledfn XX01XXXX 1  LED0=LINK-ALL/ACT-ALL     LED1=LINK100
    //led = led | 0x00000200 ; // ledfn XX10XXXX 2  LED0=LINK-10 /ACT-ALL     LED1=LINK100
    //led = led | 0x00000300 ; // ledfn XX11XXXX 3  LED0=LINK10/ACT10         LED1=LINK100/ACT100
  //ESP_LOGE("ETH_PHY_802_3_reg","LEDfn     write =0x%08X",led);
    ESP_GOTO_ON_ERROR(eth->phy_reg_write(eth, phy_802_3->addr, 19, led  ), err, TAG, "write Ledfn failed");
    ESP_GOTO_ON_ERROR(eth->phy_reg_write(eth, phy_802_3->addr, 31, cpage), err, TAG, "Write page2 failed");
  //ESP_LOGE("ETH_PHY_802_3_reg","cpage = 0x%08X mpage =0x%08X dpage=0x%08X ledval=0x%08X ", cpage, mpage, dpage , led );

    *ledfn = led ;

    return ESP_OK;
err:
    return ret;
}
```
-------------------------------------------------------------------------------
modify the function in the file, To call the above new function function 
esp-idf-v5.1\components\esp_eth\src\esp_eth_phy_rtl8201.c

-------------------------------------------------------------------------------
```
static esp_err_t rtl8201_init(esp_eth_phy_t *phy)
{
    esp_err_t ret = ESP_OK;
    phy_802_3_t *phy_802_3 = esp_eth_phy_into_phy_802_3(phy);

    /* Basic PHY init */
    ESP_GOTO_ON_ERROR(esp_eth_phy_802_3_basic_phy_init(phy_802_3), err, TAG, "failed to init PHY");

    /* Check PHY ID */
    uint32_t oui;
    uint8_t model;
    uint32_t ledfun;

    ESP_GOTO_ON_ERROR(esp_eth_phy_802_3_read_oui(phy_802_3, &oui), err, TAG, "read OUI failed");
    ESP_GOTO_ON_ERROR(esp_eth_phy_802_3_read_manufac_info(phy_802_3, &model, NULL), err, TAG, "read manufacturer's info failed");
    ESP_GOTO_ON_FALSE(oui == 0x732 && model == 0x1, ESP_FAIL, err, TAG, "wrong chip ID");

    /* Set RTL8201 LedFun to Desired value *********************/
    ESP_GOTO_ON_ERROR(esp_eth_phy_802_3_update_ledFunction(phy_802_3, &ledfun), err, TAG, "update ledfun  failed");
    ESP_LOGE("ESP_ETH_PHY_RTL8201.C","RTL8201 Ledfunction Value = 0x%08X ", ledfun );
    return ESP_OK;
err:
    return ret;
}

```

Regards
Mrutyunjay