espressif/esp-idf #12677: I get a lot of undefined CONFIG_... errros

shlomo-edu, 2023-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-491-g30eb2918c6

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I clone the project.
I am trying to compile the BLE gatt_server example
1. I see in Visual Studio Code that all the include files from "freetos" and below, are all marked with red underline (which means that the it can't find them).
When I try to compile it (with 
I get errors.
The fits error is: unknown type name 'esp_bluedroid_config_t'
 on this line:
`esp_bluedroid_config_t bluedroid_cfg = BT_BLUEDROID_INIT_CONFIG_DEFAULT();
`
I also tried to copy the code (the C file) to another project that I did manage to compile (HID) and there I didn't have the problem with the red line (it found the include files), but I had several undeclared CONFIG_ variable such as:
CONFIG_BT_LE_COEX_PHY_CODED_TX_RX_TLIM_EFF

### What is the actual behavior?

1. I should find the include files.
2. It should compile.

### Steps to reproduce.

1. clone the ESP-IDF
2. Open Visual Sudio Clode
2. Open Folder: C:\ESP32-BLE\esp-idf\examples\bluetooth\bluedroid\ble\gatt_server
3. Click on: View->Command Palette...
4. Select: ESP-IDF: Set Espressif device rarget
5. Click on: View->Command Palette...
6: Select: ESP-IDF: Build, Flush, Start...

### Build or installation Logs.

```plain
[18/32] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/ble_server.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/ble_server.c.obj 
C:\Users\shlomoa\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -IC:/ESP32-WROOM/hid/build/config -IC:/ESP32-WROOM/hid/main -IC:/Users/shlomoa/esp/esp-idf/components/newlib/platform_include -IC:/Users/shlomoa/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/shlomoa/esp/esp-idf/components/freertos/esp_additions/include/freertos -IC:/Users/shlomoa/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/shlomoa/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_hw_support/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/shlomoa/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -IC:/Users/shlomoa/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -IC:/Users/shlomoa/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -IC:/Users/shlomoa/esp/esp-idf/components/heap/include -IC:/Users/shlomoa/esp/esp-idf/components/log/include -IC:/Users/shlomoa/esp/esp-idf/components/soc/include -IC:/Users/shlomoa/esp/esp-idf/components/soc/esp32s3/. -IC:/Users/shlomoa/esp/esp-idf/components/soc/esp32s3/include -IC:/Users/shlomoa/esp/esp-idf/components/hal/esp32s3/include -IC:/Users/shlomoa/esp/esp-idf/components/hal/include -IC:/Users/shlomoa/esp/esp-idf/components/hal/platform_port/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_rom/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_rom/include/esp32s3 -IC:/Users/shlomoa/esp/esp-idf/components/esp_rom/esp32s3 -IC:/Users/shlomoa/esp/esp-idf/components/esp_common/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_system/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_system/port/soc -IC:/Users/shlomoa/esp/esp-idf/components/esp_system/port/include/private 
-IC:/Users/shlomoa/esp/esp-idf/components/xtensa/include -IC:/Users/shlomoa/esp/esp-idf/components/xtensa/esp32s3/include -IC:/Users/shlomoa/esp/esp-idf/components/lwip/include -IC:/Users/shlomoa/esp/esp-idf/components/lwip/include/apps -IC:/Users/shlomoa/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/shlomoa/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/shlomoa/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/shlomoa/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/shlomoa/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/shlomoa/esp/esp-idf/components/efuse/include -IC:/Users/shlomoa/esp/esp-idf/components/efuse/esp32s3/include -IC:/Users/shlomoa/esp/esp-idf/components/driver/include -IC:/Users/shlomoa/esp/esp-idf/components/driver/deprecated -IC:/Users/shlomoa/esp/esp-idf/components/driver/esp32s3/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_pm/include -IC:/Users/shlomoa/esp/esp-idf/components/mbedtls/port/include -IC:/Users/shlomoa/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/shlomoa/esp/esp-idf/components/mbedtls/mbedtls/library -IC:/Users/shlomoa/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_app_format/include -IC:/Users/shlomoa/esp/esp-idf/components/bootloader_support/include -IC:/Users/shlomoa/esp/esp-idf/components/bootloader_support/bootloader_flash/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_partition/include -IC:/Users/shlomoa/esp/esp-idf/components/app_update/include -IC:/Users/shlomoa/esp/esp-idf/components/spi_flash/include -IC:/Users/shlomoa/esp/esp-idf/components/pthread/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_timer/include -IC:/Users/shlomoa/esp/esp-idf/components/app_trace/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_event/include -IC:/Users/shlomoa/esp/esp-idf/components/nvs_flash/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_phy/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_phy/esp32s3/include -IC:/Users/shlomoa/esp/esp-idf/components/vfs/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_netif/include -IC:/Users/shlomoa/esp/esp-idf/components/wpa_supplicant/include -IC:/Users/shlomoa/esp/esp-idf/components/wpa_supplicant/port/include -IC:/Users/shlomoa/esp/esp-idf/components/wpa_supplicant/esp_supplicant/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_wifi/include -IC:/Users/shlomoa/esp/esp-idf/components/unity/include -IC:/Users/shlomoa/esp/esp-idf/components/unity/unity/src -IC:/Users/shlomoa/esp/esp-idf/components/cmock/CMock/src -IC:/Users/shlomoa/esp/esp-idf/components/console -IC:/Users/shlomoa/esp/esp-idf/components/http_parser -IC:/Users/shlomoa/esp/esp-idf/components/esp-tls -IC:/Users/shlomoa/esp/esp-idf/components/esp-tls/esp-tls-crypto -IC:/Users/shlomoa/esp/esp-idf/components/esp_adc/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_adc/interface -IC:/Users/shlomoa/esp/esp-idf/components/esp_adc/esp32s3/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_adc/deprecated/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_eth/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_gdbstub/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_gdbstub/xtensa -IC:/Users/shlomoa/esp/esp-idf/components/esp_gdbstub/esp32s3 -IC:/Users/shlomoa/esp/esp-idf/components/esp_hid/include -IC:/Users/shlomoa/esp/esp-idf/components/tcp_transport/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_http_client/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_http_server/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_https_ota/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_lcd/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_lcd/interface -IC:/Users/shlomoa/esp/esp-idf/components/protobuf-c/protobuf-c -IC:/Users/shlomoa/esp/esp-idf/components/protocomm/include/common -IC:/Users/shlomoa/esp/esp-idf/components/protocomm/include/security -IC:/Users/shlomoa/esp/esp-idf/components/protocomm/include/transports -IC:/Users/shlomoa/esp/esp-idf/components/esp_local_ctrl/include -IC:/Users/shlomoa/esp/esp-idf/components/esp_psram/include -IC:/Users/shlomoa/esp/esp-idf/components/espcoredump/include -IC:/Users/shlomoa/esp/esp-idf/components/espcoredump/include/port/xtensa -IC:/Users/shlomoa/esp/esp-idf/components/wear_levelling/include -IC:/Users/shlomoa/esp/esp-idf/components/sdmmc/include -IC:/Users/shlomoa/esp/esp-idf/components/fatfs/diskio -IC:/Users/shlomoa/esp/esp-idf/components/fatfs/vfs -IC:/Users/shlomoa/esp/esp-idf/components/fatfs/src -IC:/Users/shlomoa/esp/esp-idf/compone [... cut ...]
[106/106] cmd.exe /C "cd /D C:\ESP32-WROOM\hid\build\bootloader\esp-idf\esptool_py && C:\Users\shlomoa\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:/Users/shlomoa/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/ESP32-WROOM/hid/build/bootloader/bootloader.bin"
Bootloader binary size 0x5030 bytes. 0x2fd0 bytes (37%) free.
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

Halley826, 2023-11-27:
Please use the following command to update the submodules: git submodule update.
I have tested using the commit you provided and it is functioning properly:
[image]

zhp0406, 2023-11-28:
@shlomo-edu 
Have you checked if the issue still exists after using 'git submodule update'?

shlomo-edu, 2023-11-28:
It probly solved the unresolved issues, but now I get the following error:
C:/Users/shlomoa/esp/esp-idf/components/xtensa/include/xtensa/xtensa_api.h:104:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  104 | {
      | ^
line 104 is the { in this code:
```
/*
-------------------------------------------------------------------------------
  Call this function to set the specified (s/w) interrupt.
-------------------------------------------------------------------------------
*/
static inline void xt_set_intset(unsigned int arg)
{
    xthal_set_intset(arg);
}

```
And then the rest of the file is full of errors, so I assume that the problem is:
static inline void xt_set_intset(unsigned int arg)
but I don't know what?

zhp0406, 2023-11-29:
The error still exists within the components, indicating that not all updates to the components have taken effect.

zhp0406, 2023-11-29:
For the installation of the tool, you can take a look at the documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html

At this point, none of the examples should be buildable. Not only the BLE examples, because your environment has not been set up yet.

shlomo-edu, 2023-11-29:
I downloaded the: Universal Online Installer 2.23 Windows 10, 11 Size: 4 MB
From: https://dl.espressif.com/dl/esp-idf/?idf=4.4
I installed it. It asked me where to install it, and I gave the path: to C:\ESP-NEW
But, it installed something in C:\ESP-NEW and some other stuff in: C:\Espressif
See the attached picture.
At the end it displayed a message to build: idf.py build
So I tried to do it in: C:\ESP-NEW, but it doesn't have CMakeLists.txt
So I tried to compile in: C:\Espressif\frameworks\esp-idf-v5.1.2
But now I receive the following errors:
1. -- Detecting CXX compiler ABI info - failed
2. The C++ compiler   "C:/MinGW/bin/mingw32-gcc.exe"  is not able to compile a simple test program.
3. mingw32-gcc.exe: error: CreateProcess: No such file or directory
    ninja: build stopped: subcommand failed.  
  
[image]

shlomo-edu, 2023-11-29:
But now it does compile from Visual Studio Code.

closed 2023-11-29 by shlomo-edu