espressif/esp-idf #12678: l2cap client can't connect the other device

jjj19950710, 2023-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

After the bt_l2cap_client searches for the service, the l2cap client can initialize normally，and connect any devices.****

### What is the actual behavior?

Now bt_l2cap_client initialize failed,and send disconnect request.
the log is: 
```
I (37361) L2CAP_TAG: Found a target device, address 58:1c:f8:e6:cb:e7, name smylbook3
I (37371) L2CAP_TAG: Cancel device discovery ...
I (37381) L2CAP_TAG: event: 1
W (39241) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
I (39241) L2CAP_TAG: event: 16

I (39501) SDP_TAG: ESP_SDP_SEARCH_COMP_EVT: status:0

I (39501) SDP_TAG: Remote device address: 58:1c:f8:e6:cb:e7

I (39501) SDP_TAG: Remote device record count: 1

I (39511) SDP_TAG: Remote device rfcomm channel number: -1

I (39511) SDP_TAG: Remote device l2cap psm: 0xffffffff

I (39521) L2CAP_TAG: ESP_BT_L2CAP_CL_INIT_EVT: status:1

W (41501) BT_HCI: hci cmd send: disconnect: hdl 0x81, rsn:0x13
W (41591) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x16
I (41591) L2CAP_TAG: event: 17
```

### Steps to reproduce.

1. Change the name of the Bluetooth device that needs to be connected to a phone or computer Bluetooth.
```
#define TARGET_DEVICE_NAME            "smylbook3"
```
2.Change uuid:
```
static const uint8_t  UUID_UNKNOWN[] = {0x00, 0x00, 0x18, 0x01, 0x00, 0x00, 0x10, 0x00,
                                           0x80, 0x00, 0x00, 0x80, 0x5F, 0x9B, 0x34, 0xFB
                                       };
```
3.Compile and Run by idf.py

xiongweichao, 2024-03-05:
Hi @jjj19950710 ,

Sorry for replying to you so late. Because GATT (Generic Attribute) does not include the `GoepL2capPsm attribute`, l2cap psm cannot be found. This example is intended for use in developing OBEX functionality. 
If you want to establish a gatt connection with a phone, I suggest you use this [example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_client).

Thanks

xiongweichao, 2024-03-26:
@jjj19950710 May I ask if the problem still exists?

Sherry616, 2024-04-02:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.