espressif/esp-idf #12679: Cache inconsistency when using esp_partition_mmap and esp_partition_write_raw

pgrawehr, 2023-11-27:
### Board

ESP32 Dev Module

### Device Description

Using an ESP32 Dev Module with 4MB of flash

### Hardware Configuration

No additional hardware required

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Now this gets a bit complicated, I hope I can explain it in an understandable way. I have a (large) program that uses `esp_partition_write_raw` to write data sequentially to a special flash partition. That partition is simultaneously mapped to RAM using `esp_partition_mmap`, so that I can read from it as if it was RAM. 

For safety, I am doing a `memcmp` after writing each block, to make sure the data gets written correctly. With Arduino Core V2.X this always worked flawlessly, but now with 3.X I suddenly get comparison errors between what I've written to the flash and what I read back. When using `esp_partition_read_raw` to force a read from the flash, the comparison always succeeds. 

I first thought I had a broken flash chip, so I wrote the attached flash test program, that first didn't show any error at all, but then I reduced the block size and whoa, the bug showed up. (see comment in line 55, "int writeLen = ...")

At the moment, I have no other explanation than assuming that writing to flash doesn't update the CPU cache line we've just been reading from. Is there a way to force invalidating the CPU cache? As said, I have never observed this issue with the 2.X core.

The sample code below requires this `partitions.csv` file:

```
# ESP32 partition map. Total flash size reserved 4 MiB
# Name        | Type     | SubType | Offset  | Size    | Flags
nvs,            data,      nvs,      0x9000,   0x6000,
phy_init,       data,      phy,      0xf000,   0x1000,
factory,        app,       factory,  0x10000,  0x64000,
remaining,      data,      0x06,0x74000,  0x38C000,
```

### Sketch

```cpp
#include "esp_partition.h"
const char* FLASH_TAG = "Flash";
const esp_partition_t* partition; // the test partition
size_t length; // The partition length
size_t blockSize; // the size of a flash block
byte* testBuf; 
byte* mappedAddress; // Base address of memory mapped region
uint32_t map_handle; // Memory map handle
void setup() {
  Serial.begin(115200);
[... 193 lines cut ...]
[ 22202][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0x140
[ 22211][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0x168
[ 22220][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0x190
[ 22229][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0x1b8
[ 22238][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0x1e0
....
.... Many lines....
....
[ 33148][E][ESP_FlashTest.ino:67] TestPattern(): [Flash] Error verifying flash with mapped access
[ 43156][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0xc030
[ 43165][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0xc058
[ 43173][E][ESP_FlashTest.ino:67] TestPattern(): [Flash] Error verifying flash with mapped access
[ 53182][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0xc080
[ 53191][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0xc0a8
[ 53199][E][ESP_FlashTest.ino:67] TestPattern(): [Flash] Error verifying flash with mapped access
[ 63208][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0xc0d0
[ 63217][I][ESP_FlashTest.ino:56] TestPattern(): [Flash] Flashing pattern 0x75 to offset 0xc0f8
[ 63225][E][ESP_FlashTest.ino:67] TestPattern(): [Flash] Error verifying flash with mapped access
.... 
.... More stuff
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igrr, 2023-11-27:
Thanks for reporting the issue @pgrawehr. We do have a test for this scenario, but seems like it uses the legacy `spi_flash_write*` API. Perhaps there is a difference in the behavior of `esp_flash_*` API which we didn't consider. We'll investigate this.

https://github.com/espressif/esp-idf/blob/master/components/spi_flash/test_apps/flash_mmap/main/test_flash_mmap.c#L438-L451

polhenarejos, 2024-04-10:
I have the same problem as @pgrawehr . Any news or workaround on this?

pgrawehr, 2024-04-14:
@polhenarejos No, I did not really find a workaround yet. The problem doesn't seem to occur for me if I don't read the memory until everything is written. But there's no guarantee for that, I suppose.

polhenarejos, 2024-04-15:
What I'm doing to "flush" is to `unmap()` and `mmap()` again, but I am not sure how intensive is this.

pgrawehr, 2024-04-15:
If I remember correctly, that didn't help for me. But you can try it with the sketch above.