espressif/esp-idf #12680: 看门狗没有如期工作

aboyuanna, 2023-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-dirty

### Espressif SoC revision.

ESP32-WROOM-32E-N4 chip revision: v3.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

定制版

### Power Supply used.

External 3.3V

### What is the expected behavior?

我正在开发一个智能插座，当我通过按键切换继电器时，芯片会被电磁干扰导致程序跑飞了，这时看门狗应该重启芯片。

### What is the actual behavior?

按键按下无反应，log输出一堆乱码，看门狗没有重启芯片

### Steps to reproduce.

这个问题在我们的电路板上才能复现。

### Debug Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:10028
ho 0 tail 12 room 4
load:0x40078000,len:17452
load:0x40080400,len:4620
entry 0x40080694
[0;32mI (30) boot: ESP-IDF v5.0.1 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 13:41:25[0m
D (30) bootloader_flash: XMC chip detected by RDID (00204016), skip.[0m
D (35) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)[0m
[0;32mI (42) boot: chip revision: v3.0[0m
D (46) qio_mode: Probing for QIO mode enable...[0m
D (51) qio_mode: Raw SPI flash chip id 0x204016[0m
D (55) qio_mode: Manufacturer ID 0x20 chip ID 0x4016[0m
[0;32mI (60) qio_mode: Enabling default flash chip QIO[0m
D (66) qio_mode: Initial flash chip status 0x2[0m
D (70) qio_mode: QIO mode already enabled in flash[0m
D (75) qio_mode: Enabling QIO mode...[0m
[... 512 lines cut ...]
X+v
jq�|�Aj�*�ɪ�R�	����a,�ˢBA,��|����*�����T���-
H+���|��kAn �(X����Kj|�n���l
*r��P� ��*��-�����-��(N�Dbn`V"Z����R
�`;ib~&�R`�L�8~k�m����o�a�be�v%C|�t�-���'�X��'p�,P��PB�X�����,��'p�,P�X~�-jS�|�Aj�G�H~��R�	����a,�ˢBA,��*
��*�����T����-
�+��|���*���ɪ�P�+��𪭹�,���*r��P۠��*��-����--��*f�f�vsebf(����"��jJ$dJ�%�ɛf8�~��r����o���be�jN8깪t�,��t�-9��'�X���'�X���B�-����B�X����jX#@)�z�EX��kt�쭷,��٫��ˁ�A��|�
�kAn �(X�������X��+��|�5~A�
�ɪP�	���X�b���
*--����
��m���8�-X��#7f�j�`V"���-�"��L�cbv#�%�ɻf8�~��r�-��宩�VF-�dN8|��t�,��B�-���'�X��-?ׯ�,	�B�-����B�X����tuH*��z�EX��kt�R�+�*٫��ˁ�A��|�
�����(X�������X��+���|�5~A�
�ɪP�	���X�b���
Dr��P� �����-
��-X���#7f�v�ebf����R
�b�cbv!�%��d8�~t�Ӭ���o���be�jN8깪��,	�t�,��t�X���-�ׯP��'p�,P�pt�X����Ѫ��\���AH�"�ɪ�R�	����a�ˢBA,��*����*���(X������X��+��|���*���ɪ�P�+��𪭹�,���*r��Pۨ���������t�/�6�dv�bj$"���-�"��L�cbv#�%�ɹf8���m�-���宩�VF-�dN8|��t�,��B�-9��t�X���-?ׯP��'p�,P�pt�X����ф��\���(z�l �(X�+t�챩�X����+��|�
��* �ɪ�T����-
H+���|�5�*"�ɪ�P�+���aŹ����*-������
��m��8�-X��D'�n%�Ffp���R��b�cbv#�%;�f8���m�-���宩�VF-�d���'p�,P��t�X���'p�,P��PB�X������,	�'p�,P��+v�m\���P�
���(X�+t����m���+��|�
�kAn �(X�������X��+��|���*���ɪ�P�+����,���*-�����
��m���8�-X��#7f�j�`V"���-�"��L�cbv!�%��d8�~��r����o���be�v%C|�t�X���'�X��B�-9�ȃ'p�,P��t�X����B�-9���jX#@)��(z�l �(X�+t�챩�X����+��|�
�kAn
�ɪT���al �A.1�*���*�����P�+����X�b����������� ��
```

aboyuanna, 2023-11-28:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13482885/sdkconfig.txt)

zikalino, 2023-11-28:
这个问题可以始终如一地重现，还是会不时发生？
我认为它可能与软件无关。您能分享一些关于硬件设计的更多细节吗？与此同时，我们将在内部讨论这个问题，并尝试找出可能的解决方案。

Dazza0, 2023-11-29:
Hi @aboyuanna, 请用逻辑分析仪看看乱码。有可能是普通的 LOG，但是在不同的波特率输出。

> 这个问题在我们的电路板上才能复现

能分享以下电路吗？会不会 ESP32 供电遇到过电流或欠电压的情况？

aboyuanna, 2023-11-29:
@zikalino @Dazza0 这个问题在切换继电器的时候就有概率出现。
[相关资料.zip](https://github.com/espressif/esp-idf/files/13498545/default.zip)
zip文件里有能够复现这个现象的代码、原理图。当log出现乱码时，芯片是完全没有反应的，只会不停输出乱码。我们现在关注的问题是ESP32在这种情况下为什么不自动重启？

Dazza0, 2023-11-29:
> 当log出现乱码时，芯片是完全没有反应的

理论上，如果 ESP32 还在输出 log，就应该还在跑的 （表示 ESP32 内部时钟和  UART 外设还在跑）。所以才想用逻辑分析一看看这些乱码是什么。

一个简单的方法来测试 ESP32 是否停掉了是开个 task （或者用 [Idle hook](https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32/api-reference/system/freertos_additions.html#esp-idf-tick-and-idle-hooks)) 定时将一个 GPIO 拉高/低 。

aboyuanna, 2023-11-29:
> 理论上，如果 ESP32 还在输出 log，就应该还在跑的 （表示 ESP32 内部时钟和 UART 外设还在跑）。所以才想用逻辑分析一看看这些乱码是什么。
>
> 一个简单的方法来测试 ESP32 是否停掉了是开个 task （或者用 [Idle [...]

你可以看一下源码，我创建了一个esp timer每隔一秒切换继电器。当log输出乱码时，继电器就不会切换了，而且这时按键也没反应。稍后我会用逻辑仪看看乱码是什么内容。

ESP-Marius, 2024-07-08:
Sorry for the slow replies, is there still any outstanding issues here?

closed 2024-07-29 by aboyuanna

PING020903, 2024-08-08:
不能通过增加金属屏蔽层避免吗