espressif/esp-idf #12681: Calling rtc_gpio_deinit() for any GPIO will disable wake on GPIO for diffeent GPIOs

xyzzy42, 2023-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-582-gab03c2ea13

### Espressif SoC revision.

ESP32-S3 v0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 DevkitC

### Power Supply used.

USB

### What is the expected behavior?

After correctly enabling wake on GPIO, calling `rtc_gpio_deinit()` on a **completely different GPIO** will not disable wake on GPIO.

### What is the actual behavior?

`rtc_gpio_deinit()` only disable the supplied GPIO from RTC mode and does not affect other functions of other GPIOs.

### Steps to reproduce.

1. Modify `examples/system/light_sleep/main/gpio_wakeup.c` to add `rtc_gpio_deinit(13)` to end of `example_register_gpio_wakeup()`.  Where `13` is unused GPIO not related to wake on boot button used in test.
2. Wake on gpio example no longer works.

### More Information.

The problem is in `rtcio_ll_function_select()`:
```c
    } else if (func == RTCIO_LL_FUNC_DIGITAL) {
        CLEAR_PERI_REG_MASK(rtc_io_desc[rtcio_num].reg, (rtc_io_desc[rtcio_num].mux));
        SENS.sar_peri_clk_gate_conf.iomux_clk_en = 0;
        // USB Serial JTAG pad re-enable won't be done here (it requires both DM and DP pins not in rtc function)
        // Instead, USB_SERIAL_JTAG_USB_PAD_ENABLE needs to be guaranteed to be set in usb_serial_jtag driver
    }
```

If any GPIO is changed to digital function, then iomux_clk_en is set to 0.  This bit is not documented much in ESP32-S3 docs, but it appears that once it is disabled wake on GPIO no longer works.  Perhaps it turns off clock to iomux when in lightsleep?  

It is turned on by call to `rtcio_ll_wakeup_enable()`.

Perhaps it should only be turned off when no GPIOs are in RTC function or wakeup function.

I suspect that there is a related bug if two GPIOs are set to `RTCIO_LL_FUNC_RTC`, then only one is changed to `RTCIO_LL_FUNC_DIGITAL` but the other is not.  `iomux_clk_en` will be turned off even through the other GPIO is still in RTC function mode.

Another related bug is `rtcio_ll_wakeup_disable()` does not disable `iomux_clk_en`.  So enabling wakeup and then disabling wakeup will leave the iomux clock on despite no remaining users.

pedrogsfaria, 2025-02-19:
I am facing a similar problem, any updates on how to solve it?

esp-wzh, 2025-02-20:
Thanks for the report, the fix MR has been merged internally and will be synced to github soon.

esp-wzh, 2025-02-25:
Fixed by https://github.com/espressif/esp-idf/commit/79ee0927d1f2e1e137b0eb55e260ec365af12315.

Alvin1Zhang, 2025-02-25:
Thanks for reporting and sorry for slow turnaround, feel free to reopen.

closed 2025-02-25 by Alvin1Zhang