espressif/esp-idf #12689: [v5.1.2][AES][ESP32-s3] mbedtls_aes_crypt_cbc returning nondeterministic results despite same input data

chipweinberger, 2023-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-s3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

Given the same input data & iv, the encrypted data should be the same

### What is the actual behavior?

The output data is not always the same

### Steps to reproduce.

1. I'm using AES decrypt during my OTA process, over wifi
2. I'm using `CONFIG_MBEDTLS_AES_USE_INTERRUPT=n`
3. [sdkconfig.txt](https://github.com/espressif/esp-idf/files/13496681/sdkconfig.txt)

### Debug Logs.

```plain
// run #1
I (163944) pd ota session: checksumD 792277 // iv
I (163944) pd ota session: checksumE 1612320 // input len
I (163944) pd ota session: checksumF 205499032 // input bytes
I (163954) pd ota session: checksumG 140200246 // output

// run #2
I (84744) pd ota session: checksumD 792277 // iv
I (84744) pd ota session: checksumE 1612320 // input len
I (84744) pd ota session: checksumF 205499032 // input bytes
I (84744) pd ota session: checksumG 13996798 // output
```

### More Information.

I've made simple checksums of the input data & output data. 

The input data always has the same checksum, but the output does not!

```
uint64_t checksumD = 0;
uint64_t checksumE = 0;
uint64_t checksumF = 0;
uint64_t checksumG = 0;

static pd_buf_t decryptAesCbc(pd_ota_session_t* ctx,
                                 const pd_buf_t input, 
                                    pd_error_t* error)
{
    pd_err_unset(error);

    pd_buf_t decrypted = pd_buf_empty();
    {
            decrypted = pd_buf_fill_zeros(input.len);

            // checksumD
            for(int i = 0; i < 16; i++) {
                checksumD += ctx->aesIv[i];
            }

            // checksumE
            checksumE += input.len;

            // checksumF
            for(int i = 0; i < input.len; i++) {
                checksumF += input.bytes[i];
            }

            // decryptAesCbc
            int ret = 0;
            mbTry(mbedtls_aes_crypt_cbc,(&ctx->aesCtx, MBEDTLS_AES_DECRYPT,
                input.len, ctx->aesIv, input.bytes, decrypted.bytes));

            // checksumG
            for(int i = 0; i < decrypted.len; i++) {
                checksumG += decrypted.bytes[i];
            }
        }

        pd_err_success(error);
    }
end:
    eOnFail(pd_buf_delete(&decrypted));
    eOnFail(decrypted = pd_buf_empty());
    return decrypted;
}
```

chipweinberger, 2023-11-29:
@mahavirj

chipweinberger, 2023-11-29:
This is what ChatGPT said:

If mbedtls_aes_crypt_cbc is returning nondeterministic results despite being given the same input data, it's important to carefully check several aspects of your usage of the function to identify the cause. Here are some key factors to consider:

Initialization Vector (IV):

* The IV for CBC mode must be unique and should not be reused with the same key. However, for the same output, the same IV must be used for encryption and decryption.
Ensure that the IV is properly initialized before each encryption or decryption operation. If the IV is not set or is set incorrectly (e.g., uninitialized memory), it can lead to nondeterministic results.

Key Initialization:

* Make sure the key is correctly initialized and is the same for every operation.
Check if the key size matches the expected size for the AES variant you are using (AES-128, AES-192, AES-256).
Input Data:

*  that the input data (including its length) is the same for each operation.
Remember that CBC mode requires the input size to be a multiple of the block size (16 bytes for AES). If your data does not meet this requirement, you need to apply padding.
Context Reuse:

* u are reusing the mbedtls_aes_context, ensure it is properly reset or reinitialized before each operation. Context reuse without proper reinitialization can lead to unexpected results.
Memory Issues:

* or any memory corruption issues or unintended modifications of the input data, key, or IV between operations.
Use tools like Valgrind to detect memory issues that could be affecting your program's behavior.
Library Version and Configuration:

* Ensure you are using a compatible and stable version of the Mbed TLS library.
Check if the library is configured correctly for your needs. Mbed TLS can be fine-tuned with various compile-time options, which might affect its behavior.
Error Checking:

* Ensure that you check the return value of mbedtls_aes_crypt_cbc for errors after each call.
Investigate any errors returned by the function to understand what might be going wrong.
Thread Safety:

* If your application is multi-threaded, ensure that there are no race conditions affecting the shared resources like keys, IVs, or context.

Environmental Factors:

* Consider if there are any environmental factors that could be affecting the operation, such as changes in the system's state or external modifications of data.

By methodically checking each of these areas, you should be able to pinpoint the cause of the nondeterministic behavior. In cryptographic operations, small changes or errors in the setup can lead to large differences in output, so attention to detail is crucial

chipweinberger, 2023-11-29:
it looks like in v5.0 theres a new way to set the decryption key. ill try that

```
int mbedtls_aes_setkey_dec(mbedtls_aes_context *ctx, const unsigned char *key,
                           unsigned int keybits)
```

chipweinberger, 2023-11-29:
note if i set `CONFIG_MBEDTLS_AES_USE_INTERRUPT=y` , i get

```
E (38001) esp-aes: Timed out waiting for completion of AES Interrupt
E (38001) esp-aes: esp_aes_dma_wait_complete failed
```

chipweinberger, 2023-11-29:
It is fixed by setting `CONFIG_MBEDTLS_HARDWARE_AES=n`!!!!

chipweinberger, 2023-11-29:
I found another bug in my code. I was using `esp_aes_context` instead of `mbedtls_aes_context`. **But it does not fix anything.**

chipweinberger, 2023-11-29:
It is fixed by setting `CONFIG_MBEDTLS_HARDWARE_AES=n` !!!!

Harshal5, 2023-11-29:
Hello @chipweinberger,

If possible could you share the values of the input and iv that you are experimenting with?  I tried out some AES-CBC calculations with some random values but they seem to work fine. We even have an AES-CBC [test case](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/test_apps/main/test_aes.c#L72-L117) in ESP-IDF and that seems to work well too.

chipweinberger, 2023-11-29:
@Harshal5 my code is quite large and complex. I wish I had the time to debug it further.

My guess is that it might be related to wifi & ota

my ota binary is 1.5MB and I transfer it in chunks of ~16KB where each chunk is sent in a separate http POST request.

I also have bluetooth enabled, and am using PSRAM. I can't think of anything else to say.

Harshal5, 2023-11-29:
@chipweinberger, could you try executing the test [mbedtls AES performance](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/test_apps/main/test_aes_perf.c#L20-L75) with the sdkconfig that you have provided  above? (below are the steps for the same) I tried it on my side and the encryption seems to work well.

Steps to execute the mbedtls test:
1. `cd components/mbedtls/test_apps`
2. Copy the sdkconfig file that is linked above.
3. Run `idf.py build flash monitor`
4. Enter the number for the test "mbedtls AES performance" (26 maybe)

chipweinberger, 2023-11-29:
works.

note: I use a watchdog timeout in my sdkconfig. it's unrelated of course.

```
Running mbedtls AES self-tests...
  AES note: built-in implementation.
  AES-ECB-128 (dec): passed
  AES-ECB-128 (enc): passed
  AES-ECB-192 (dec): passed
  AES-ECB-192 (enc): passed
  AES-ECB-256 (dec): passed
  AES-ECB-256 (enc): passed

  AES-CBC-128 (dec): passed
  AES-CBC-128 (enc): passed
  AES-CBC-192 (dec): passed
  AES-CBC-192 (enc): passed
  AES-CBC-256 (dec): passed
  AES-CBC-256 (enc): passed

  AES-CFB128-128 (dec): passed
  AES-CFB128-128 (enc): passed
  AES-CFB128-192 (dec): passed
  AES-CFB128-192 (enc): passed
  AES-CFB128-256 (dec): passed
  AES-CFB128-256 (enc): passed

  AES-OFB-128 (dec): passed
  AES-OFB-128 (enc): passed
  AES-OFB-192 (dec): passed
  AES-OFB-192 (enc): passed
  AES-OFB-256 (dec): passed
  AES-OFB-256 (enc): passed

  AES-CTR-128 (dec): passed
  AES-CTR-128 (enc): passed
  AES-CTR-128 (dec): passed
  AES-CTR-128 (enc): passed
  AES-CTR-128 (dec): passed
  AES-CTR-128 (enc): passed

  AES-XTS-128 (dec): passed
  AES-XTS-128 (enc): passed
  AES-XTS-128 (dec): passed
  AES-XTS-128 (enc): passed
  AES-XTS-128 (dec): passed
  AES-XTS-128 (enc): passed

Verified no APB corruption from operations
MALLOC_CAP_8BIT usage: Free memory delta: 0 Leak threshold: -1024 
MALLOC_CAP_32BIT usage: Free memory delta: 0 Leak threshold: -1024 
/IDF/components/mbedtls/test_apps/main/test_mbedtls.c:31:mbedtls AES self-tests:PASS
Test ran in 1151ms

-----------------------
1 Tests 0 Failures 0 Ignored 
OK
Enter next test, or 'enter' to see menu
E (8089) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (8089) task_wdt:  - IDLE (CPU 0)
E (8089) task_wdt: Tasks currently running:
E (8089) task_wdt: CPU 0: testTask
E (8089) task_wdt: CPU 1: IDLE
E (8089) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x42017CCB:0x3FC97920 0x403777C9:0x3FC97960 0x40048D03:0x3FCB0C20 |<-CORRUPTED
0x42017ccb: pd_xtrans_resp_set_hdr(pd_xtrans_t*, char const*, char const*, pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_xtrans.cpp:649

0x403777c9: rtc_isr_noniram_enable at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_hw_support/rtc_module.c:202

0x40048d03: uart_rx_one_char in ROM
```

Harshal5, 2023-11-30:
The test passes so I am not sure if there could be a `sdkconfig` migration issue in this case. Would it be possible for you to share a minimal example or any more information for me to reproduce the issue?

chipweinberger, 2023-11-30:
the test is probably not stressful enough imo. testing PSRAM would be the first thing to check imo.

But unfortunately I can'y justify the time to create a minimal example. I plan to use software AES regardless because I think it would more reliable.

Harshal5, 2024-05-07:
Closing the issue for now as the issue is not reproducible at our end, please feel free to re-open the issue as needed.

closed 2024-05-07 by Harshal5