espressif/esp-idf #12691: Cannot connect ESP32-S3 WROOM Bluooth to BLE

shlomo-edu, 2023-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-S3 WROOM

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Visual Studio Code

### Power Supply used.

USB

### What is the expected behavior?

I managed to compile and run: ble  gatt_server example
Now I am trying to pair it from Windows 11.
I see the ESP_GATTS_DEMO in the list and I click on it.
It tries to connect: Connecting....
And I see in the Terminal that it tries to connect, and then it says: 
I (305799) GATTS_DEMO: update connection params status = 16, min_int = 16, max_int = 32,conn_int = 30,latency = 0, 
timeout = 400
E (333819) BT_BTM: Device not found
And then at the end (after retrying)
W (1168119) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
I (1168119) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13

### What is the actual behavior?

I want to see that it manage to connect.

### Steps to reproduce.

1. Run the example on the device (ESP32-S3 WROOM from FREENOVE)
2. In Windows Settings (add bluettoth device window), click on the :  ESP_GATTS_DEMO

...

### Debug Logs.

```plain
I (303789) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote d8:f8:83:db:c1:2f:
I (303789) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote d8:f8:83:db:c1:2f:
I (303789) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (303799) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (305799) GATTS_DEMO: update connection params status = 16, min_int = 16, max_int = 32,conn_int = 30,latency = 0, 
timeout = 400
E (333819) BT_BTM: Device not found

W (333819) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
I (333819) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (1138649) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote d8:f8:83:db:c1:2f:
I (1138649) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote d8:f8:83:db:c1:2f:
I (1138649) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (1138659) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (1140669) GATTS_DEMO: update connection params status = 16, min_int = 16, max_int = 32,conn_int = 30,latency = 0, timeout = 400
E (1168119) BT_BTM: Device not found

W (1168119) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
I (1168119) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
```

zhp0406, 2023-11-30:
@shlomo-edu 
thanks for report the problem.I will check it.

shlomo-edu, 2023-11-30:
I want to add something to this problem.
I know that BLE doesn't need to pair in order to send/receive messages.
So I tried sending messages from my iPhone by using BluetoothLE app.
It works OK, I can send a message, and then receive something wich I still need to understand what it is.
So this means that it connect to iPhone (and also to Android cell phone).
I also tried to change the: ESP_BT_MODE_BLE to ESP_BT_MODE_BTDM (because I thought that this will enable me to pair in Windows), but in this case it fails in the initialization phase:
E (449) BT_INIT: invalid mode 3, controller support mode is 1
E (459) GATTS_DEMO: app_main enable controller failed: ESP_ERR_INVALID_ARG
And the server is not in the list of bluetooth devices (in Windows).
Same thing when I tried: ESP_BT_MODE_CLASSIC_BT

zhp0406, 2023-11-30:
@shlomo-edu 
I reproduced the issue on my end. On the Windows side, the connection was initiated, but the ESP32 did not respond to the pairing request within 30 seconds, leading to an **ATT timeout** and **disconnection**. I will work on fixing this problem.
[image]

zhp0406, 2023-11-30:
@shlomo-edu 
Classic Bluetooth is only supported by ESP32; other chips do not support it. If you have set ESP_BT_MODE_BTDM on other chips, you may encounter the following errors:

E (411) BLE_INIT: invalid mode 3, controller support mode is 1
E (411) GATTS_DEMO: app_main enable controller failed: ESP_ERR_INVALID_ARG
I (421) main_task: Returned from app_main()

shlomo-edu, 2023-11-30:
Does this means that PCs with only Bluetooth (without BLE) cannot connect to my ESP chip?

zhp0406, 2023-11-30:
@shlomo-edu 
you are right

shlomo-edu, 2023-11-30:
What about the paring from Windows?
The truth is that I don't need it in my application, but if you will have a solution for it than I woul like to have this option in the future.

zhp0406, 2023-11-30:
According to my observation here, Windows initiates pairing proactively, regardless of whether you need it or not. I'm not sure why Windows does this. If the ESP32 does not respond to Windows' pairing request, the connection will be terminated by Windows after 30 seconds.

zhp0406, 2023-11-30:
What do you need to do with the ESP32-S3? Could you please specify your requirements?

shlomo-edu, 2023-11-30:
We have a BLE logger that we connect to from our application, and we don't do pairing.
When I tried to do pairing from Windows to my ESP32-S3 WROOM, the client displayed the error message:
I (1168119) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I don't know what is 0x13, but I think it tells you why the client didn't responed with in 30 seconds.

zhp0406, 2023-12-01:
So, if ESP32-S3 is serving as a server and needs to connect with Windows, it must enable SMP (Secure Simple Pairing) functionality. MacOS, Android, iOS, and similar platforms do not initiate pairing requests actively, so they can be used directly.

If you intend to use ESP32-S3 with Windows, SMP functionality must be enabled. Only by enabling SMP can ESP32-S3 respond to pairing requests from Windows.

You can refer to the example "esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server." This example enables SMP functionality.
Then, based on that example, add your own services.

shlomo-edu, 2023-12-03:
I received a message from: Halley826
That I should write:
`rsp.attr_value.handle = param->read.handle;`
But this is not what I want to do.
I want to add a long message (longer than 4 bytes) and to send it not as a response to a read event.
This (send a respond of 4 bytes) I can do now.

Halley826, 2023-12-03:
esp_ble_gatts_send_indicate

shlomo-edu, 2023-12-03:
What do you mean?

shlomo-edu, 2023-12-03:
What do you mean?

zhp0406, 2023-12-04:
The client needs to register for notifications from the server using the API esp_ble_gattc_register_for_notify.

If notifications are registered but the client does not receive notifications or indications from the server,

the ESP_GATTC_REG_FOR_NOTIFY_EVT event will be triggered. In this event, it is necessary to write
the server's characteristic descriptor using the API esp_ble_gattc_write_char_descr.

Once the write is complete, the server will raise the ESP_GATTS_WRITE_EVT event.
In the ESP_GATTS_WRITE_EVT event, check whether the characteristic descriptor is being written.
If so, send esp_ble_gatts_send_indicate.

Then the client can receive notifications from the server in the ESP_GATTC_NOTIFY_EVT event.

zhp0406, 2023-12-04:
```
/**
 * @brief           Send indicate or notify to GATT client.
 *                  Set param need_confirm as false will send notification, otherwise indication.
 *                  Note: the size of indicate or notify data need less than MTU size,see "esp_ble_gattc_send_mtu_req".
 *
 * @param[in]       gatts_if: GATT server access interface
 * @param[in]       conn_id - connection id to indicate.
 * @param[in]       attr_handle - attribute handle to indicate.
 * @param[in]       value_len - indicate value length.
 * @param[in]       value: value to indicate.
 * @param[in]       need_confirm - Whether a confirmation is required.
 *                  false sends a GATT notification, true sends a GATT indication.
 *
 * @return
 *                  - ESP_OK : success
 *                  - other  : failed
 *
 */
esp_err_t esp_ble_gatts_send_indicate(esp_gatt_if_t gatts_if, uint16_t conn_id, uint16_t attr_handle,
                                      uint16_t value_len, uint8_t *value, bool need_confirm);
```

shlomo-edu, 2023-12-04:
@zhp0406 
I see that all the examples that use `esp_ble_gattc_register_for_notify` are clients examples.
Can I use it in the server?

shlomo-edu, 2023-12-04:
So I don't understand.
The senario is this:
I want to build a bridge from our logger to a PC.
Now I can communicate with the logger.
And I can receive and reply messages from the PC via the BLE.

I don't understand how do I send the messages that I recieve from the logger to the PC.
Please show me the code how to do it in the gatt_server.c

shlomo-edu, 2023-12-04:
OK, now it works for me.
I just do this:
`    esp_ble_gatts_send_indicate(last_gatts_if, 0, gl_profile_tab[PROFILE_A_APP_ID].char_handle,
        strlen(notify_data), (uint8_t *) notify_data, false);
 `
Thank you very much.

closed 2023-12-04 by shlomo-edu

profill, 2024-09-26:
我也遇到了同样得问题我的app需要在电脑或者手机上与ESP32-S3使用BLE互通数据但是能发现却无法连接I (1137736) GATTS_TABLE_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (1138256) GATTS_TABLE_DEMO: 更新连接参数状态 
= 0, 最小间隔 = 16, 最大间隔 = 48, 当前间隔 = 42, 延迟 = 0, 超时 = 400
E (1167706) BT_BTM: Device not found

W (1167706) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
I (1167706) GATTS_TABLE_DEMO: ESP_GATTS_DISCONNECT_EVT, reason = 0x13
I (1167716) GATTS_TABLE_DEMO: 广播启动成功

esp-zhp, 2024-09-27:
@profill 
esp-idf/examples/bluetooth/bluedroid/ble/gatt_server_service_table
这个 example 是没有 SMP 功能的，这样会更加节省资源。

如果要和 windows 平台使用，你需要使用另外一个 example（esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server），该 example 支持打开 SMP，并设置好了相关的参数。