espressif/esp-idf #12694: Help with ESP32C3 built in JTAG Debugging

VeloceEmbedded, 2023-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

To be able to use JTAG Debugging for VS Code IDE

### What is the actual behavior?

I am not able to use JTAG debugging due to remote connection not being set

### Steps to reproduce.

1. Open Terminal on VS Code and enter "openocd -f board/esp32c3-builtin.cfg"
2. then click on Debug 
[image]

3. This shows me an error
[image]

I am not able to understand what file is It asking for because the miDebuggerPaths exists
[image]

...

### Build or installation Logs.

```plain
When I open Terminal on my VS Code, I see these logs:

Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.39.2\cmd\
git version 2.39.2.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.1.1

Adding ESP-IDF tools to PATH...
    C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.24.0\bin
    C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    C:\Espressif\frameworks\esp-idf-v5.1.1\tools

Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.1.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.1.1\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

D:\My-Project\ESP32-Wroom-C3>

Now as far as I understand this says that ESPIDF 5.1.1 was configured properly. 

Then I do, 
D:\My-Project\ESP32-Wroom-C3>openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (58:CF:79:D7:14:B4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)    
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Memory protection is enabled. Reset target to disable it...
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Error: Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Error: Failed to get flash size!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected

MY tasks.json File is:
{
    "tasks": [
        {
            "type": "cppbuild",
            "label": "C/C++: gcc.exe build active file",
            "command": "C:\\MinGW\\bin\\gcc.exe",
            "args": [
                "-fdiagnostics-color=always",
                "-g",
                "${file}",
                "-o",
                "${fileDirname}\\${fileBasenameNoExtension}.exe"
            ],
            "options": {
                "cwd": "${fileDirname}"
            },
            "problemMatcher": [
                "$gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            },
            "detail": "Task generated by Debugger."
        }
    ],
    "version": "2.0.0"
}

and My launch.json file is
{
    "version": "0.2.0",
    "configurations": [
      {
        "name": "ESP32 OpenOCD",
        "type": "cppdbg",
        "request": "launch",
        "cwd": "${workspaceFolder}/build",
        "program": "${workspaceFolder}/build/MyTracker.elf",
        "miDebuggerPath": "C:/Espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin/riscv32-esp-elf-gdb.exe",
        "setupCommands": [
            {"text": "set remote hardware-watchpoint-limit 2"}, 
            {"text": "target remote 127.0.0.1:3333"},
            {"text": "monitor reset halt"},
            {"text": "flushregs"},
            {"text": "thb app_main"}
        ]
      }
    ]
  }

The file paths are correct.
```

erhankur, 2023-11-30:
@VeloceEmbedded which application did you load? Can you load hello-world example and try again?

meakashrao, 2023-11-30:
I also tried modifying my tasks.json

{
    "tasks": [
        {
            "type": "cppbuild",
            "label": "C/C++: gcc.exe build active file",
            "command": "C:\\MinGW\\bin\\gcc.exe",
            "args": [
                "-fdiagnostics-color=always",
                "-g",
                "${file}",
                "-o",
                "${fileDirname}\\${fileBasenameNoExtension}.exe"
            ],
            "options": {
                "cwd": "${fileDirname}"
            },
            "problemMatcher": [
                "$gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            },
            "detail": "Task generated by Debugger."
        },
        {
            "label": "OpenOCD (ESP32-C3)",
            "type": "shell",
            "command": "C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/openocd.exe",
            "args": [
                "-f",
                "board/esp32c3-builtin.cfg"
            ],
            "isBackground": true,
            "problemMatcher": {
                "owner": "cpp",
                "pattern": {
                    "regexp": ".*",
                    "file": 1,
                    "location": 2,
                    "message": 3
                },
                "background": {
                    "activeOnStart": true,
                    "beginsPattern": "^Info : [^ ]+ server started at",
                    "endsPattern": "^Info : [^ ]+ server stopped"
                }
            },
            "presentation": {
                "reveal": "always",
                "focus": false,
                "panel": "shared"
            }
        }
    ],
    "version": "2.0.0"
}

Still no luck, please some help to get it configured

VeloceEmbedded, 2023-11-30:
I even configured my tasks.json

{
    "tasks": [
        {
            "type": "cppbuild",
            "label": "C/C++: gcc.exe build active file",
            "command": "C:\\MinGW\\bin\\gcc.exe",
            "args": [
                "-fdiagnostics-color=always",
                "-g",
                "${file}",
                "-o",
                "${fileDirname}\\${fileBasenameNoExtension}.exe"
            ],
            "options": {
                "cwd": "${fileDirname}"
            },
            "problemMatcher": [
                "$gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            },
            "detail": "Task generated by Debugger."
        },
        {
            "label": "OpenOCD (ESP32-C3)",
            "type": "shell",
            "command": "C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/openocd.exe",
            "args": [
                "-f",
                "board/esp32c3-builtin.cfg"
            ],
            "isBackground": true,
            "problemMatcher": {
                "owner": "cpp",
                "pattern": {
                    "regexp": ".*",
                    "file": 1,
                    "location": 2,
                    "message": 3
                },
                "background": {
                    "activeOnStart": true,
                    "beginsPattern": "^Info : [^ ]+ server started at",
                    "endsPattern": "^Info : [^ ]+ server stopped"
                }
            },
            "presentation": {
                "reveal": "always",
                "focus": false,
                "panel": "shared"
            }
        }
    ],
    "version": "2.0.0"
}

Still no luck, please help

erhankur, 2023-11-30:
Did you check here?

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md

I don't see a breakpoint in your launch.json

VeloceEmbedded, 2023-11-30:
added that afterwards, still the same

erhankur, 2025-03-08:
closed due to inactivity

closed 2025-03-08 by erhankur