espressif/esp-idf #12696: 希望可以优化编程指南的描述

caidamao, 2023-11-30:
### Is your feature request related to a problem?

FAQ:https://espressif-docs.readthedocs-hosted.com/projects/esp-faq/zh-cn/latest/software-framework/storage/nvs.html

closed 2023-11-30 by caidamao

Alvin1Zhang, 2023-12-05:
Thanks for reporting, the updated version of the FAQ you mentioned is available at https://docs.espressif.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/storage/nvs.html#nvs-api, feel free to reopen.

caidamao, 2023-12-06:
> Thanks for reporting, the updated version of the FAQ you mentioned is available at https://docs.espressif.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/storage/nvs.html#nvs-api, feel free to reopen.

thanks！