espressif/esp-idf #12699: nvs_entry_find_in_handle present in main but not releases

IoTisan, 2023-11-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

nvs_entry_find_in_handle is present in `esp-idf/components/nvs_flash/include/nvs.h` in main branch, but not in any releases.

It's listed in the documentation, but is removed from this file in releases.

Is there a reason for this?

igrr, 2023-11-30:
In ESP-IDF, new features are developed on the `master` branch. For example, currently development of IDF v5.3 is on-going on `master`. When we get closer to the v5.3 release, we will create a branch release/v5.3 from `master` and the new features will be available there. Older release branches have been branched off `master` at earlier points in time, so as you understand, newer features are not available  there.

The feature in question was added in https://github.com/espressif/esp-idf/commit/c19203be0eea and will be part of the future v5.3 release.

IoTisan, 2023-11-30:
Thanks for the explanation. I look forward to the upcoming release.

closed 2023-11-30 by IoTisan