espressif/esp-idf #12700: esp_hw_support/port/esp32/CMakeLists.txt specifies missing include directory

Hairy-Lee, 2023-11-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

`./v5.1/components/esp_hw_support/port/esp32/private_include` no longer exists (it was in v4.4 along with associated *.h). 

However, it is still specified in `./v5.1/components/esp_hw_support/port/esp32/CMakeLists.txt` on line 2:
`target_include_directories(${COMPONENT_LIB} PUBLIC private_include)`

When creating a JSON Compliation Database this non-existant folder subsequently gets added to compile_commands.json

0xjakob, 2023-12-01:
@Hairy-Lee Thanks for letting us know, we'll prepare a fix for this.

0xjakob, 2023-12-04:
The fix is prepared and merged, it should appear on the `master` branch soon.

closed 2023-12-06 by espressif-bot