espressif/esp-idf #12705: Can I turn off assert warnings in the static library?

jjj19950710, 2023-12-01:
### Is your feature request related to a problem?

I don't want information like "ASSERT-WARN (1 8), in lc_task. c" to be printed.This print source code is in the static library libbtdm_app.a.Can I turn off it.

jjj19950710, 2023-12-12:
I can't calling the API esp_bt_controller_set_log_level.It doesn't seem to be defined.
```
 error: implicit declaration of function 'esp_bt_controller_set_log_level'; did you mean 'esp_bt_controller_mem_release'? [-Werror=implicit-function-declaration]
```

ESP-YTGerd, 2023-12-12:
Hi, @jjj19950710 
Did you extern it before call the `esp_bt_controller_set_log_level`?

jjj19950710, 2023-12-13:
now I can close those WARNING logs.thanks!

Indastri, 2024-01-09:
Seems that this issue was closed. If the issue still persists feel free to reopen this ticket or create a new one. Cheers.

closed 2024-01-09 by Indastri