espressif/esp-idf #12706: ESP32-C3 documentation has no "SD Pull-up Requirements" page

llamasoft, 2023-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The ESP32-C3 documentation has no "SD Pull-up Requirements" page.

The page exists for [ESP32](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html), [ESP32-S3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/sd_pullup_requirements.html), etc, but not for ESP32-C3.

I'm not sure if this omission was intentional or not.  I understand that the ESP32-C3 doesn't support SDMMC but the pull-up requirements still apply to SPI mode as well.

igrr, 2023-12-02:
It is true that there are pull-up requirements for  SD cards in SPI mode, but as it is, the page only covers the SD mode. (With the exception of one paragraph which is also relevant to SPI.) To avoid confusion, it is only being shown in the programming guide of chips which do have the SD Host peripheral.

If you'd like to suggest the content which should be displayed for chips with only the SPI peripheral, you are most welcome to do so!

llamasoft, 2023-12-06:
That's understandable.
If the note on pull-ups can be moved anywhere, it might make sense to move it to the [SD SPI Host Driver](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/sdspi_host.html) and/or [SD/SDIO/MMC Driver](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/storage/sdmmc.html) pages as that's where it's most likely to be useful.