espressif/esp-idf #12708: assert failed: block_locate_free tlsf.c:566 (block_size(block) >= size)

hamza2657, 2023-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.1

### Espressif SoC revision.

ESP32-WROOM DEVKitC-V4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP-IDF

### Power Supply used.

USB

### What is the expected behavior?

I'm trying to use BLE and Wifi driver simultaneously. and turn on each when needed and deinitialize when done.
  

### What is the actual behavior?

when i try to deinit wifi it is not properly deinitlizing the wifi stack and when i try to initialize it again heap memory corruption error log appear on screen and the esp reboot.

### Steps to reproduce.

when i initialize the wifi it gives logs that confim that wifi is properly initialized and when i deinitialize it and then try to initialize it it gives assert failed error "assert failed: block_locate_free tlsf.c:566 (block_size(block) >= size)"
so when i checked the log during the deinit phase i observe that the drive is not proper deinitializing the rx tx static and dynamic buffer so in the next run when it try to initialize those buffers which are already present in the heap so the error occures. and due to panic attack it restarts the esp32.

### Debug Logs.

```plain
Init phase code:
one time init. of netif event and callbacks:
code:
void one_time_wifi_initliazation(void)
{
//Initialize NVS
	esp_err_t ret = nvs_flash_init();
	if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
	  ESP_ERROR_CHECK(nvs_flash_erase());
	  ret = nvs_flash_init();
	}
	ESP_ERROR_CHECK(esp_event_loop_create_default());
	esp_event_handler_instance_t instance_any_id;
	esp_event_handler_instance_t instance_got_ip;
	ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler,NULL, &instance_any_id));
	ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,IP_EVENT_STA_GOT_IP,&event_handler,NULL,&instance_got_ip));
	
} 
and this function will use repeatedly to init. the wifi on a single life of the program:
void wifi_init_sta(void)
{
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

	     wifi_config_t wifi_config = {
	        .sta = {
	            .ssid = "FreeWifi",
	            .password = "12345678",
	        },
	    };
	    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
	    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
	    ESP_ERROR_CHECK(esp_wifi_start());

	    ESP_LOGI(TAG, "wifi_init_sta finished.");
    rgb_led_wifi_app_started();
}

Debug Logs on first try which is quite a success:
I (9141) wifi:Init data frame dynamic rx buffer num: 32
I (9141) wifi:Init static rx mgmt buffer num: 5
I (9151) wifi:Init management short buffer num: 32
I (9151) wifi:Init static tx buffer num: 16
I (9151) wifi:Init static rx buffer size: 1600
I (9161) wifi:Init static rx buffer num: 10
I (9161) wifi:Init dynamic rx buffer num: 32
I (9171) wifi_init: rx ba win: 6
I (9171) wifi_init: tcpip mbox: 32
I (9171) wifi_init: udp mbox: 6
I (9181) wifi_init: tcp mbox: 6
I (9181) wifi_init: tcp tx win: 5744
I (9191) wifi_init: tcp rx win: 5744
I (9191) wifi_init: tcp mss: 1440
I (9191) wifi_init: WiFi IRAM OP enabled
I (9201) wifi_init: WiFi RX IRAM OP enabled

debug logs during deinit phase which is i guess wrong it is not properly deinit the driver:
Wifi disconnected successfully
E (19831) wifi:NAN WiFi stop
I (19831) wifi:flush txq
I (19831) wifi:stop sw txq
I (19831) wifi:lmac stop hw txq
I (19831) wifi:Deinit lldesc rx mblock:10

and after that ble start and then when i switch 2nd time
i got these logs :
I (27761) wifi:wifi driver task: 3ffb5b28, prio:23, stack:6656, core=0
I (27761) wifi:wifi firmware version: 91b9630
I (27761) wifi:wifi certification version: v7.0
I (27761) wifi:config NVS flash: disabled
I (27761) wifi:config nano formating: disabled
I (27771) wifi:Init data frame dynamic rx buffer num: 32
I (27771) wifi:Init static rx mgmt buffer num: 5
I (27781) wifi:Init management short buffer num: 32

assert failed: block_locate_free tlsf.c:566 (block_size(block) >= size)

Backtrace: 0x40081ace:0x3ffb47f0 0x40092419:0x3ffb4810 0x40098b4d:0x3ffb4830 0x40096d65:0x3ffb4950 0x400968b9:0x3ffb4970 0x400821ae:0x3ffb4990 0x400821c1:0x3ffb49c0 0x400834c0:0x3ffb49e0 0x4015d9e9:0x3ffb4a00 0x40127f2c:0x3ffb4a30 0x4012814f:0x3ffb4a50 0x4012bd0a:0x3ffb4a80 0x401287e7:0x3ffb4aa0 0x40117082:0x3ffb4ac0 0x40117188:0x3ffb4ae0 0x401155fe:0x3ffb4b10 0x4009a411:0x3ffb4b30
0x40081ace: panic_abort at C:/Espressif/esp-idf-v5.1.2/components/esp_system/panic.c:452

0x40092419: esp_system_abort at C:/Espressif/esp-idf-v5.1.2/components/esp_system/port/esp_system_chip.c:84

0x40098b4d: __assert_func at C:/Espressif/esp-idf-v5.1.2/components/newlib/assert.c:81

0x40096d65: block_locate_free at C:/Espressif/esp-idf-v5.1.2/components/heap/tlsf/tlsf.c:566
 (inlined by) tlsf_malloc at C:/Espressif/esp-idf-v5.1.2/components/heap/tlsf/tlsf.c:1004

0x400968b9: multi_heap_malloc_impl at C:/Espressif/esp-idf-v5.1.2/components/heap/multi_heap.c:207

0x400821ae: heap_caps_malloc_base at C:/Espressif/esp-idf-v5.1.2/components/heap/heap_caps.c:176

0x400821c1: heap_caps_malloc at C:/Espressif/esp-idf-v5.1.2/components/heap/heap_caps.c:197

0x400834c0: esp_coex_common_malloc_internal_wrapper at C:/Espressif/esp-idf-v5.1.2/components/esp_coex/esp32/esp_coex_adapter.c:125

0x4015d9e9: esf_buf_alloc_dynamic at ??:?

0x40127f2c: esf_buf_setup_static at ??:?

0x4012814f: esf_buf_setup at ??:?

0x4012bd0a: pp_attach at ??:?

0x401287e7: ic_init at ??:?

0x40117082: wifi_lmac_init at ??:?

0x40117188: wifi_init_process at ??:?

0x401155fe: ieee80211_ioctl_process at ??:?

0x4009a411: ppTask at ??:?
```

### More Information.

i tried many things but it is not solving my issue and what i think is when the deinit function is not properly deinitlizing or free the allocated memory during the init phase

hamza2657, 2023-12-05:
Ghg

hamza2657, 2023-12-05:
The main issue is heap corruption and as long as in my code i didnt use heap else to store wifi configuration. But the provided library i think either they are not properly deinitialize the driver or the other case is which i figured out is the order to call the apis matter as for some hit and trial method i think its achievable to enable and disable  both the wifi and ble on user request without being needed to restart the mcu or the worse you can put the mcu to deep sleep. But there should be a straight forward instructions in the api guides or documentation to do the both as it does for the initialization.

hamza2657, 2023-12-21:
Hi @hansw123 i figured out how to do proper initialization and more importantly deinit the peripheral. So the problem doesn't exist on my side now.

hamza2657, 2023-12-21:
I found a book esp32-c3 IoT projects by espressif [GitHub](https://github.com/espressif/book-esp32c3-iot-projects) and [https](https://www.esp32.com/bookc3). The book covers everything in great detail.[image]

One thing i want to add here is that you dont have to create and register event handler on the each iteration of enabling and disabling the wifi. Only once in a full life of program will do the job fine.

closed 2023-12-21 by hamza2657

hansw123, 2023-12-21:
I'm glad you solved the problem and thank you for recognizing the esp32c3 book!