espressif/esp-idf #12711: unhandled double exception ESP32-S2 IDF 5.1.2 what is the cause in a previous well running project on IDF 5.0.1?

ralphde, 2023-12-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.4 and 5.1.2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

should work out of the box

### What is the actual behavior?

The following start up ends in extensa_vectors.S line 564 _DoubleExceptionVector:.

Previously the code was running fine until IDF 5.0.1, stopped loading in 5.0.4 and 5.1.2

what should I look at ?

Bootloader issue:

`ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x2138
load:0x4004b000,len:0x4
load:0x4004b004,len:0xc30
load:0x4004f000,len:0x3544
SHA-256 comparison failed:
Calculated: 7b4bf3281ac986894c811941b66f9443d4bc32bfd5862f7c1770019daafa3070
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004b204
[0;32mI (43) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader[0m
[0;32mI (43) boot: compile time Dec  2 2023 23:05:15[0m
D (43) bootloader_flash: non-XMC chip detected by SFDP Read (00), skip.[0m
D (49) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)[0m
[0;32mI (56) boot: chip revision: v0.0[0m
D (60) boot.esp32s2: magic e9[0m
D (63) boot.esp32s2: segments 04[0m
D (66) boot.esp32s2: spi_mode 02[0m
D (70) boot.esp32s2: spi_speed 0f[0m
D (73) boot.esp32s2: spi_size 02[0m
[0;32mI (76) boot.esp32s2: SPI Speed      : 80MHz[0m
[0;32mI (81) boot.esp32s2: SPI Mode       : DIO[0m
[0;32mI (86) boot.esp32s2: SPI Flash Size : 4MB[0m
D (91) boot: Enabling RTCWDT(9000 ms)[0m
[0;32mI (94) boot: Enabling RNG early entropy source...[0m
D (100) bootloader_flash: rodata starts from paddr=0x00008000, size=0xc00, will be mapped to vaddr=0x3f000000[0m
D (110) boot: mapped partition table 0x8000 at 0x3f008000[0m
D (115) flash_parts: partition table verified, 4 entries[0m
[0;32mI (121) boot: Partition Table:[0m
[0;32mI (124) boot: ## Label            Usage          Type ST Offset   Length[0m
[... 94 lines cut ...]
CommonProgramW6432: C:\Program Files\Common Files
ComSpec: C:\Windows\system32\cmd.exe
IDF_MAINTAINER: 1
TEMP: C:\Users\ralph\AppData\Local\Temp
SystemRoot: C:\Windows
HOMEDRIVE: C:
USERPROFILE: C:\Users\ralph
TMP: C:\Users\ralph\AppData\Local\Temp
CommonProgramFiles(x86): C:\Program Files (x86)\Common Files
NUMBER_OF_PROCESSORS: 16
IDF_PYTHON_ENV_PATH: C:\Users\ralph\.espressif\python_env\idf5.1_py3.12_env
ESP_ROM_ELF_DIR: C:\Users\ralph\.espressif\tools\esp-rom-elfs\20230320\

Operating System: windows 10
Java Runtime Version: 17.0.9+9
Eclipse Version: 4.29.0.v20230903-1000
Eclipse CDT Version: 11.3.0.202307110141
IDF Eclipse Plugin Version: 2.11.1.202310270725
ESP-IDF v5.1.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.12.0
```

### More Information.

Same code works fine on IDF 5.0.1. Problem started with 5.0.4

ralphde, 2023-12-05:
@0xjakob how can we provide our code? obviously not via here. we need an urgent solution

0xjakob, 2023-12-05:
@ralphde If you're OK sharing it privately without NDA, then you can send it to the email address of an Espressif engineer, which would be jakob.hasse@espressif.com for me (you can see the email address in the logs of IDF). I will also forward it to the relevant colleagues. If you can't share without NDA, then you would need to contact Espressif sales: sales@espressif.com.

ralphde, 2023-12-06:
@0xjakob no this is all what falls out of UART0 at startup. The code lands in extensa_vectors.S line 564 _DoubleExceptionVector before restrart. That basically loops due to setting in sdkconfig.

ESP-Marius, 2023-12-18:
Issue seems to be in the cache init code, where we actually do not properly handle `ESP32S2_DATA_CACHE_0KB`. It falls back to configuring cache the same way as `ESP32S2_DATA_CACHE_16KB` 

We'll fix this.

ralphde, 2024-01-02:
it seem to be much more problems with memory management including partition handling

ralphde, 2024-01-02:
any progress on this?

Icarus113, 2024-01-04:
It's because 0 size dcache condition is not configured well. It can be reproduced on master as well. We'll fix it on master then bring the fix to previous release versions.

You may try this for now:

[s2_dcache_0_size_v5.txt](https://github.com/espressif/esp-idf/files/13826100/s2_dcache_0_size_v5.txt)

ralphde, 2024-01-04:
OK I understood that before, however, even with 8 kb dcache I have further
issues which may be related to data chashing as well, like global variables
do not work as expected and the NVS component does as well cause panicing.

See my last issue posts, can these issue be linked to a cashing issue in
general?

Best Regards

Mit freundlicher Empfehlung

Ralph Deffke

Icarus113, 2024-01-05:
This sounds like another issue. You may follow-up it in https://github.com/espressif/esp-idf/issues/12917 you posted.

The crash in this issue is caused by the cache wrong configurations. This issue will be closed after the corresponding commit is synced here.

Feel free to reopen and leave comments.

closed 2024-01-08 by espressif-bot