espressif/esp-idf #12716: The ESP crashes when I merge two projects.

shlomo-edu, 2023-12-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have two projects that are derived from the examples:
1. ble/gatt_server
2. hid/hid_host_example
Each of them works OK.
I copied the: hid_host_example.c to the gat_server/main folder, and also copied the usb/*.h files from the hid project to the ble project (so it can be compiled).
I left the hid_host_example.c unchanged, and only changed the app_main() name to ble_server() so only the hid will be called.
but now it crashes after it loads (and then crashes in an endless loop).
Here is the log output.
I looks to me that I need to copy more c files from the hid project.
How do I know what files?

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375825
0x40375825: esp_restart_noos_dig at C:/Users/shlomoa/esp/esp-idf/components/esp_system/esp_system.c:64 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x165c
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2d9c
entry 0x403c9900
I (29) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (29) boot: compile time 13:32:06
I (29) boot: chip revision: v0.1
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=22e8ch (142988) map
I (123) esp_image: segment 1: paddr=00032eb4 vaddr=3fc96d00 size=0370ch ( 14092) load
I (126) esp_image: segment 2: paddr=000365c8 vaddr=40374000 size=09a50h ( 39504) load
I (137) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8178ch (530316) map
I (233) esp_image: segment 4: paddr=000c17b4 vaddr=4037da50 size=09298h ( 37528) load
I (249) boot: Loaded app from partition at offset 0x10000
I (249) boot: Disabling RNG early entropy source...
I (261) cpu_start: Pro cpu up.
I (261) cpu_start: Starting app cpu, entry point is 0x40375370
0x40375370: call_start_cpu1 at C:/Users/shlomoa/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (275) cpu_start: Pro cpu start user code
I (275) cpu_start: cpu freq: 160000000 Hz
I (275) cpu_start: Application information:
I (278) cpu_start: Project name:     gatt_server_demos
I (284) cpu_start: App version:      v5.1.2-dirty
I (290) cpu_start: Compile time:     Nov 29 2023 13:31:16
I (296) cpu_start: ELF file SHA256:  d8da97eaa2766bfd...
I (302) cpu_start: ESP-IDF:          v5.0.2-dirty
I (307) cpu_start: Min chip rev:     v0.0
I (312) cpu_start: Max chip rev:     v0.99 
I (317) cpu_start: Chip rev:         v0.1
I (321) heap_init: Initializing. RAM available for dynamic allocation:
I (329) heap_init: At 3FC9E2B8 len 0004B458 (301 KiB): DRAM
I (335) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (341) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (348) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (355) spi_flash: detected chip: gd
I (358) spi_flash: flash io: dio
W (362) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (376) coexist: coexist rom version e7ae62f
I (381) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (401) ESP32_MODULE: HID Host example
I (431) ESP32_MODULE: USB shutdown
I (431) BT_INIT: BT controller compile version [80abacd]
I (431) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (471) system_api: Base MAC address is not set
I (471) system_api: read default base MAC address from EFUSE
I (471) BT_INIT: Bluetooth MAC: 34:85:18:77:3d:12

I (501) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (511) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

W (511) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
I (521) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (521) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44

I (531) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (531) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (541) GATTS_DEMO: the gatts demo char length = 3

I (551) GATTS_DEMO: prf_char[0] =11

I (551) GATTS_DEMO: prf_char[1] =22

I (561) GATTS_DEMO: prf_char[2] =33

I (561) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44

I (571) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (581) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (581) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44

ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x42009dcb
0x42009dcb: usb_lib_task at C:/Espressif/frameworks/esp-idf-v5.1.2/examples/bluetooth/bluedroid/ble/gatt_server/main/hid_host_example.c:204 (discriminator 1)

file: "./main/hid_host_example.c" line 204
func: usb_lib_task
expression: usb_host_uninstall()

abort() was called at PC 0x4037ed37 on core 0
0x4037ed37: _esp_error_check_failed at C:/Users/shlomoa/esp/esp-idf/components/esp_system/esp_err.c:47

Backtrace: 0x40375e86:0x3fca4050 0x4037ed41:0x3fca4070 0x40384be2:0x3fca4090 0x4037ed37:0x3fca4100 0x42009dcb:0x3fca4130 0x40381b89:0x3fca4160
0x40375e86: panic_abort at C:/Users/shlomoa/esp/esp-idf/components/esp_system/panic.c:423

0x4037ed41: esp_system_abort at C:/Users/shlomoa/esp/esp-idf/components/esp_system/esp_system.c:153

0x40384be2: abort at C:/Users/shlomoa/esp/esp-idf/components/newlib/abort.c:38

0x4037ed37: _esp_error_check_failed at C:/Users/shlomoa/esp/esp-idf/components/esp_system/esp_err.c:47

0x42009dcb: usb_lib_task at C:/Espressif/frameworks/esp-idf-v5.1.2/examples/bluetooth/bluedroid/ble/gatt_server/main/hid_host_example.c:204 (discriminator 1)

0x40381b89: vPortTaskWrapper at C:/Users/shlomoa/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: d8da97eaa2766bfd

roma-jam, 2023-12-04:
1. Thanks. Just a couple of things, regarding the file: 
- Please, pay attention that `GPIO_NUM_0` is used twice (redefined): as a `APP_QUIT_PIN` and as `GPIO_INPUT_IO_0`.
- The logic to check the `APP_QUIT_PIN` with `while (gpio_get_level(APP_QUIT_PIN) != 0)` is not fully correct, when we use `portMAX_DELAY`. Please, refer the up-to-date example for HID: [hid_host_example.c, line 466](https://github.com/espressif/esp-idf/blob/03414a15508036c8fc0f51642aed7a264e9527df/examples/peripherals/usb/host/hid/main/hid_host_example.c#L466C1-L466C1)

2. Please, refer the links in this comment https://github.com/espressif/esp-idf/issues/12712#issuecomment-1837874535. There is a full information and links to the docs, that will help you to get everything right. 

3. Does the examples crash without merging? If no, I think there is a chance to that something was merged not clearly during  examples merging. 

Please, check everything and if there will be more questions, do not hesitate to ask.

roma-jam, 2023-12-04:
Hi @shlomo-edu, 

Sure, there might be some potential problems when the GPIO which is used for app closing handled in a wrong way . Based on the `hid_host_example.txt` you have shared, the code is from the previous version of HID Host example. Please, refer to the last changes regarding the `APP_QUIT_PIN` changes in the commit here: https://github.com/espressif/esp-idf/commit/375907299ea5bd69ee229c07a2fb7bebfa65d354

Also, please pay attention that current version of HID Driver from idf-extra-components (v1.0.1) opening the HID device after attaching, which means that if there is a call to close the driver and uninstall the `usb_host` library based on the `APP_QUIT_PIN` state while USB HID Device still connected to the ESP host, there could be a potential error during `hid_host_uninstall()`.
So, be sure that there is no connected devices while you want to close the HID driver and uninstall the `usb_host` library.
That is not your current case, but when the same `GPIO_NUM_0` is used twice  - it could be achieved easily. 

Anyway, the last point is only for the HID driver and we are going to change that logic in the next version of HID Driver (https://github.com/espressif/idf-extra-components/pull/214). 

Try to cleanup the GPIO definitions or at least make them independent. 
If there will be anything else, feel free to return with questions.

shlomo-edu, 2023-12-04:
I found the problem.
The problem was that I had this line:
`            printf("\r\=== nSending command %x to microlog ===\r\n\r\n",command);
`
And the \= is unprintable character. should be \n====
Thank you very much for your help.

closed 2023-12-04 by shlomo-edu

shlomo-edu, 2023-12-05:
Hi Roma,

Thanks for all your help.
Now that we have solved all the major problems (communication with the BLE and the HID), I have several questions regarding the ESP product itself.
You may not know the answer for them because it is not related to the firmware, so maybe you can refer the questions to someone else in your company.

  1.  We have purchased the ES32-S3 WROOM FREENOVE because you told me that because it will solve the “host” problem till you fix your example to handle this problem. Do you have a version that can be used with ES32-S3 now?

  2.  We purchased the DevKit, what is the difference between the DevKit and the regular product, and how do we purchase the regular product.

  3.  We see that in all the versions of the ESP32 that we have, the USB are lined with the PCB board (the plastic), but we need them to be a little "stand out” so we can put it in a box (see the picture at the bottom).

  4.  Do you know of any company that sell this product in a box with a battery (a complete product)?

Thanks,
Shlomo.