espressif/esp-idf #12719: sdmmc example doesn't work in esp32s3usb-otg

suryasid09, 2023-12-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3775-gb4268c874a

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3 USB OTG

### Power Supply used.

USB

### What is the expected behavior?

The example "sd-mmc" should be working without any changes in esp32s3-usb-otg development board.

### What is the actual behavior?

Even if I enable pull-up specifically by below code. It doesn't seem to work. 

```
      gpio_config_t io_conf_34 = {
          .pin_bit_mask = (1ULL<< slot_config.d3),
          .mode = GPIO_MODE_INPUT,
          .intr_type = GPIO_INTR_DISABLE ,
          .pull_up_en = GPIO_PULLUP_ENABLE,
          .pull_down_en = GPIO_PULLDOWN_DISABLE
      };
      gpio_config(&io_conf_34);
```

[image]

### Steps to reproduce.

I ran the example provided in the IDF version stated above.
I tried giving enabling internal pull-up, still, it didn't work.

igrr, 2023-12-05:
@suryasid09 The board does have SD card interface, which is usable with SDMMC peripheral: https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-otg/user_guide.html#sd-card-interface. Are you setting the GPIOs correctly?

suryasid09, 2023-12-05:
I already changed the GPIO pin as described in the website but it doesn't work. 
Also, isn't SDMMC and SDIO different?  I got confused because it doesn't write SDMMC explicitly. 

[image]

igrr, 2023-12-06:
I have tried this example with the same settings as in your screenshot, and they work on my side with both `master` and `release/v5.2` branches.

Could you please change the log level to "Debug" and post the log in case when the card initialization fails?

igrr, 2023-12-06:
@feifei112587 Your issue is about a different example (sdio), not related to the topic of the issue. Please open a separate issue for your case.

feifei112587, 2023-12-06:
I have opened a new issue, but there is no answer

adokitkat, 2024-07-22:
Cannot reproduce and no response, closing.

closed 2024-07-22 by adokitkat