espressif/esp-idf #12722: SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY depends on SPIRAM_USE_MALLOC

ftab, 2023-12-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Reporting as per https://github.com/espressif/esp-idf/issues/9011#issuecomment-1837046892

See https://github.com/espressif/esp-idf/blob/master/components/esp_psram/esp32/Kconfig.spiram#L265

I cannot enable task stack in SPIRAM without also enabling SPIRAM allocation by malloc() (instead of only through heap_caps_malloc). However, enabling SPIRAM allocation by malloc() causes all sorts of other things to prefer internal RAM instead of SPIRAM as I want, and my internal RAM is exhausted before I finish booting.

@igrr is it as simple as just changing that line to `depends on SPIRAM`?

0xjakob, 2023-12-05:
@ftab Which chip and version of ESP-IDF are you using?

And what do you mean by
> However, enabling SPIRAM allocation by malloc() causes all sorts of other things to prefer internal RAM instead of SPIRAM as I want,...

Without `SPIRAM_USE_MALLOC` *disabled*, `malloc()` should only allocate internal memory. When enabled, `malloc()` should allocate memory above a certain size on SPIRAM.

ftab, 2023-12-05:
@0xjakob thanks for the quick response!

> @ftab Which chip and version of ESP-IDF are you using?

ESP32, 5.1.2

> And what do you mean by the following?
> 
> > However, enabling SPIRAM allocation by malloc() causes all sorts of other things to prefer internal RAM instead of SPIRAM as I want,...
> 
> Without `SPIRAM_USE_MALLOC` _disabled_, `malloc()` should only allocate internal memory. When enabled, `malloc()` should allocate memory above a certain size on SPIRAM.

When I enable SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY on the default ESP-IDF 5.1.2, I have to set SPIRAM_USE to SPIRAM_USE_MALLOC and my application fails to boot because it runs out of internal RAM. I don't know exactly why yet. I have checked that BT_ALLOCATION_FROM_SPIRAM_FIRST is true, SPIRAM_TRY_ALLOCATE_WIFI_LWIP is true, and all other similar options are set to true. I haven't fully investigated, there's a lot going on in my application (ESP-ADF, WiFi, Bluetooth A2DP/AVRCP/SPP, Bluetooth LE, esp console, esp rainmaker/insights, audio DSP, etc.). My application only successfully boots when SPIRAM_USE is set to SPIRAM_USE_CAPS_MALLOC, but then that disables SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY.

> Edit: Note that this behavior for allocating FreeRTOS objects [changed in version 5.1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.1/system.html?highlight=spiram_use_malloc). Hence your IDF version is of interest, too.

Yes, I saw this and this is the reason I wanted to upgrade to this version of ESP-IDF, so I could use task stack in external memory and free up some internal memory. But in the default esp-idf 5.1.2 it is not possible to use task stack in external memory when SPIRAM_USE method is SPIRAM_USE_CAPS_ALLOC

I have made a fork of 5.1.2 that changes SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY to depend on SPIRAM instead of SPIRAM_USE_MALLOC, and the application seems to work, but I don't know enough to know whether this is safe.

ftab, 2023-12-18:
The only issue I have had so far with this arrangement is that esp-idf/components/spi_flash/cache_utils.c is still blissfully unaware of SPIRAM being a valid location for task stack:

```c
static inline bool esp_task_stack_is_sane_cache_disabled(void)
{
    const void *sp = (const void *)esp_cpu_get_sp();

    return esp_ptr_in_dram(sp)
#if CONFIG_ESP_SYSTEM_ALLOW_RTC_FAST_MEM_AS_HEAP
        || esp_ptr_in_rtc_dram_fast(sp)
#endif
    ;
}
```

Thus anything calling spi_flash_disable_interrupts_caches_and_other_cpu() which includes
- marking an OTA partition as valid
- seemingly any NVS access?

causes an assert failure and I have to move that task back to internal RAM, meaning I'm almost out again.

jaggi73, 2024-07-04:
Came upon this in a google search and I think I see something related.  I am using the esp-adf and created my own element that uses himem.  By default, adf wants to put element task stacks in SPIRAM as long as it is enabled.  The problem is spi_flash_disable_interrupts_caches_and_other_cpu() fails as soon as I try to map himem with the task stack in SPIRAM.  I had to specify this particular task stack to be in internal memory and then things work as they should

closed 2024-08-21 by espressif-bot

0xjakob, 2024-08-22:
@ftab We just merged https://github.com/espressif/esp-idf/commit/4c492cc85fec216dc9f20607411fc9d991996882, which allows SPIRAM task stack for statically allocated tasks.

ftab, 2024-08-22:
> @ftab We just merged https://github.com/espressif/esp-idf/commit/4c492cc85fec216dc9f20607411fc9d991996882, which allows SPIRAM task stack for statically allocated tasks.

Thanks! You're my hero