espressif/esp-idf #12723: Xtensa Special Register symbolic names #defines with very short identifiers

nebkat, 2023-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

No excessively short macros defined in framework headers.

### What is the actual behavior?

https://github.com/espressif/esp-idf/blob/master/components/xtensa/esp32/include/xtensa/config/specreg.h
https://github.com/espressif/esp-idf/blob/master/components/xtensa/esp32s2/include/xtensa/config/specreg.h
https://github.com/espressif/esp-idf/blob/master/components/xtensa/esp32s3/include/xtensa/config/specreg.h

Are indirectly included in `<freertos/FreeRTOS.h>`, meaning a large number of very short identifiers are macro replaced to numbers, causing build failures if those identifiers are used elsewhere in code.

### Steps to reproduce.

1. `#include <freertos/FreeRTOS.h>`
2. Use `BR`, `PS`, `DDR`, `SAR`, `MR`, `EPC`, `EPS`, `INTERRUPT`, `ICOUNT`, etc as identifiers.

### Build or installation Logs.

```plain
In file included from C:/Espressif/frameworks/esp-idf-v5.1/components/xtensa/include/xtensa/xtruntime.h:30,
                 from C:/Espressif/frameworks/esp-idf-v5.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:72,
                 from C:/Espressif/frameworks/esp-idf-v5.1/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:59,
                 from C:/Espressif/frameworks/esp-idf-v5.1/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:71,
                 from C:/Users/Nebojsa/Work/Troo/mdf-firmware/troo_idf_components/freertos_cxx/include/freertos/ring_buffer.hpp:6,
                 from C:/Users/Nebojsa/Work/Troo/mdf-firmware/main/include/app/core/event/event_loop.hpp:9,
                 from C:/Users/Nebojsa/Work/Troo/mdf-firmware/main/include/app/main.hpp:9,
                 from C:/Users/Nebojsa/Work/Troo/mdf-firmware/main/src/main.cpp:6:
C:/Espressif/frameworks/esp-idf-v5.1/components/xtensa/esp32s3/include/xtensa/config/specreg.h:86:25: error: expected unqualified-id before numeric constant
   86 | #define PS              230
      |                         ^~~
C:/Users/Nebojsa/Work/Troo/mdf-firmware/troo_idf_components/external/mp-units/mp-units/src/systems/si/include/mp-units/systems/si/unit_symbols.h:494:23: note: in expansion of macro 'PS'
  494 | inline constexpr auto PS = peta<siemens>;
      |                       ^~
```

Dazza0, 2023-12-05:
@nebkat Unfortunately there's not much we can do in this case. The include chain you are seeing is as follows:

`freertos/FreeRTOS.h` <- `portable.h` <- `portmacro.h` <- `<xtensa/config/...h>`

`portmacro.h` requires use of the Xtensa macros because a significant a large number of FreeRTOS porting macros are implemented as inline functions/asm indirectly inside `portmacro.h`. They are implemented inline to increase efficiency.

As an alternative, are you able to `undef` those identifiers?

nebkat, 2023-12-05:
@Dazza0 Thanks for the quick response!

The chain and reasoning make sense, but could we rename the macros, even just to `XT_PS` or `REG_PS`? 

I am able to `#undef`, just it pollutes my library code that isn't strictly used with ESP-IDF.

Dazza0, 2023-12-12:
@nebkat Upon further review, we've decided to rename the `specreg.h` and prefix them with `XT_...`. We'll leave a compatibility header with the old names just in case anyone needs them.

closed 2025-06-23 by espressif-bot