espressif/esp-idf #12728: BLE: Changing the device name take affect only after the PC connects to the device.

shlomo-edu, 2023-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using the gatt_server under BLE.
I have a default device name:
`#define TEST_DEVICE_NAME    "MICRO_BLE_GATEWAY"
esp_err_t set_dev_name_ret = esp_ble_gap_set_device_name(TEST_DEVICE_NAME);
`
Which it calls under the event: 
`ESP_GATTS_REG_EVT`
Every time I connect to a logger, I get an event and I read the logger's name, and then I call
`esp_ble_gap_set_device_name(logger_name)` with the new name.
But
What I see it that the Windows pairing dialog (Bluetooth and other devices) shows only: "MICRO_BLE_GATEWAY" and doesn't change it, even I I close the setting and opens it.
Another bluetooth app that I use from iPhone, changes the device name only after I connect to the device and read something.
So I think that I need to call the esp_ble_gap_set_device_name() only under the  ESP_GATTS_REG_EVT.
So how do I trigger the event, or how do I make it publish the new device name?

zhp0406, 2023-12-06:
The ESP_GATTS_REG_EVT event is triggered by esp_ble_gatts_app_register. However, esp_ble_gap_set_device_name is not bound to the ESP_GATTS_REG_EVT event. You can also call this API in other places.

shlomo-edu, 2023-12-06:
I tried calling `esp_ble_gap_set_device_name(logger_name)` it didn't help.
I also tried the following:
`   esp_ble_gatts_app_unregister(PROFILE_A_APP_ID);
   esp_ble_gatts_app_unregister(PROFILE_B_APP_ID);
   esp_ble_gatts_app_register(PROFILE_A_APP_ID);
   esp_ble_gatts_app_register(PROFILE_B_APP_ID);
`
But is didn't create a new `ESP_GATTS_REG_EVT` event.
So I don't know what to do.
What I see is that if I connect to the ESP with the old name (MICRO_BLE_GATEWAY) than, it changes the name in the list.
This is only in the iPhone, in the PC it doesn't change at all.

shlomo-edu, 2023-12-06:
One more thing.
I tried to solve the problem by calling:
`esp_ble_gap_start_advertising(&adv_params);`
Every one second.
It didn't solve this problem.
But I am left with a question, do I need to advertize every second?

zhp0406, 2023-12-20:
**esp_ble_gap_set_device_name** does not change the content of the advertising data. If you want to modify the advertising data, you can use **esp_ble_gap_config_adv_data_raw**.

zhp0406, 2024-02-28:
@shlomo-edu Has your issue been resolved?

shlomo-edu, 2024-02-28:
I am now using ESP-AT instead of the actual code.
The problem exists also there.
I came to the conclusion that the problem is not in the server (the ESP32 code) but in the clients (the scanner, or the Window),
So I am closing the issue.

closed 2024-02-28 by shlomo-edu