espressif/esp-idf #12730: No trasport-key message in ZigBee HA example

dominiktopyla, 2023-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32D0WDQ6

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I want to sniff communication between 2 esp32-c6 to observe ZigBee message with key-transport key encrypted with 'ZigBeeAliance09' value. First flashed with HA_on_off_switch package and second with HA_on_off_light

### What is the actual behavior?

Now i haven't seen the message with key-transport key encrypted with 'ZigBeeAliance09' value. But i see in idf monitor mode that network is established and I can switch on/off light on another board using switch board.

### Steps to reproduce.

1. Start sniffing communication on selected channel  
2. Flash 1. board with HA_on_off_light
3. Flash 2. board with HA_on_off_switch
4. Observe results in pcap file 

### Debug Logs.

```plain
I (364) sleep: Configure to isolate all GPIO pins in sleep state
I (370) sleep: Enable automatic switching of GPIO sleep configuration
I (377) coexist: coex firmware version: b6d5e8c
I (383) coexist: coexist rom version 5b8dcfa
I (388) app_start: Starting scheduler on CPU0
I (393) main_task: Started on CPU0
I (393) main_task: Calling app_main()
I (403) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (403) phy_init: phy_version 230,c773401,Oct 30 2023,15:07:16
I (443) phy: libbtbb version: 7243671, Oct 30 2023, 15:07:30
I (453) main_task: Returned from app_main()
I (463) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (463) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
I (463) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (473) ESP_ZB_ON_OFF_SWITCH: Start network formation
I (473) ESP_ZB_ON_OFF_SWITCH: Formed network successfully (Extended PAN ID: 40:4c:ca:ff:fe:43:80:a0, PAN ID: 0x1f31, Channel:25)
I (953) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (953) ESP_ZB_ON_OFF_SWITCH: Network steering started
I (13193) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (13213) ESP_ZB_ON_OFF_SWITCH: New device commissioned or rejoined (short: 0x411b)
I (13223) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (13243) ESP_ZB_ON_OFF_SWITCH: Found light
I (13243) ESP_ZB_ON_OFF_SWITCH: Try to bind On/Off
I (13243) ESP_ZB_ON_OFF_SWITCH: Bound successfully!
I (13243) ESP_ZB_ON_OFF_SWITCH: The light originating from address(0x411b) on endpoint(10)
I (195703) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
```

### More Information.

I haven't changed  HA_on_off_light or  HA_on_off_switch. I am using hackrf with GnuRadio to sniff communication. I was able to sniff ZigBee message with key-transport key encrypted with 'ZigBeeAliance09' value for different vendors on setup coordinator end-deviice, so sniffing setup is not an issue. This the screenshot of part of sniffed communication.
[image]

xieqinan, 2024-02-29:
@dominiktopyla ，

The esp-zigbee example employs 'ZigBeeAliance09' as the default global_default_TC_link_key. Could you try capturing the Zigbee packet using [rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) and Wireshark? You can follow the steps outlined in the [documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark).

Additionally, the esp-zigbee SDK offers an API for setting the global_default_TC_link_key using [esp_zb_secur_TC_standard_preconfigure_key_set()](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_secur.h#L125).

xieqinan, 2024-04-15:
@dominiktopyla ，

Regarding the issue, do you have any updates?

closed 2024-10-10 by Alvin1Zhang