espressif/esp-idf #12733: ESP32-C6 Unable to join Phlips Hue zigbee network

Paveldeg, 2023-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using "HA_on_off_light" sample code from the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk), my Hue bridge does not see the light, and I get repeating console messages like this 
`" ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)"`

Is ESP32-C6 generally compatible with Philips Hue Zigbee bridge?

xieqinan, 2024-02-29:
Hello @Paveldeg

This issue will be handled [here](https://github.com/espressif/esp-zigbee-sdk/issues/168). please close this one.

chshu, 2024-03-01:
The issue was addressed by this comment: https://github.com/espressif/esp-zigbee-sdk/issues/168#issuecomment-1952370582

closed 2024-03-01 by chshu