espressif/esp-idf #12734: httpd_req_recv can result in ECONNRESET if server want to return error if a user sent more data than expected

rretanubun, 2023-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My esp-idf codebase is 4.4.6 based

my code is similar to this example: https://github.com/espressif/esp-idf/blob/03414a15508036c8fc0f51642aed7a264e9527df/examples/protocols/http_server/simple/main/main.c#L260-L289

But I want to do something like this:
```
/* An HTTP POST handler with a specific expected input */
static esp_err_t echo_post_handler(httpd_req_t *req)
{
    char buf[100];
    int ret, remaining = req->content_len;

    if (remaining > sizeof(buf) {
        // We get more data than we are expecting on this transaction
        httpd_resp_send_err(req, HTTPD_400_BAD_REQUEST, NULL);
        // ^^^^^^^^^^^^^^^^^ what I see is that this error code is not sent
        // and the client experience ECONNRESET instead
        return ESP_FAIL
    } else {
        /* Read the data for the request */
        if ((ret = httpd_req_recv(req, buf,
                        MIN(remaining, sizeof(buf)))) <= 0) {
            if (ret == HTTPD_SOCK_ERR_TIMEOUT) {
                /* Retry receiving if timeout occurred */
                continue;
            }
            return ESP_FAIL;
        }

        /* Log data received */
        ESP_LOGI(TAG, "=========== RECEIVED DATA ==========");
        ESP_LOGI(TAG, "%.*s", ret, buf);
        ESP_LOGI(TAG, "====================================");
    }
}
```

`CONFIG_HTTPD_ERR_RESP_NO_DELAY` is enabled on my skdconfig. What am I missing?

hmalpani, 2023-12-13:
Hello @rretanubun 
I tried the example and I was able to send the error to the client without any error
PTAL below code and the screenshot:

### Code
```
static esp_err_t echo_post_handler(httpd_req_t *req)
{
    char buf[10];
    int ret = ESP_OK, remaining = req->content_len;

    if (remaining > sizeof(buf)) {
        // We get more data than we are expecting on this transaction
        httpd_resp_send_err(req, HTTPD_400_BAD_REQUEST, NULL);
        return ESP_OK;
    } else {
        /* Read the data for the request */
        while(remaining) {
            if ((ret = httpd_req_recv(req, buf, MIN(remaining, sizeof(buf)))) <= 0) {
                if (ret == HTTPD_SOCK_ERR_TIMEOUT) {
                    /* Retry receiving if timeout occurred */
                    continue;
                }
                return ESP_FAIL;
            }
            httpd_resp_send_chunk(req, buf, ret);
            remaining -= ret;
            /* Log data received */
            ESP_LOGI(TAG, "=========== RECEIVED DATA ==========");
            ESP_LOGI(TAG, "%.*s", ret, buf);
            ESP_LOGI(TAG, "====================================");
        }
    }
    httpd_resp_send_chunk(req, NULL, 0);
    return ESP_OK;
}
```

[image]

hmalpani, 2023-12-27:
Closing this issue. Please feel free to reopen if you have any more updates.
Thanks!

closed 2023-12-27 by hmalpani