espressif/esp-idf #12737: Wifi mesh cannot form a third layer through networking when wifi mesh and ble enable together

fzboffice, 2023-12-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32_DEVKIT_V4

### Power Supply used.

USB

### What is the expected behavior?

WiFi mesh networking forms the third layer

### What is the actual behavior?

Wifi mesh cannot form a third layer through networking
or
The wifimesh network forms the third layer, and devices on the third layer cannot receive or send data.

### Steps to reproduce.

1. I use esp-idf\examples\mesh\internal_communication project.
2. Enable BT in menuconfig.
3. set CONFIG_MESH_AP_CONNECTIONS to 1 in menuconfig.(Only 3 development boards are needed for testing)
4. Init BLE.
5. Call esp_mesh_fix_root(true).
6. Call esp_mesh_set_type(MESH_ROOT);(other development board call esp_mesh_set_type(MESH_NODE)).
7. Start wifi mesh.

### Debug Logs.

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7140
load:0x40078000,len:15500
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080640
I (29) boot: ESP-IDF v5.2-dev-4169-gbf816394d4 2nd stage bootloader
I (29) boot: compile time Dec  6 2023 16:26:19
[... 332 lines cut ...]
I (18677) mesh: [SCAN][ch:1]AP:7, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (18677) mesh: [FAIL][64]root:0, fail:64, normal:0

I (18717) mesh_main: heap: | default 79 kb | internal 80 kb | spiram 0 kb | freeRTOS free heap 79 kb | freeRTOS min free heap 78 kb |
I (18807) mesh: [SCAN][ch:1]AP:7, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (18807) mesh: [FAIL][65]root:0, fail:65, normal:0

I (18937) mesh: [SCAN][ch:1]AP:4, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (18937) mesh: [FAIL][66]root:0, fail:66, normal:0

I (19067) mesh: [SCAN][ch:1]AP:5, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19067) mesh: [FAIL][67]root:0, fail:67, normal:0

I (19197) mesh: [SCAN][ch:1]AP:5, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19197) mesh: [FAIL][68]root:0, fail:68, normal:0

I (19327) mesh: [SCAN][ch:1]AP:4, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19327) mesh: [FAIL][69]root:0, fail:69, normal:0

I (19457) mesh: [SCAN][ch:1]AP:5, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19457) mesh: [FAIL][70]root:0, fail:70, normal:0

I (19587) mesh: [SCAN][ch:1]AP:6, other(ID:0, RD:0), MAP:1, idle:0, candidate:0, root:1, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19587) mesh: [FAIL][71]root:0, fail:71, normal:0
```

fzboffice, 2023-12-06:
Here is the demo.
[internal_communication.zip](https://github.com/espressif/esp-idf/files/13578548/internal_communication.zip)

fzboffice, 2023-12-11:
My project is very urgent,anyone can help me?

zhangyanjiaoesp, 2023-12-11:
we will look into this and feedback to you soon

fzboffice, 2023-12-18:
@zhangyanjiaoesp Hi,has this issue been confirmed?

zhangyanjiaoesp, 2023-12-18:
@fzboffice yes, we can reproduce this problem. However, we do not recommend to use wifi mesh and wifi-BT coexist together. Can you describe what you want to achieve with coexistence and wifi mesh? Maybe there will be a better choice.

fzboffice, 2023-12-19:
@zhangyanjiaoesp Our project is a mesh network with Bluetooth gateway function,Our project has been running steadily on IDF4.4 for 3 years.Recently, one of our nodes needs to use an RGB screen and we have to switch to idf5.x.

zhangyanjiaoesp, 2023-12-19:
@fzboffice Do you enable BT on all mesh device in your project? Or just the root node or the leaf node ?

fzboffice, 2023-12-19:
@zhangyanjiaoesp Yes,every node enable BT.Bluetooth gateway function is mandatory in the project.

zhangyanjiaoesp, 2023-12-19:
Do you enable the mesh PS function?
[image]

When the third node can't connect to the layer2 node, does the layer2 node still works well ? Can you please provide the log for the layer2 node?

fzboffice, 2023-12-19:
@zhangyanjiaoesp PS is disabled.
Sometimes, when the layer3 device is powered on, the layer2 device can receive messages from root, but cannot send messages to root.
Sometimes, when the third layer device is powered on, the second layer device can perform normal IO.
The log of layer2:
`
I (536) cpu_start: Multicore app
I (536) quad_psram: This chip is ESP32-D0WD
I (536) esp_psram: Found 8MB PSRAM device
I (538) esp_psram: Speed: 80MHz
I (542) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (549) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (1031) esp_psram: SPI SRAM memory test OK
I (1043) cpu_start: Pro cpu start user code
I (1043) cpu_start: cpu freq: 240000000 Hz
I (1043) cpu_start: Application information:
I (1051) cpu_start: App version:      release/2023021701-478-g097ab32
I (1059) cpu_start: Compile time:     Dec 19 2023 17:43:34
I (1065) cpu_start: ELF file SHA256:  80685e4602b22c76...
I (1071) cpu_start: ESP-IDF:          v5.3-dev-1043-g8c9e29898f
I (1077) cpu_start: Min chip rev:     v0.0
I (1082) cpu_start: Max chip rev:     v3.99 
I (1087) cpu_start: Chip rev:         v1.0
I (1092) heap_init: Initializing. RAM available for dynamic allocation:
I (1099) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1105) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1112) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1118) heap_init: At 3FFC7398 len 00018C68 (99 KiB): DRAM
I (1124) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1130) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1137) heap_init: At 4009ECCC len 00001334 (4 KiB): IRAM
I (1143) esp_psram: Adding pool of 4032K of PSRAM memory to heap allocator
I (1151) spi_flash: detected chip: generic
I (1155) spi_flash: flash io: qio
I (1160) esp_core_dump_flash: Init core dump to flash
I (1171) esp_core_dump_flash: Found partition 'coredump' @ 690000 65536 bytes
I (1211) esp_core_dump_flash: Core dump data checksum is correct
I (1211) esp_core_dump_flash: Found core dump 42500 bytes in flash @ 0x690000
I (1214) coexist: coex firmware version: c02915e
I (1220) main_task: Started on CPU0
I (1224) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1232) main_task: Calling app_main()
I (1847) gpio: GPIO[2]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1854) button: IoT Button Version: 3.1.3
I (1858) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1898) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
I (1898) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1906) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1915) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1924) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1933) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1943) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1952) gpio: GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1961) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2228) BTDM_INIT: BT controller compile version [fa43201]
I (2229) BTDM_INIT: Bluetooth MAC: 98:cd:ac:b0:57:c2

I (2232) phy_init: phy_version 4790,b35933d,Dec 12 2023,10:52:08
W (2510) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
I (2597) wifi:wifi driver task: 3ffcc1e4, prio:23, stack:3584, core=1
I (2641) wifi:wifi firmware version: 1160f50
I (2642) wifi:wifi certification version: v7.0
I (2642) wifi:config NVS flash: enabled
I (2642) wifi:config nano formating: disabled
I (2646) wifi:Init data frame dynamic rx buffer num: 128
I (2651) wifi:Init static rx mgmt buffer num: 5
I (2655) wifi:Init management short buffer num: 32
I (2660) wifi:Init static tx buffer num: 8
I (2664) wifi:Init tx cache buffer num: 32
I (2667) wifi:Init static rx buffer size: 1600
I (2671) wifi:Init static rx buffer num: 8
[... 331 lines cut ...]

W (59872) wifi:m f null

W (60179) wifi:m f null

W (60486) wifi:m f null

W (60794) wifi:m f null

W (61101) wifi:m f null

W (61408) wifi:m f null

W (61715) wifi:m f null

W (62022) wifi:m f null

W (62330) wifi:m f null

W (62637) wifi:m f null

W (62944) wifi:m f null

W (63251) wifi:m f null
`

zhangyanjiaoesp, 2023-12-21:
we will debug the issue on v5.2 and feedback you soon

fzboffice, 2024-01-02:
@zhangyanjiaoesp Hi,what’s going on with this issue?

zhangyanjiaoesp, 2024-01-03:
@fzboffice we are still debugging this issue

zhangyanjiaoesp, 2024-01-04:
> @zhangyanjiaoesp Our project is a mesh network with Bluetooth gateway function,it has been running steadily on IDF4.4 for 3 years.Recently, one of our nodes needs to use an RGB screen and we have to switch to idf5.x.

@fzboffice How do you use the WiFi Mesh + BT on IDF4.4 ?
1. Is it the same way with v5.2? 
2. Do you enable the MESH PS ? 
3. Do you open the Bluetooth gateway after the mesh network has been set up?
4. Do you use the MDF project?

zhangyanjiaoesp, 2024-01-08:
@fzboffice 
Please replace the wifi lib in `components/esp_wifi/lib/esp32` and retest.
[wifi_lib_32_1bce45a.zip](https://github.com/espressif/esp-idf/files/13859550/wifi_lib_32_1bce45a.zip)

wifi firmware version: 1bce45a

fzboffice, 2024-01-09:
@zhangyanjiaoesp I have a compilation error  

```D:/esp/tool-idf/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a(lmac.o):(.wifi0iram.55+0x30): undefined reference to `dbg_func_idf'
D:/esp/tool-idf/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a(lmac.o): in function `lmacTxFrame':
(.wifi0iram.55+0x1f0): undefined reference to `dbg_func_idf'
D:/esp/tool-idf/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a(lmac.o): in function `lmacProcessCollisions_task':
(.text.lmacProcessCollisions_task+0x36): undefined reference to `dbg_func_idf'
.o):(.wifiextrairam.35+0x18): undefined reference to `dbg_func_idf'
D:/esp/tool-idf/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a(lmac.o): in function `lmacProcessTxComplete':
(.wifiextrairam.35+0xc7): undefined reference to `dbg_func_idf'
D:/esp/tool-idf/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a(lmac.o):(.text.lmacProcessTxTimeout+0x37): more undefined references to `dbg_func_idf' follow'```

zhangyanjiaoesp, 2024-01-09:
will provide a new lib for you

zhangyanjiaoesp, 2024-01-09:
[wifi_lib_32_56aa398.zip](https://github.com/espressif/esp-idf/files/13869246/wifi_lib_32_56aa398.zip)

fzboffice, 2024-01-09:
@zhangyanjiaoesp After my simple testing, this lib works very well. Can you provide the wifi lib for ESP32S3 and test it again

zhangyanjiaoesp, 2024-01-11:
> @zhangyanjiaoesp Everything seems normal so far,thank you for your help.Please merge into V5.2 branch if possible.

yes, we will merged it into v5.2.

fzboffice, 2024-01-12:
@zhangyanjiaoesp Hey,today I discovered that in a 3-layer network, every approximately 1 minute, devices in the second layer will disconnect from the root and immediately reconnect successfully. The reason for disconnecting is 4(ASSOC_ EXPIRE)

zhangyanjiaoesp, 2024-01-16:
> @zhangyanjiaoesp Hey,today I discovered that in a 3-layer network, every approximately 1 minute, devices in the second layer will disconnect from the root and immediately reconnect successfully. The reason for disconnecting is 4(ASSOC_ EXPIRE)

can you provide the logs?

fzboffice, 2024-01-16:
@zhangyanjiaoesp I can't reproduce any more.Once reproduced, I will report the log again.

closed 2024-03-06 by fzboffice